dresden-elektronik/deconz-rest-plugin #2373: New device support request: Heiman HS1SA-N Smoke sensor

Sjonnie2018, 2020-01-24:
I am not sure if this is the proper way to address my issue or request. I have opened an issue sometime last week (issue #2323 ) to get the Heiman HS1SA**-N** working with ConBee II. I have had great support form @ebaauw and @SwoopX but unfortunatley it didn't lead to a solution.

The original HS1SA (is there an original version??) is stated as been supported but the HS1SA-N apparently is not. So I hereby ask kindly and humbly to get this sensor supported :)

Node info:
/home/jan/Pictures/DeConz screenshots/Node info HS1SA-N (no. 3).png

Basic cluster info:
/home/jan/Pictures/DeConz screenshots/Basic Cluster info HS1SA-N (no. 3)

Power configuration cluster info:
/home/jan/Pictures/DeConz screenshots/Power Configuration info HS1SA-N (no. 3).png

Identify cluster info:
/home/jan/Pictures/DeConz screenshots/Identify Cluster info HS1SA-N (no. 3).png

Diagnostics cluster info:
/home/jan/Pictures/DeConz screenshots/Diagnostics Cluster info HS1SA-N (no. 3).png

IAS Zone cluster info:
/home/jan/Pictures/DeConz screenshots/IAS Zone Cluster info HS1SA-N (no. 3).png

IAS WD cluster info:
/home/jan/Pictures/DeConz screenshots/IAS WD Cluster info HS1SA-N (no. 3).png

OTAU cluster info:
/home/jan/Pictures/DeConz screenshots/OTAU Cluster info HS1SA-N (no. 3).png

I tried to get the debug info from deconz but I could not get that to work on my Raspberry Pi 3B+

I hope you can add this sensor to the list of supported devices!

TIA

Greetz,

Sjonnie

ebaauw, 2020-01-24:
Sorry, these screenshots are largely useless, as you didn't read the attributes.  Looking at the screenshots from the other issue, the sensor should already be supported by REST API plugin.  And looking at the name on the node, the REST API plugin has created at least one resource for it.  I don't think there's much more we can do here, but just to be sure, can you list the resource that the REST API plugin has created for this sensor?

SwoopX, 2020-01-24:
@Sjonnie2018 as I can tell from #2323, the device is recognized properly. Please take note, a supported device from the REST plugin perspective does not mean it is visible in Phoscon. Those are decoupled and Phoscon is closed source. I admit this is kinda inconvenient, but only Dresden Elektronik has influence on that.

Since a Phoscon update has already been announced with some improvements, let's hope this also goes for "dynamic" device integration.

Sjonnie2018, 2020-01-26:
@ebaauw : where can I find this resource? I have browsed my Pi and thoroughly went through the DeConz GUI but I can't find what you're asking for. Sorry for my Noobness :-|

@SwoopX : I understand. I have tried to contact the team through the regular channels but haven't had any reply whatsoever. At this time even a simple: sorry, we can't help you would have been nice. Saves me barking up a dead tree ;)
I just read commit #2375 and saw an update for the sensor. Hope it gets merged :-) Tx!

ebaauw, 2020-01-26:
You need to retrieve it through the API, by doing a GET of `/sensors`.

Sjonnie2018, 2020-01-26:
Wish I could :(

Had some reading to do because I did not understand how I could do that. Stumbled upon this guide: https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Installed the REST plugin for FireFox and enabled the "Authenticate App" in the web interface. After I resend the command http://192.168.16.115:8090/api I get this:
[image]

Maybe in the future they develop a tool to upload development skills into one's brains. Until then I am dependent on your knowledge and willingness to help me :|

How can I get the sensor reading?

ebaauw, 2020-01-26:
Don't mess with basic authentication; the username (API key) is specified in the URL.
Try to GET `/api/config`.  This is an unauthenticated request, that doesn't require a username (API key).  If that returns some JSON, at least you know the connection to the deCONZ web server is OK.

To create a username (API key), you need to enable _Authenticate App_ and POST to `/api`, with a body of:

`{"devicetype": "`_application_`#`_host_`"}`

The API will respond with the 10-character API key.  After that, you can issue a GET of `/api/`_apikey_`/sensors`.

I use the `ph` command line utility, that I included with homebridge-hue.  When running `ph createuser`, `ph` creates the API key and stores it in `~/.ph`, so you don't need to specify it with each request.  You'd simply issue `ph get /sensors`.  It also formats the JSON for better readability of the result.

Sjonnie2018, 2020-01-26:
Hi Erik,

Thanks for your continued support!

After some fiddling I got the sensor reading. I copied them her in whole to avoid missing a specific syntax you might be looking for:

`{
"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"eb798a96890a63209bcbd99e7d7923df","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":false,"daylight":true,"lastupdated":"2020-01-26T15:29:15","status":180,"sunrise":"2020-01-26T07:34:35","sunset":"2020-01-26T16:23:14"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:04:f6:0c-01"},
"10":{"config":{"battery":88,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"0c0293c16670b9ffa7a5991cd063cd50","manufacturername":"LUMI","modelid":"lumi.weather","name":"Kelder achter","state":{"humidity":5319,"lastupdated":"2020-01-26T15:21:39"},"swversion":"20161129","type":"ZHAHumidity","uniqueid":"00:15:8d:00:04:60:0a:d1-01-0405"},
"11":{"config":{"battery":88,"on":true,"reachable":true},"ep":1,"etag":"0c0293c16670b9ffa7a5991cd063cd50","manufacturername":"LUMI","modelid":"lumi.weather","name":"Kelder achter","state":{"lastupdated":"2020-01-26T15:21:39","pressure":1012},"swversion":"20161129","type":"ZHAPressure","uniqueid":"00:15:8d:00:04:60:0a:d1-01-0403"},
"12":{"config":{"battery":88,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"e05ea0764e8413eb83affc4d5fb2892b","manufacturername":"LUMI","modelid":"lumi.weather","name":"Voorraadkast","state":{"lastupdated":"2020-01-26T15:22:45","temperature":1877},"type":"ZHATemperature","uniqueid":"00:15:8d:00:04:5c:b2:61-01-0402"},
"13":{"config":{"battery":88,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"e05ea0764e8413eb83affc4d5fb2892b","manufacturername":"LUMI","modelid":"lumi.weather","name":"Voorraadkast","state":{"humidity":4887,"lastupdated":"2020-01-26T15:22:45"},"type":"ZHAHumidity","uniqueid":"00:15:8d:00:04:5c:b2:61-01-0405"},
"14":{"config":{"battery":88,"on":true,"reachable":true},"ep":1,"etag":"e05ea0764e8413eb83affc4d5fb2892b","manufacturername":"LUMI","modelid":"lumi.weather","name":"Voorraadkast","state":{"lastupdated":"2020-01-26T15:22:45","pressure":1011},"type":"ZHAPressure","uniqueid":"00:15:8d:00:04:5c:b2:61-01-0403"},
"15":{"config":{"battery":91,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"734c949ac52fc0758448f60eb1955bc5","manufacturername":"LUMI","modelid":"lumi.weather","name":"Slaapkamer","state":{"lastupdated":"2020-01-26T15:34:41","temperature":1540},"swversion":"20161129","type":"ZHATemperature","uniqueid":"00:15:8d:00:04:18:e3:64-01-0402"},
"16":{"config":{"battery":91,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"734c949ac52fc0758448f60eb1955bc5","manufacturername":"LUMI","modelid":"lumi.weather","name":"Slaapkamer","state":{"humidity":5787,"lastupdated":"2020-01-26T15:34:41"},"swversion":"20161129","type":"ZHAHumidity","uniqueid":"00:15:8d:00:04:18:e3:64-01-0405"},
"17":{"config":{"battery":91,"on":true,"reachable":true},"ep":1,"etag":"734c949ac52fc0758448f60eb1955bc5","manufacturername":"LUMI","modelid":"lumi.weather","name":"Slaapkamer","state":{"lastupdated":"2020-01-26T15:34:41","pressure":1011},"swversion":"20161129","type":"ZHAPressure","uniqueid":"00:15:8d:00:04:18:e3:64-01-0403"},
"2":{"config":{"alert":"none","battery":null,"group":"22949","on":true,"reachable":true},"ep":1,"etag":"eb798a96890a63209bcbd99e7d7923df","manufacturername":"IKEA of Sweden","mode":3,"modelid":"TRADFRI remote control","name":"TRÅDFRI remote control","state":{"buttonevent":1002,"lastupdated":"2020-01-26T12:38:11"},"swversion":"1.2.223","type":"ZHASwitch","uniqueid":"14:b4:57:ff:fe:74:da:4a-01-1000"},

"3":{"config":{"battery":100,"on":true,"reachable":false},"ep":1,"etag":"eb798a96890a63209bcbd99e7d7923df","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0","state":{"fire":null,"lastupdated":"none","lowbattery":null,"tampered":null},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:75:7f:f5-01-0500"},

"4":{"config":{"battery":null,"on":true,"reachable":true},"ep":1,"etag":"26ba61d1ac5c1590bdf7d9a32c44e770","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (2)","state":{"fire":false,"lastupdated":"2020-01-26T14:24:31","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:e7:bc-01-0500"},

"5":{"config":{"battery":100,"on":true,"reachable":true},"ep":1,"etag":"eb798a96890a63209bcbd99e7d7923df","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (3)","state":{"fire":false,"lastupdated":"2020-01-26T13:19:44","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:ec:43-01-0500"},

"6":{"config":{"battery":100,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"8f68cabc7bf668dc343cbec9da80cf63","manufacturername":"LUMI","modelid":"lumi.weather","name":"Woonkamer","state":{"lastupdated":"2020-01-26T15:01:01","temperature":1928},"swversion":"20161129","type":"ZHATemperature","uniqueid":"00:15:8d:00:04:60:0a:cf-01-0402"},
"7":{"config":{"battery":100,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"8f68cabc7bf668dc343cbec9da80cf63","manufacturername":"LUMI","modelid":"lumi.weather","name":"Woonkamer","state":{"humidity":4426,"lastupdated":"2020-01-26T15:01:01"},"swversion":"20161129","type":"ZHAHumidity","uniqueid":"00:15:8d:00:04:60:0a:cf-01-0405"},
[... cut ...] tate":{"lastupdated":"2020-01-26T15:01:01","pressure":1012},"swversion":"20161129","type":"ZHAPressure","uniqueid":"00:15:8d:00:04:60:0a:cf-01-0403"},
"9":{"config":{"battery":88,"offset":0,"on":true,"reachable":true},"ep":1,"etag":"0c0293c16670b9ffa7a5991cd063cd50","manufacturername":"LUMI","modelid":"lumi.weather","name":"Kelder achter","state":{"lastupdated":"2020-01-26T15:21:39","temperature":1587},"swversion":"20161129","type":"ZHATemperature","uniqueid":"00:15:8d:00:04:60:0a:d1-01-0402"}}`

ebaauw, 2020-01-26:
OK, you have `/sensors` resources for three different `Smokesensor-N-3.0` sensors.

Two seem to be working fine (at least their `state` was updated), but `/sensors/3`, with mac address 00:15:8d:00:02:75:7f:f5 doesn't.  Probably it hasn't been enrolled.  Could you check in the GUI the _Cluster Info_ panel for the _IAS Zone_ cluster?  After reading the attributes, the _Zone State_  should be `Enrolled`.  If not, write the RaspBee/ConBee mac address to the _IAS_CIE_Address_ field.

Sjonnie2018, 2020-01-26:
Again... Thank you :-)

Found the MAC address for the stick by removing the /sensors from the URL. I put that in the IAS-CIE address field and pressed "write". After about a minute I get "write failed". I pressed the button on the device during the writing process.

[image]

I will try to reset the device through the deCONZ GUI to see if I can get it reconnected...

Keep you posted!

ebaauw, 2020-01-26:
That’s not the ZigBee mac address of the ConBee. It starts with 00212e and is 16 hex digits long (instead of 12).  Look for the blue node in the GUI or check the value for the other smoke sensors.

Sjonnie2018, 2020-01-26:
Resetting the device provided more information at least. I do now miss some basic information in DeCONZ gui but it might show up later. The result of http://192.168.16.115:8090/api/77C17A564E/sensors is now:

"3":{"config":{"battery":100,"on":true,"reachable":true},"ep":1,"etag":"1e8f61305ec5376a340441140fa20f18","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0","state":{"fire":null,"lastupdated":"none","lowbattery":null,"tampered":null},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:75:7f:f5-01-0500"},"4":{"config":{"battery":null,"on":true,"reachable":true},"ep":1,"etag":"794fba68d915a92c7bdc37cf9e992d73","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (2)","state":{"fire":false,"lastupdated":"2020-01-26T14:24:31","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:e7:bc-01-0500"},"5":{"config":{"battery":100,"on":true,"reachable":true},"ep":1,"etag":"794fba68d915a92c7bdc37cf9e992d73","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (3)","state":{"fire":false,"lastupdated":"2020-01-26T13:19:44","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:ec:43-01-0500"},

Progress :)

Sjonnie2018, 2020-01-26:
> That’s not the ZigBee mac address of the ConBee. It starts with 00212e and is 16 hex digits long (instead of 12). Look for the blue node in the GUI or check the value for the other smoke sensors.

Found the ZigBee MAC address in the GUI. Not sure if I need it at present (see post above)

ebaauw, 2020-01-26:
No, don’t get worried; just enroll them, by writing the address.

Sjonnie2018, 2020-01-26:
Will give it a go tomorrow. Got some other things to do now ;)

Tomorrow I will report back on the results. Thank you again!

Sjonnie2018, 2020-01-27:
I am starting to believe I can only enroll two of these devices. Writing the IAS_CIE_Address on the first two devices took only one try and was completed with the better part of a second. I just can't get the third device (number 4) enrolled. Restarted the Deconz-gui and tried several times with both of the below sequences:

- Press the button on the device and within one second press Write in the gui;
- Press Write in gui and then press the button on the device within 2 seconds.

Both failed.

The result of http://192.168.16.115:8090/api/77C17A564E/sensors is now:
"3":{"config":{"battery":100,"on":true,"reachable":true},"ep":1,"etag":"d464842c717265e80e6c243ce937bfdf","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0","state":{"fire":false,"lastupdated":"2020-01-27T17:16:45","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:75:7f:f5-01-0500"},

"4":{"config":{"battery":null,"on":true,"reachable":false},"ep":1,"etag":"4888be95f045c9ae5f2de9f5cd6fdc9f","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (2)","state":{"fire":false,"lastupdated":"2020-01-26T14:24:31","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:e7:bc-01-0500"},

"5":{"config":{"battery":100,"on":true,"reachable":true},"ep":1,"etag":"39605aa1dba9df9d62c13ba6596a1998","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (3)","state":{"fire":false,"lastupdated":"2020-01-27T17:20:07","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:ec:43-01-0500"},

Sometimes things are not as easy as you want them to be ;-)

Any thoughts?

Greetz,

Sjonnie

ebaauw, 2020-01-27:
Number 4 also hasn't updated `config.battery`.  My best guess: it has dropped off the network.  You might try power cycling it (disconnect/reconnect the battery).  If that doesn't help, reset and re-pair it.

I run deCONZ with the GUI enabled all the time.  It's a bit tricky on reboot, since deCONZ will crash if started before the graphical environment is up.  Use a timer to delay the startup of the deconz-gui service, see #1736.

SwoopX, 2020-01-27:
I'd probably for a reset of all devices as well, just to be sure.

Sjonnie2018, 2020-01-28:
@ebaauw : I power cycled the device and now it is enrolled. In the past 24 hours however something happened and now the other two devices show up as not enrolled. There have been no power outages and I haven't reset any other device. Only thing I have done is switching back to deconz instead of deconz-gui because I couldn't control my lights anymore from Phoscon or Domoticz. So it appears that the connection/communication is not long lasting. Mind boggling....
@SwoopX : That shouldn't be necessary. All (17) other devices are working perfectly. Not loosing connection and communicating fine with the stick.

@ both of you: thank you for your continued support! I really appreciate that!

Greetz,

Sjonnie

SwoopX, 2020-01-28:
> @SwoopX : That shouldn't be necessary. All (17) other devices are working perfectly. Not loosing connection and communicating fine with the stick.

Just to be clear here, I meant only the smoke sensors ;) Really all would be a bit exaggerated and of no use.

Sjonnie2018, 2020-01-29:
LOL! I misunderstood you :)

After 24 hours the other two devices are still not reachable. Would the proposed merge #2380  help get things sorted? I just wondered if I should wait for that?

Greetz,

Sjonnie

SwoopX, 2020-01-30:
I wonder if it may be a good idea to also include the enrollment status in the API? What's the worst case being not enrolled? The alarm goes off when you're not at home and you do not get any notification from you software?

Sjonnie2018, 2020-01-31:
Didn't change anything last 2 days. Out of curiosity I checked the rest api this morning. Apart from the battery on device 4  the results look a lot better:

`"3":{"config":{"battery":100,"on":true,"reachable":true},"ep":1,"etag":"101bee792036b25b6fcf7f70e939df2f","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0","state":{"fire":false,"lastupdated":"2020-01-27T17:16:45","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:75:7f:f5-01-0500"},"4":{"config":{"battery":null,"on":true,"reachable":true},"ep":1,"etag":"101bee792036b25b6fcf7f70e939df2f","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (2)","state":{"fire":false,"lastupdated":"2020-01-28T17:26:48","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:e7:bc-01-0500"},"5":{"config":{"battery":100,"on":true,"reachable":true},"ep":1,"etag":"101bee792036b25b6fcf7f70e939df2f","manufacturername":"HEIMAN","modelid":"SmokeSensor-N-3.0","name":"SmokeSensor-N-3.0 (3)","state":{"fire":false,"lastupdated":"2020-01-27T17:20:07","lowbattery":false,"tampered":false},"swversion":"2000-0001","type":"ZHAFire","uniqueid":"00:15:8d:00:02:92:ec:43-01-0500"},`

popy2k14, 2020-03-03:
I noticed the following:
Before 2.05.74 Update i have paired 5x Heiman sensors with success and all values (battery, ...) are reported frequently.

After updating to 2.05.74 i have paired another two of them.
**The now show up as "Unkown" device inin the light category in phoscon, is this normal behaviour?**

Sadly, non of them reporting the battery percentage until now... have to wait a few hours/days, maybe this will change...

ebaauw, 2020-03-03:
On what version did you pair the other five?  The `/lights` resource is for activating the siren (the _IAS WD_ cluster).  It should have a `type` of _Warning device_.  Support for this was introduced in June 2019, see #1593.

popy2k14, 2020-03-03:
Not so long ago, about a month ago.
I was on the stable release before v74, I think it was  V2_05_72.

Will look tomorrow if it's an warning device.

Is the behavior of the battery percentage normal?

There is a difference on the 5x and 2x. The 5x are Schwager ZHS1 and the 2x are Heiman. 

Could this be issue?

(the Schwaiger are rebranded Heiman, aren't they?)

Thx

SwoopX, 2020-03-03:
> the Schwaiger are rebranded Heiman, aren't they?

Yes, they are.

ebaauw, 2020-03-03:
> There is a difference on the 5x and 2x. The 5x are Schwager ZHS1 and the 2x are Heiman.

The devices are whitelisted on _Model Identifier_.  If that's changed between the Heiman and Schwager, the REST API plugin might treat them differently.

> Is the behavior of the battery percentage normal?

I see no mention of `ZHS1` nor to `Schwager` in the code.  Maybe the REST API plugin didn't setup the binding and attribute reporting for the battery.  Could you list the resource(s) for a Heiman and for a Schwager?

SwoopX, 2020-03-03:
Hm, maybe I was a bit too fast. I also got a Schwaiger lying around. Packaging says "ZHS 20", but it's purely Heiman, also the modelID starting with SMOK. That may be different for the ZHS1 though...

Can you provide a screenshot of the cluster info panel for the basic cluster from deconz GUI?

popy2k14, 2020-03-04:
Sorry guys for not beeing clear.
I meant "Schwaiger ZHS 20" and not "Schwager ZHS 1".
Typed this on my phone and made some typos.

The model id of the 5x working Schwaiger ZHS 20 is: SMOK_YDLV10
The model id of the 2x issue devices Heiman HS1SA-M/HS1SA-N is: SmokeSensor-N-3.0

@ebaauw Is the modelid "SmokeSensor-N-3.0" implemented to be detected as Heiman?

Thanks

SwoopX, 2020-03-04:
Feels like we're missing some sensors here in contrast to the supported devices in de_web_plugin.cpp

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/c69b8323051a90dd78035025ffe174df7370f440/bindings.cpp#L1884-L1895

popy2k14, 2020-03-04:
ok, that makes sense why they will appear other than the 5x "SMOK_YDLV10".

Can you please guide manually througt the binding/reporting settings of those devices, so i can setup the battery percentage reporting?

Thanks

ebaauw, 2020-03-05:
Urs

popy2k14, 2020-03-05:
How can i "fix" the two devices so they working like an "ZHS 20"?

Thank you

SwoopX, 2020-03-05:
To enable binding, add the binding dropbox within deconz GUI via "Panel". Then drop endpoint 01 (configuration tool) as source and the power configuration cluster from your end device as source, then press bind.

[image]

popy2k14, 2020-03-05:
Thanks a lot!
Binded the Power Configuration Cluster and readed manually the battery percentage.
After that i have readed/writed/readed (just to be sure) the Reporting settings for battery percentage which was configured for 1/7200/1.

Everytime when you try to read/write be sure to shortly press the pair button to wake the device.

Will see in the comming days how it goes.

Thanks a lot for support and the commit!

PS.: When will this be available as binary for rpi?
Is there any nightly build or something like this?

SwoopX, 2020-03-05:
> PS.: When will this be available as binary for rpi?
> Is there any nightly build or something like this?

I'm afraid there are no nightlies, but you can always compile the plugin yourself if desired. There is no ETA for the next version, but my gut feeling is that it shouldn't take any longer than 1-2 weeks.

popy2k14, 2020-03-05:
ok, thanks for clearing things up.
Did the manual binding and hope i am good to go :-)

popy2k14, 2020-03-06:
@SwoopX 
The battery report is working. thx for this.

How can i get rid of the "unkown" light which got added for the fire sensor (without breaking it)?

thanks

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

cscashby, 2020-08-03:
@SwoopX you mention 'a few weeks' but I see similar issues with HS1SA-N and would love the binding to be included. Have the changes been merged in a specific version yet? If so can you confirm version numbers?

Mimiix, 2020-08-03:
@cscashby As far as i can tell, it is already there included in this pull: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2532/commits/5f60db804087933e424b5661d0640c1e98459077

If you still have issues, please open a new [Bug report](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Bug+report&template=bug_report.md&title=)