dresden-elektronik/deconz-rest-plugin #2379: deCONZ GUI crashes on `float` value

ebaauw, 2020-01-26:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1654#issuecomment-578460731.

manup, 2020-01-27:
Hmm strange, not much special code wise. Perhaps the second line makes the trouble.

The U32 and float are in a C union.

[image]

Is the debug print visible before crash?

ebaauw, 2020-01-27:
I think it did.  Took deCONZ long enough to crash after that, though.
```
Jan 27 19:38:28 pi2 deCONZ[3030]: 19:38:28:138 float: 0x00000000
...
Jan 27 19:38:36 pi2 deCONZ[3030]: 19:38:36:656 Poll light node Terrace 2
Jan 27 19:38:39 pi2 systemd[1]: deconz-gui.service: Main process exited, code=dumped, status=11/SEGV
Jan 27 19:38:46 pi2 systemd[1]: deconz-gui.service: Unit entered failed state.
Jan 27 19:38:46 pi2 systemd[1]: deconz-gui.service: Failed with result 'core-dump'.
```
On second and third attempt, I didn't see the message, though, and deCONZ crashed quicker after pressing _Read_.

Core dump (of first attempt): https://app.box.com/s/sa0qwgh5l6iibhded5o4wan6luq3w1wi.

manup, 2020-01-31:
Thanks for the core dump, this helped. I've added the missing pieces in the attribute editor to show and write float values in various formats like 1000, 1000.0 and 1e3. Will be available in 2.05.73.

[image]

ebaauw, 2020-01-31:
Thanks.  You might want to check the spelling of precision.  Or is this taken from general.xml?

ebaauw, 2020-02-19:
> Will be available in 2.05.73.

I can now read the attribute normally.  Thanks.

closed 2020-02-19 by ebaauw