dresden-elektronik/deconz-rest-plugin #2380: New device support request: Xiaomi Mijia Zigbee 3.0 Smart Home Light Sensor

sumo-89, 2020-01-26:
Hi!
Please add support for this device. :)

Completely covered "Mesured value" is "0".

[image]
[image]
[image]
[image]

SwoopX, 2020-01-26:
Can you please also provide some screenshots of the basic and the power configuration cluster after the attributes have been read?

I do see we have a new manufacturer code here.

sumo-89, 2020-01-26:
Sure! 
Thank you! :)

[image]
[image]

SwoopX, 2020-01-26:
Thanks. Have you already checked if the sensors are available via REST API? You may check out this fur further guidance https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2373#issuecomment-578511439

Just to highlight this upfront, the sensor will not be visible via Phoscon as it is closed source.

sumo-89, 2020-01-27:
Hi!
Sorry, I tried to follow that thread but I don't get it... :( I use Postman, so I know how get all sensors from REST, but it just doesn't show up. I tried resetting it, restarting deconz etc. I can read the attributes.

If I can see it (like I you can see in the screenshoots above) have I managed to include it or not then? Or is that the issue?

SwoopX, 2020-01-27:
> I use Postman, so I know how get all sensors from REST, but it just doesn't show up.

That's fine then. It tells us that the device indeed isn't supported. Just wanted to doucle check since in the past, device integration was requested although the sensors were there.

wvuyk, 2020-01-27:
[image]
This would have done it :-)

davidclancy, 2020-02-04:
When will this be supported

SwoopX, 2020-02-04:
I'd expect with version .73. Not sure when DE will release it.

LukasA83, 2020-02-17:
I've tested it today with .73. I can connect the sensor and see it in REST API. If I press the button on the sensor and manually click "read" on the Illuminance measurement Cluster it is updating the Measured Value. 
By forcing an update via the button via any interaction in VNC interface, I see no update of the value. Any configuration issue?

davidclancy, 2020-02-17:
I can now connect to the sensor but do not see in the connected sensor list. Any ideas?

Regards

SwoopX, 2020-02-18:
Phoscon? If yes, this will probably not happen any time soon since Phoscon is closed source and devices need to be integrated seperately.

davidclancy, 2020-02-18:
Yes this is using Phoscon via Home Assistant. HA seems to see the sensor but as unknown. although it does not show in the Phoscon list.

eserero, 2020-02-18:
@LukasA83 can you explain more what you did? 
how did you get the button press to refresh the data?
my sensor never refresh the data unless I pressed it and click "read", but shouldn't the sensor report light changes every few seconds without any need for user intervention?

SwoopX, 2020-02-19:
@eserero You need to add the binding in deconz GUI manually. Bind cluster illuminance with the level control cluster of the conbee.

LukasA83, 2020-02-20:
Battery is no reported into rest api, always indicates "null":
```
{
  "config": {
    "battery": null,
    "on": true,
    "reachable": true,
    "temperature": null,
    "tholddark": 12000,
    "tholdoffset": 7000
  },
  "ep": 1,
  "etag": "20307667724a20efdd08ead6e29bb8b3",
  "manufacturername": "LUMI",
  "modelid": "lumi.sen_ill.mgl01",
  "name": "Lightsensor outside garage",
  "state": {
    "dark": true,
    "daylight": false,
    "lastupdated": "2020-02-20T01:09:32",
    "lightlevel": 0,
    "lux": 0
  },
  "swversion": "20191118",
  "type": "ZHALightLevel",
  "uniqueid": "04:cf:8c:df:3c:77:62:02-01-0400"
}
```

SwoopX, 2020-02-20:
Already included automatic binding and battery status in a new pull request.

davidclancy, 2020-02-20:
Still can not see it in Phoscon via Home Assistant

SwoopX, 2020-02-20:
Have you compiled my repo to check it, or...?

zladukas, 2020-02-26:
I can confirm, that device is working fine:

{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true,
    "temperature": null,
    "tholddark": 12000,
    "tholdoffset": 7000
  },
  "ep": 1,
  "etag": "a3264bada36855929d175008da7344be",
  "manufacturername": "LUMI",
  "modelid": "lumi.sen_ill.mgl01",
  "name": "Light sensor",
  "state": {
    "dark": true,
    "daylight": false,
    "lastupdated": "2020-02-26T20:45:50",
    "lightlevel": 6021,
    "lux": 4
  },
  "swversion": "20191118",
  "type": "ZHALightLevel",
  "uniqueid": "04:cf:8c:df:3c:77:xx:xx-01-0400"
}

SwoopX, 2020-02-26:
Sounds great, then I guess we're done here.

davidclancy, 2020-02-28:
works in HA now but still not seen in the Deconz interface in Sensors

SwoopX, 2020-02-28:
> works in HA now but still not seen in the Deconz interface in Sensors

This will not happen unless DE integrates it. Phoscon is closed source.

drthanwho, 2020-03-21:
If it helps I can also confirm that the device works with HA. I can see it as a device (sensor.lightlevel_4) on HA and doesn't show up on deconz integration as expected, which isn't really a problem anyway.

Thanks for adding support :)

jensflorian, 2020-04-21:
I can also confirm this device works with Openhab 2.4.5 and can be accessed via the Deconz REST API. @SwoopX thanks for adding the support!

I don't understand why Dresden won't add this into phoscon. Real values are much better than the built in virtual daylight sensor.

drthanwho, 2020-04-30:
I've been having some issue with the sensor the last two days. I was working for weeks until two days ago where it stopped syncing with HA. Weirdly however looking at the mesh network it still showed up as connected and with last seen status updating frequently. 

After attempting to pair again (holding down button) it updated in HA once and hasn't since.

callebridholm, 2020-05-31:
Anyone got this sensor working in deCONZ? I have tried adding my sensor several times now running deCONZ 2.05.77 but is not receiving any data:

    "98": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0,
            "tholddark": 12000,
            "tholdoffset": 7000
        },
        "ep": 1,
        "etag": "0761fcf84eeb0e6ec003343dd44623f8",
        "lastseen": "2020-05-31T20:12:25.285",
        "manufacturername": "LUMI",
        "modelid": "lumi.sen_ill.mgl01",
        "name": "Light Sensor",
        "state": {
            "dark": null,
            "daylight": null,
            "lastupdated": "none",
            "lightlevel": null,
            "lux": null
        },
        "swversion": "20191118",
        "type": "ZHALightLevel",
        "uniqueid": "04:cf:8c:df:3c:77:3c:74-01-0400"
    },

drthanwho, 2020-05-31:
@callebridholm yeah I've had mine working just fine with deCONZ in Home Assistant. Only once it dropped off and had to pair it again. Apart from that it's working fine.

SwoopX, 2020-05-31:
@callebridholm the binding got probably MIA. You can set it manually, use the `illuminance measurement` cluster for 1) and the `measured value` for 2) (values have been mentioned below as they would have been set by default for your device).

How to manually bind and enable attribute reporting on the example of the current position for windows covering devices

1. Manually bind a cluster
In deconz GUI, select Panels and then Bind dropbox.
For your device as well as the coordinator (blue node), press the most right bullet to expand the available clusters.
For the coordinator: Drag&drop 01 Home Automation Endpoint as destination.
For the device: Select the windows covering cluster as source, also drag&drop it. Take note that this must be a server cluster (colored blue).
Press `bind`.

2. Manually set up attribute reporting
Then you need to set up attribute reporting. In most cases, there are default settings already defined and should be placed into action either after a deconz restart or power-cycling the device. However, if values do not get updated upon the taken action or not at all, attributes reporting has presumably not been set up properly.
Select windows covering cluster of your device, double click on 0x0008 (current position).
On the lower end of the dialog, enter the values
_Min. report interval: 5
Max. report interval: 300
Reportable change: 2000_
then press `write config`.

callebridholm, 2020-06-01:
> @callebridholm the binding got probably MIA. You can set it manually, use the `illuminance measurement` cluster for 1) and the `measured value` for 2) (values have been mentioned below as they would [...]

Hi,
I tried with the first step 1 but have problem to bind them, think I did the correct thing. Tried all four different lines with blue numbers but same problem with timeout. 
[image]

SwoopX, 2020-06-01:
Wake the device by exposing it to a light source and then darken it.

callebridholm, 2020-06-01:
> Wake the device by exposing it to a light source and then darken it.

Exposed to flash light 5 minutes and then in black box 8 hours but still same problems with binding

SwoopX, 2020-06-02:
That is indeed strange. Have you also tried to reset and re-pair already?

callebridholm, 2020-06-03:
> That is indeed strange. Have you also tried to reset and re-pair already?

Yes! Now I tried to via VNC mark it and then choose "Delete selected nodes". The device is then remove but still visible when executing this command in postman:
http://localhost:40850/api/56478788EA/sensors/98

I then restarted deCONZ plugin but then the Light sensor is back. Is it a better way to completely remove a device in deCONZ?

SwoopX, 2020-06-03:
Just physically reset the device. There should be a button somewhere which has to be held for like 5 - 15 secs.

callebridholm, 2020-06-05:
> Just physically reset the device. There should be a button somewhere which has to be held for like 5 - 15 secs.

Tried that several times without luck. Today I tried it at our summer house with a Conbee 1 and fw 26330500 and it work directly and updates lux values rapidly. So my problems is probably very old firmware at my Conbee stick at home, It has fw 260B0500 :)

SwoopX, 2020-06-05:
That might be indeed a valid reason ;) But as you can see, it uses to work.

Mimiix, 2020-06-06:
As OP's original question has been answered, please open a new issue.

closed 2020-06-06 by Mimiix

slurpey, 2021-06-17:
@callebridholm hi, I've tried your newbie instructions and somewhat got things going but nothing ends up showing in Home Assistant or in Node-Red Deconz integration.  I seem to fail to follow your "Select windows covering cluster of your device, double click on 0x0008 (current position)."
[image]

Is this what you mean?  In select windows, it's actually the 0000 box I need to click on?
I've tried this many times, restarted, even updated firmware and it seems it reacts (if I press the physical button, the little node blinks blue)

Apologies, this interface is very scary for a casual user.  Thanks.

callebridholm, 2021-06-17:
> @callebridholm hi, I've tried your newbie instructions and somewhat got things going but nothing ends up showing in Home Assistant or in Node-Red Deconz integration. I seem to fail to follow your "Select windows covering cluster of your device, double click on 0x0008 (current position)."
> [image]
> 
> Is this what you mean? In select windows, it's actually the 0000 box I need to click on? If so, I did try to "write" the interval times etc but it would end up in error.
> I've tried this many times, restarted, even updated firmware and it seems it reacts (if I press the physical button, the little node blinks blue)
> 
> Apologies, this interface is very scary for a casual user. Thanks.
> 
> [image]

Last months when I added some of these sensors it work perfect automatically and you can now see the sensors in Phoscon UI

slurpey, 2021-06-17:
Hmmm. I'll give it another try tomorrow. I have 40 various devices and I only got it to show in the deconz after pairing in phoscon (while it said phoscon couldn't pair). Strange

slurpey, 2021-06-18:
I confirm, this works now in Phoscon.  Thanks @callebridholm 
version: 2.11.05 / 5/9/2021
Firmware: 26660700

If using from HASSIO addon - in Home Assistant, go to Supervisor > Deconz
Look at top of page if you need to upgrade.  This started working after I upgraded to version: 6.9.0