dresden-elektronik/deconz-rest-plugin #2381: Xiaomi Aqara Smart Cube only creates the digital buttonevent sensor, rotation missing

CC1337, 2020-01-26:
Hi,
I have two Xiaomi Aqare smart cubes, bought from different shops. The first one exposes two sensor devices in the REST Plugin as well as in my ioBroker - as expected.
After pairing the second one, only one sensor for the digital button events was created, Deleting it and pairing again did not help.
I compared the model, both MFKZQ01LM, the one with the rotation reading has all other information printed in Chinese.
REST API reading:
`
"109":{
   "config":{
      "battery":91,
      "on":true,
      "reachable":true,
      "temperature":2100
},
   "ep":2,
   "etag":"3fe0f63a33989f2e6df8900f9ee01e42",
   "manufacturername":"LUMI",
   "mode":1,
   "modelid":"lumi.sensor_cube.aqgl01",
   "name":"1OG Büro Zauberwürfel",
   "state":{
      "buttonevent":7000,
      "gesture":0,
      "lastupdated":"2020-01-25T22:47:59"
},
   "type":"ZHASwitch",
   "uniqueid":"00:15:8d:00:03:49:5d:b5-02-0012"
}
`
deCONZ
[image]
Phoscon (no rotation degrees printed here as well, obviously)
[image]

For comparison a "working" one with rotation readings:
[image]
[image]

Not sure how I can debug further, would be great to get some advice in which direction to dig deeper. Not sure if maybe some versions of the cube do not support rotation.

Best, CC

SwoopX, 2020-01-26:
I guess that's a rather common issue unfortunately. I also had to pair it a couple of times to have both values available. Just repeat and check. Make also sure that you not only do some gestures while searchung for sensors, but also do move angle related gestures. I doubt, just enter the sensor search again and do some angle movements.

CC1337, 2020-01-26:
Well, that was easier than expected. Great hint, worked without even deleting it.
Open network, click pair button again, rotate etc., done.
Did not expect it to be so silly, thanks :)

closed 2020-01-26 by CC1337