dresden-elektronik/deconz-rest-plugin #2382: Cannot get DeCONZ start version > 2.05.70, 2.05.69 works

fsandblom77, 2020-01-27:
Having problem starting the newer versions of DeCONZ. Version 2.05.69 is the last one working.
Have tried to install both by APT and also binary .deb package without any success.

**Error:** 
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/bin/deCONZ)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Network.so.5: no version information available (required by /usr/bin/deCONZ)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/bin/deCONZ)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/bin/deCONZ)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/bin/deCONZ)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5: no version information available (required by /usr/bin/deCONZ)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
/usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)

Follow different issues but non of them has solved my problem. Ex #2280

I'm using Stretch (Raspian). RPI 3 with Raspbee. 

**OS**
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I have tried version 2.05.70, 71 and 72. All gives me the same error above.
Any body have a clue how to solve this problem?
Thanks

closed 2020-06-02 by stale[bot]

basti0681, 2021-12-20:
Similar issue with libQt5Gui.so.5.

I am not able to start any version  >2.13.01 (incl. latest release):

`/usr/bin/deCONZ: relocation error: /usr/bin/deCONZ: symbol _ZNK12QFontMetrics17horizontalAdvanceERK7QStringi version Qt_5 not defined in file libQt5Gui.so.5 with link time reference`

OS: Ubuntu 18.04.6 LTS (GNU/Linux 4.14.180-178 armv7l)

Mimiix, 2021-12-20:
@basti0681 Did you install the new dependency?
libqt5qml5

basti0681, 2021-12-20:
Yes, it is:
libqt5qml5 is already the newest version (5.9.5-0ubuntu1.1).

I also compared the dependency via 'dpkg -I' between version 2.13.01 and 2.13.04 and both have the same dependency .

I know starting with 2.13.02 the bases was changed but does that matter?