dresden-elektronik/deconz-rest-plugin #2383: Feature Request: Virtual Device

Alpakat2, 2020-01-27:
I would like it if there is a feature where I can create a virtual device like a light bulb or a plug and use it as a normal device: add it to a group, control it and so on, so that if I change something I get a WebSocket notifications and can send it to a none Zigbee device like a NodeMCU.

esseki, 2020-02-23:
Hi,

did you have a look at CLIP sensors ? https://dresden-elektronik.github.io/deconz-rest-doc/sensors/#changestate

I use a bunch of those virtual sensors and they work pretty well. The only thing that i cannot guarantee 100% is that they actually send Websocket Notifications. But i think they do.

closed 2020-03-30 by Alpakat2

popy2k14, 2020-07-23:
@esseki will they be exposed to alexa, so you can switch with voice command?

Currently running habridge for that. Sadly i have always issues with double devices.

thx