dresden-elektronik/deconz-rest-plugin #2386: Deconz accepts unsupported parameters and reports wrong result

chbndrhnns, 2020-01-28:
I have a light of this type:

```
"13": {
        "etag": "2684fc93b06048bb8e955559d8f8f2a9",
        "hascolor": false,
        "manufacturername": "MLI",
        "modelid": "tint Smart Switch",
        "name": "Treppe",
        "swversion": "2.5.3_r47",
        "type": "On/Off light",
        "uniqueid": "ee:ee:03:02:01:03:75:e0-01"
    }
```

If I send a request like below, it is accepted and a wrong state is reported back.

```
curl --location --request PUT 'http://X/lights/13/state' \
--header 'Content-Type: text/plain' \
--data-raw '{
	"on": true,
	"bri": 100
}'
```

The response is:

```json
[
    {
        "success": {
            "/lights/13/state/on": true
        }
    },
    {
        "success": {
            "/lights/13/state/bri": 100
        }
    }
]
```

But then the light is in fact not switched on unless I remove the 'bri' key.

ebaauw, 2020-01-28:
Deconz sends a _Move to Level (with On/Off)_ command when you specify both `on` and `bri`.  When you specify only `on`, it sends an _On_ command.  It think the light doesn’t support the first command.  You can check by issuing the command from the deCONZ GUI (_Cluster Info_panel for tge light’s _Level Control_ cluster) and see how the light behaves.

chbndrhnns, 2020-01-29:
The reason I discovered that issue is that this light is shown in Home Assistant with the attribute `supported_features: 41`: HA thinks it supports brightness settings and thus sends the value to deconz.
We could also discuss why this is the case but for this issue I wanted to raise the issue that, IMHO, deconz *accepts* the request and the response indicates it has been processed but, in fact, that's wrong.

I believe deconz should either send back an error about the request or silently ignore the `bri` parameter and just perform the state change to on.

The REST API exposes these values:

```json
{
"state": {
        "alert": "none",
        "on": false,
        "reachable": true
    }
}
```

ebaauw, 2020-01-29:
Hm, the state doesn’t show `bri`, so HA is wrong to assume it does support brightness.  And the REST API plugin is wrong to accept it.  Handling of API errors is notoriously bad in deCONZ, so I’m not surprised, but not taking into account that the device doesn’t support `bri` is a more severe bug.

chbndrhnns, 2020-01-29:
Do you know from which message or API call HA derives the supported attributes? I wanted to debug that but was stuck so far.

ebaauw, 2020-01-29:
I don’t know HA, but the REST API provides two means: the `type` and the attributes in the `state` object.  Both are reported when doing a GET of the light resource.

chbndrhnns, 2020-01-30:
Both means you talked about have no indication for me that a brightness setting is supported. Let me file an issue with HA about the wrong value for `supported_features`.

Can we keep this issue open about the error handling? I believe the return values from deconz are incorrect no matter what is wrong in HA.

ebaauw, 2020-01-30:
> Can we keep this issue open about the error handling?

You could try, but eventually the stale bot will wear you down.  See #34.

chbndrhnns, 2020-01-30:
😳

Klagio, 2020-02-28:
> 😳

hi, is your device Muller Light Tint Smart switch, the relay?

chbndrhnns, 2020-02-28:
Yes, it's a relay.

Klagio, 2020-02-28:
> Yes, it's a relay.

Ok, good I just bought it from Amazon. So it is not working with deconz, or it has problems?

chbndrhnns, 2020-02-28:
> > Yes, it's a relay.
> 
> Ok, good I just bought it (ordered) from Amazon. So it is not working with deconz, or it has problems?
> 
> My use would to close a circuit (which triggers the garage door) and stays normally open

It works for me unless you send unsupported commands like written above.

Klagio, 2020-03-10:
> > > Yes, it's a relay.
> > 
> > 
> > Ok, good I just bought it (ordered) from Amazon. So it is not working with deconz, or it has problems?
> > My use would to close a circuit (which triggers the garage door) and stays normally open
> 
> It works for me unless you send unsupported commands like written above.

Hi, I am trying unsuccesfully to pair it: I connected only L and N to main, in the phoscon app I click to add  "switch" then press 5 times the reset button , then ... nothing .. is the above the correct way??

ebaauw, 2020-03-10:
Wired switches (used to smartify dumb lights) are exposed as lights.

chbndrhnns, 2020-03-10:
> Hi, I am trying unsuccesfully to pair it: I connected only L and N to main, in the phoscon app I click to add new "switch", "other", then press 5 times the reset button , then ... nothing .. is the above the correct way??

According to the [manual](http://static.toom.de/produkte/bilder/9240797/bedienungsanleitung_9240797.pdf) (German version), you need to reset the switch first and the try to connect it to deconz. I remember I needed a couple of tries for that, as well.

Klagio, 2020-03-10:
> According to the [manual](http://static.toom.de/produkte/bilder/9240797/bedienungsanleitung_9240797.pdf) (German version), you need to reset the switch first and the try to connect it to deconz. I [...]

In phoson is a LIGHT or a SWITCH? Because I tried with a switch, when back home later I will try add new LIGHT.
can't download the manual: isn't the reset pressing the button 5 times?

chbndrhnns, 2020-03-10:
> n phoson is a LIGHT or a SWITCH? Because I tried with a switch, when back home later I will try add new LIGHT.

In Phoscon, you need to add it as a light.

Klagio, 2020-03-10:
> > n phoson is a LIGHT or a SWITCH? Because I tried with a switch, when back home later I will try add new LIGHT.
> 
> In Phoscon, you need to add it as a light.

Ok, perfect, it added immediately in the phoscon app as a light.

BTW I thought (my mistake if it is not so), I could use it as a relay with dry contact (to drive a NO or NC contact), but I guess its not ..?

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

chbndrhnns, 2020-06-05:
> And the REST API plugin is wrong to accept it. Handling of API errors is notoriously bad in deCONZ, so I’m not surprised, but not taking into account that the device doesn’t support `bri` is a more severe bug.

It is not resolved

chbndrhnns, 2020-06-05:
I am not able to re-open. Who would do it?

Mimiix, 2020-06-05:
I will provide a To-Do label to make sure it gets looked into. Thanks for the quick reply @chbndrhnns !

chbndrhnns, 2020-06-26:
This issue is not resolved.

ebaauw, 2020-06-26:
> This issue is not resolved.

What is the latest version Of deCONZ you tried?  You should now see an error if you try to set `bri` on a `/lights` resource that doesn’t expose it

chbndrhnns, 2020-06-27:
I am using 2.05.75 and I see the exact behavior that I described in my initial post.

chbndrhnns, 2020-07-18:
This issue is not resolved

Mimiix, 2020-07-20:
@ebaauw Is this something for the core?

ebaauw, 2020-07-20:
No it’s the REST API alright.  Should be solved for the `/lights` API in latest releases.

Mimiix, 2020-07-20:
@ebaauw That probably would be https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2849

@chbndrhnns Can you update to atleast .78 and try again?

chbndrhnns, 2020-07-20:
I tested with .78. It works as described by @ebaauw now:

```
curl --location --request PUT "http://192.168.2.111:40850/api/$API_KEY/lights/13/state" \
--header 'Content-Type: text/plain' \
--data-raw '{
        "on": true,
        "bri": 100
}'
[{"error":{"address":"/lights/13/state","description":"parameter, bri, not available","type":6}},{"success":{"/lights/13/state/on":true}}]%
```

closed 2020-07-20 by chbndrhnns