dresden-elektronik/deconz-rest-plugin #2387: Use deconz with wifi but as client not access point.

Smanar, 2020-01-28:
Hello, I know someone with the bug than this one https://github.com/dresden-elektronik/phoscon-app-beta/issues/16#issuecomment-569258542

Everytime deconz is launched he loose the wifi and need to use

```
sudo systemctl stop deconz-wifi
sudo service hdcpcd restart
```

For the wifi works again. But at next restart the service is enabled again, and wifi broken again.

It's something normal, the only way is renaming the deconz-wifi service ?

Qonstrukt, 2020-01-30:
As far as I found yes, that is the only way. Even if you disable the `deconz-wifi` service, it will just be enabled again on the next reboot. Quite annoying as dezonz-wifi itself isn't working correctly anymore.

Smanar, 2020-01-30:
For the moment he have used
>sudo systemctl mask deconz-wifi

But seriously, for me it's something critic. I can't imagine how much people want to use the wifi gateway as client. It's not something "exotic".

k1ng440, 2020-02-20:
this gave me a hard time. thank you @Smanar

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

adbig, 2021-01-29:
Hi,

I confirm that this issue is not resolved.
On a fresh raspbian buster install  with deCONZ package 2.09.01 it's impossible to disable deconz-wifi with systemctl. At the next reboot, the service is enable again. It is a very big problem since it break my Wifi configuration and my gateway is unreachable.

Thank you in advance.

Adrien

SwoopX, 2021-01-30:
This is very far from being impossible. This is not a deconz topic, but general OS knowledge.

`systemctl mask deconz-wifi`

adbig, 2021-01-30:
Yes, it's far from being impossible I mask the service and it works.
I misspoke when I said "it's not possible". I should have said that it is not normal that by simply installing a package on my Debian it automatically break my network configuration. I think there should be a configuration file in which we can enable this service or not, but it should not be enabled by default.