dresden-elektronik/deconz-rest-plugin #2388: Older ikea round button creates a group without ID (or UI has some error)

dakipro, 2020-01-29:
Hi, 
I am on ubuntu with :latest docker and just started adding devices. I can add lights without problems, but when I try to add an older round switch (honesty I didn't know there are two different models, picture attached) the UI does not allow me to click on its group. I tried resetting the gateway, resetting switches and lights and everything, but it just won't work, here are console logs in the browser that appear on click that mention undefined group id

`manual poll
main.js?032385:3 current site: group/lights
main.js?032385:3 goto: control/switch/00:0b:57:ff:fe:c9:ac:7d/group/undefined
main.js?032385:3 current site: group/lights
main.js?032385:3 goto: control/group/1
main.js?032385:3 current site: group/lights
main.js?032385:3 already on: control/group/1
main.js?032385:3 /scene/groupchanged took 0 ms
group.js?600959:3 check FLS-PP lp special groups for 1
group.js?600959:3 fls-pp count 0
`

So the E1810 remote works fine, but E1524 doesn't (I forgot to take a photo without back cover, it has also different battery placement.

[image]

[image]

So if I click on the remote group, it selects the first above it so there is nothing I can assign to the old-remote group

dakipro, 2020-01-30:
I've now tested on windows by resetting the gateway and adding lights and switches again, and again same behavior. I can add switch, but cannot assign anything on its group via GUI. 
I've briefly looked into the database and it looks like all groups and switches have the ID, i can also see switch events in the console. Could it be that it is only the GUI that doesn't work for some reason?
Should I try to assign lights to the switch directly via REST? Anyone on quick guide what to try?
I want to have switches and lights independent of the gateway, this is very crucial tome. But still I would like to use deconz to automate them

Smanar, 2020-01-30:
You are speaking about UniqueID ?

I have ATM a E1524 that works (but include with a previous version)

You can see a difference in json API for theses 2 remote ?

Have you tried the old webapp ( in phosocn/help) ?

Qonstrukt, 2020-01-30:
I'm having the same issue, didn't think much of it, as I just assigned the remotes to self-made groups. But something seems wrong though.

dakipro, 2020-01-30:
@Smanar  I am very new to deconz (as in 2 days testing now) so I am just looking at things that make some sense to me (rather than that knowing what I am talking about). Can you please give me some guidance on how to proceed with testing? I can upload the database here if that helps. I do not get any error message when using UI on the photo, I just noticed some things popup in console (pasted above) and the group not being selected. What caught my eye was this line from console 
 
>  goto: control/switch/00:0b:57:ff:fe:c9:ac:7d/group/undefined 

And I would assume that instead of "undefined" a group ID should be there, as it is when I click on self-created groups. That is why I thought that group perhaps has a wrong ID in database, but in database all look "ok" (as in all groups look similar). 

@Qonstrukt  I can do that as well, but I don't think self-made groups work when the stick is offline, which is a must for me and waf in the house. Or do they? I understood that switch creates a group that you add lights to if you want it to work when deconz stick is offline. 

I will try today after work to add some lights manually to the switch group through the api, perhaps it is just the UI that is struggling with something

Smanar, 2020-01-30:
Ha ok, when you have said "without id" I thought you have see some json without ID.
To use the API https://dresden-elektronik.github.io/deconz-rest-doc/
But yes, I think like you the problem is from the UI.

But have you tried the "old webapp" ?
You can too make group using the api in emergency, but not realy friendly.

dakipro, 2020-01-30:
Thanks @Smanar for the tip about old ui (I didn't even see there is that option). I now connected remote and I can "enter" details on the switch group and I can assign lights to it
(I do not see the new switch there, only the old one)
I can also go to the light and assign it to the switch-group, which is what I did now. 
Since I did this remotely I cannot test or see if it actually did anything with the lights and the switch, and in "new" ui I still cannot access switch-group to see if it did anything, but the light still says it is not added to any group in the new UI. 
I must sit at the actual pc to test it a bit more if it is UI error (which I could work around either with postman or perhaps old ui) or if it is some more serious issue with the ikea-switch support

Qonstrukt, 2020-01-30:
> @Qonstrukt I can do that as well, but I don't think self-made groups work when the stick is offline, which is a must for me and waf in the house. Or do they? I understood that switch creates a group that you add lights to if you want it to work when deconz stick is offline.

I think you're right on that one. Luckily the Raspbee and Deconz installation has been very stable for me, so it hasn't been an issue (yet).
A positive side effect seems to be that lights which wandered of the network somehow get attached again through buttons that ask for that particular lamp.

dakipro, 2020-01-30:
I have tested now @Smanar , and the old UI does not even see the groups created by the old switch, in the old UI I only see the new switch and its group. 
So it seems that there is some issue with old E1524 switches, which I have around 3 in the house. 
My firmware version is 264A0700

ebaauw, 2020-01-30:
Did you update the firmware of the old remotes?  With the new ZigBee 3 firmware, they behave like the new model, not picking a group on factory reset.  I don't think the REST API plugin takes that into account.

dakipro, 2020-01-30:
I did not update it in the last few (maybe even 6) months. I sold ikea gateway, I guess I need ikea gateway to do the update?

ebaauw, 2020-01-30:
No, you can upgrade firmware of IKEA devices using deCONZ.

Does the remote show as a HA or ZLL device in the deCONZ GUI?

dakipro, 2020-01-30:
I do not know what to look for about HA or ZZL, both new and old remote appear with their (same) picture, and on the old one it says 
Vendor IKEA of Sweden
Product TRÅDFRI remote control
Version 2.3.014

and new one has 
Vendor IKEA of Sweden
Product TRÅDFRI remote control
Version 1.2.223

But the photo is the same. From the windows app, only the NWK, IEEE and Server mask are different.

dakipro, 2020-01-30:
I found it, a bit difficult to click... 
The old one is HA, the new one is ZZL. I've googled a bit about the update, is there a summary of all the steps perhaps for ikea devices?
I might also update the lights at the same time, before I start including them all

Smanar, 2020-01-31:
Hooow, there is so much firmware.
Mine is a 1524 but with the firmware 1.2.214.

Ikea device are ZLL with old firmware then HA with new.

Serveur mask look theses one ?
0x000b570000000000
0x14b4570000000000
0x90fd9f00000000000
0xcccccc0000000000

dakipro, 2020-01-31:
I can paste the info here, but not sure what to look for. There are two groups, one 1 and second 26926. So when browser calls this endpoint
http://192.168.1.22/api/D9BDE464B8/groups?_=1580498738015
I get the fallowing:
`{"1":{"action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":[],"etag":"fb26092e0f26364b4aa5de36c1eefb7d","hidden":false,"id":"1","lights":[],"lightsequence":[],"multideviceids":[],"name":"test","scenes":[],"state":{"all_on":false,"any_on":false},"type":"LightGroup"},"26926":{"action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":["5"],"etag":"fb26092e0f26364b4aa5de36c1eefb7d","hidden":false,"id":"26926","lights":["3"],"lightsequence":[],"multideviceids":[],"name":"TRADFRI remote control 5","scenes":[],"state":{"all_on":false,"any_on":false},"type":"LightGroup"}}`

And here are the devices from this endpoint
http://192.168.1.22/api/D9BDE464B8/sensors?_=1580498738043

`{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"fb26092e0f26364b4aa5de36c1eefb7d","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":true,"daylight":false,"lastupdated":"2020-01-31T19:25:19","status":230,"sunrise":"2020-01-31T07:59:33","sunset":"2020-01-31T15:46:32"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:05:3e:33-01"},"4":{"config":{"alert":"none","battery":null,"group":null,"on":true,"reachable":true},"ep":1,"etag":"fb26092e0f26364b4aa5de36c1eefb7d","manufacturername":"IKEA of Sweden","mode":3,"modelid":"TRADFRI remote control","name":"TRÅDFRI remote control","state":{"buttonevent":1002,"lastupdated":"2020-01-30T20:44:09"},"swversion":"2.3.014","type":"ZHASwitch","uniqueid":"00:0b:57:ff:fe:c9:ac:7d-01-1000"},"5":{"config":{"alert":"none","battery":null,"group":"26926","on":true,"reachable":true},"ep":1,"etag":"fb26092e0f26364b4aa5de36c1eefb7d","manufacturername":"IKEA of Sweden","mode":3,"modelid":"TRADFRI remote control","name":"TRÅDFRI remote control Novi","state":{"buttonevent":2002,"lastupdated":"2020-01-30T20:45:01"},"swversion":"1.2.223","type":"ZHASwitch","uniqueid":"14:b4:57:ff:fe:3f:74:71-01-1000"}}`

So here are three switches, one Philips that I assume is the sunrise/sunset, and other two are my switches. one on the position 4 is the old one E1524  (with newer firmware version 2.3.014), and the one on the position 5 is the newer one E1810 with firmware 1.2.223, as seen in the json above. 

Now, I was thinking on upgrading E1810 to the latest firmware, but I guess I should wait until this is resolved. 
I've also attached the database, and while looking at it now it looks like the switches are found two times in the Sensors table. Should this be the case? I've tried resetting and deleting database several times both on win pc this is now running, and also on my ubuntu server/docker.

[zll.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4141091/zll.zip)
 
Any tips where to continue from here?

Smanar, 2020-02-01:
You have 1 group called "test" and another called "TRADFRI remote control".

The first one is not created by the remote ? So this group have a problem or the device integration miss 1 group for the second remote.

And on sensor 4, the remote haven't group too.

All is good for the remote 14:b4:57:ff:fe:3f:74:71 but not for the 00:0b:57:ff:fe:c9:ac:7d.

dakipro, 2020-02-01:
Group test is what I made correct, it is a custom group. 
TRADFRI remote control is created by the remote, not sure which. However, since my first post I have added second remote, and now there are two groups in the UI, for both remotes. One appearing since the first remote, second group appearing since I added the second remote. So the UI is getting three groups from somewhere (2 remotes + 1 test), even though the API lists only two groups. 
Quite confusing to be honest. 
[image]

Smanar, 2020-02-01:
So now all is working fine ?

dakipro, 2020-02-01:
Well no, the problem from the first post is still there. 
I cannot set "old switch with new firmware" to independently work with the lists no matter what I do, which os I use and which UI I try to use. 
in new UI i cannot click on the group, I get error message from my first post. 

And the only reason I purchased deconz is because this is supported (wheres on zigbe2mqtt it is a bit hit and miss, and difficult to set up). 
I also sent email to deconz support few months ago if this is supported, but never got the answer, however by reading on other threads I saw that this works, and it does work but either on old firmware, or on new hardware (with old firmware). 

I do not know how to mark this issue as bug or who to tag here so that this get a bit more attention, as this is one of the features deconz advertises itself to be better then much other cheaper alternatives.

Smanar, 2020-02-01:
Can you show us the new json group (with the working and not working remote) like in the previous post.

dakipro, 2020-02-01:
It is the same, browser is showing three groups, and api http://192.168.1.22/api/CEEA2F4735/config?_=1580568088132 
is replying with two groups

`{"1":{"action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":[],"etag":"e36857a2132d9a8fa5857defad0ae4d2","hidden":false,"id":"1","lights":[],"lightsequence":[],"multideviceids":[],"name":"test","scenes":[],"state":{"all_on":false,"any_on":false},"type":"LightGroup"},"26926":{"action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":["5"],"etag":"9ce73d3da1f276e79fe600b8b96fd6be","hidden":false,"id":"26926","lights":["3"],"lightsequence":[],"multideviceids":[],"name":"TRADFRI remote control 5","scenes":[],"state":{"all_on":true,"any_on":true},"type":"LightGroup"}}`

I've delete browser data, also tried in firefox, exactly same thing happens.
There is also attached database where you can see three groups in database, but not in json you mentioned

Smanar, 2020-02-01:
Ok so it's exctly same logs, even you have 3 groups displayed, you have only 1 remote group, and one of them without group too.

So the problem was here probably since inclusion.
You are using Phoson/switch/add new switch/ikea ....?  IDK if there is a difference but try with "other".

I realy don't see why the 2 remote doesn't works same.

ebaauw, 2020-02-01:
Please remember, a ZigBee group is just a 16-bit number, the group ID, which is included in broadcast messages to the group (sometimes referred to as "groupcast" messages).  There is no group object or anything in ZigBee.

Typically for a (controller) device (like a wireless switch) to send messages to a group, it needs a binding to that group.  A binding is just an entry in the device's binding table, that configures where to send messages to, from a particular endpoint and cluster.  Typically, these bindings are maintained by the REST API plugin, but you can maintain them manually through the _Binding Dropbox_ panel in the deCONZ GUI.

Typically for a (controlled) device (like a light) to respond to messages to a group, it needs to expose a server (blue) _Groups_ cluster, which maintains the list of group IDs that the device responds to.  This cluster provides commands to "join" or "leave" a group, or rather: to add or delete a group to/from the group table.  Typically these entries are maintained by the REST API plugin, but you can maintain them manually through the _Cluster Info_ panel for the _Groups_ server cluster in the deCONZ GUI.

Note that there's no forced relationship between bindings on the controller device and group tables on the controlled devices.  It's perfectly possible for the controller device to send messages to a group that no device has "subscribed" to.  It's also perfectly possible for a controlled device to have "joined" a group that no-one sends messages to.

The REST API plugin maintains `/groups` resources.  These resources contain a `lights` attribute, whose value is an array of `/lights` resource IDs.  Please understand that this does _not_ reflect the ZigBee setup, where a device (light) has added a group ID to its group table.  

> I guess that API is showing groups based on sensors info

No, the REST API plugin maintains the `/groups` resources by querying the group tables from the devices.  When creating a `/groups` resource through the API, without any lights, no ZigBee messages are sent whatsoever.  When adding (deleting) a `/lights` resource ID to (from) the group, the REST API plugin sends an _Add Group_ (_Remove Group_) command to the device corresponding to the `/lights` resource.

The REST API plugin eavesdrops on group messages, to reverse engineer the (button) events from the controller device.  When it sees such a message, it sets `config.group` in the controller's `/sensors` resource to the group ID.

Af far as I understand, on factory reset, ZLL controlling devices pick a random group ID to send messages to.  This group ID is reported on the _ZLL Commissioning_ cluster.  The REST API plugin courteously creates a `/groups` resource (with `devicemembership` linking to the `/sensors` resource ID and with an empty `lights` array) on pairing a ZLL controlling device.

ZHA or ZigBee 3 devices do _not_ pick a random group ID on factory reset - they typically send a broadcast to all devices, until they have a binding to a group.  The REST API plugin should create these bindings and the corresponding `/groups` resource on pairing a ZHA/ZigBee 3 controlling device.

> I realy don't see why the 2 remotes don't works same

The problem with the old model Trådfri remote is that it used to be a ZLL device with the older firmware, but with the recent firmware it has become a ZigBee 3 (ZHA) device.  The REST API plugin still pairs it like a ZLL device, not creating the bindings nor the `/groups` resource.  To fix this is not trivial, as the REST API plugin needs to check the remote's firmware version, but this hasn't yet been read during pairing.  In the meantime, you can remedy the situation manually by:
- Creating bindings from the remote's client (grey) _OnOff_, _Level Control_, and _Color Control_ clusters to a newly picked group ID, using the _Binding Dropbox_ panel in the deCONZ GUI;
- Add this newly picked group ID to a light, using the _Cluster Info_ panel in the deCONZ GUI.  The REST API plugin should create the `/groups` resource, next time it queries the light's group table (I think on power cycling the light or on restarting deCONZ).

> and the UI doesn't trust that but shows somethng more

I don't know Phoscon, but I'm guessing it displays groups for `/groups` resources as well as for `config.group` in `/sensors` resources.

dakipro, 2020-02-01:
Thank you very much @ebaauw for explanation, although to be honest I don't (yet) understand most of it but I understand that some things are just way to difficult. 
But I am still interested in the "you can remedy the situation manually by" part. 
I see the Bind Dropbox panel in the deCONZ app, however everything is grayed there, I can only chose Cluster group and write a number there (would this then be manually created group?)
I also cannot click the buttons no matter what I select or do, here is the screenshot, everything is grayed out, I cannot click on neither source nor destination

[image]

I would really really appreciate if you (or anyone) could write a bit more details about this so that the new user can also do it. Maybe have it on the wiki or something?
If I understand correctly, I make a group, then I bind remotes to the group, then I cluster or add lights, and then life is great. I am just missing practical steps where to click and how to confirm it. 

Btw, one more question, can I create these groups from my pc, and do a "backup", then plug inn the stick into the linux server and do some restore on the docker container? Will everything work the same?

Smanar, 2020-02-01:
You have some informations here https://www.dresden-elektronik.de/funk/software/deconz.html?file=files/dresden-elektronik/content/downloads/dokumente/funktechnik/deCONZ-BHB-en.pdf

First display cluster list on the device you want (the second circle on titlebar)
Then select the cluster you want to bind, for exemple the 0006 (on/off) and use drag and drop feature to the "source" box from the cluster number (the bleu number)
You need to bind the ouput cluster, the gray one, the blue one are input. It reality it's same for application but it can't work if the device are the good output cluster.

And yes, If you make a complete backup, all will be saved, but you need to make the backup, nothing is memorised on usb key.

ebaauw, 2020-02-01:
Check out the user manual for the GUI under the help menu.

dakipro, 2020-02-01:
Thanks @Smanar and @ebaauw , I got some progress :) 
I manage to connect the E1524 HA (firmware 2.3.014) to the light bulb so that it works without the deconz stick by fallowing these steps (writing for my future self):

1. I included both lights and switch (something I did few days ago). 
2. Then I made one "manual group1" group from the web UI. 
3. I found the switch in the deCONZ windows app, clicked on second dot to display all its functionality
4. I opened the Bind Dropbox panel, and drag gray 0006 On/Off item to the Source in the Bind panel 
5. I found the group ID from the UI, by spying its console output (resfapi will also show this ID in the groups list). 
6. I entered the group ID in the Bind panel, clicked on Bind button. 
7. I had to click the remote so that it wakes and accepts the configuration, the panel then said SUCESS above the bind button. 
8. I then went to the light details in the web ui, and assigned my new group for the light. 
9. After few moments, the switch would control the light even if the usb stick is turned of.
The fun part is that this actually fixed the group in the web ui, meaning that I can now click on the switch-group and control the light via web ui

However now the group "manual group1" I made earlier is not clickable anymore from the UI, meaning it is suffering from the same bug I described in the first part. This might affect how homeseer integration works, I am yet to test that part. 

However, the result is there (for now). I can add light to the deconz, I can add switch to it, I can bind them so they work even if the server is offline, and I can control the light from the UI (and hopefully soon homeseer). 

I guess some workaround could be done for this so that it is "one click or less" in the UI? 
Even though this might be a afterthought/temporarily solution, it might make things easier for users, or even reduce potential errors.

Here is how the groups look like from the API now after this
`{"1":{"action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":[],"etag":"e36857a2132d9a8fa5857defad0ae4d2","hidden":false,"id":"1","lights":[],"lightsequence":[],"multideviceids":[],"name":"test","scenes":[],"state":{"all_on":false,"any_on":false},"type":"LightGroup"},"26926":{"action":{"bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":null,"xy":[0,0]},"devicemembership":["5"],"etag":"9ce73d3da1f276e79fe600b8b96fd6be","hidden":false,"id":"26926","lights":["3"],"lightsequence":[],"multideviceids":[],"name":"TRADFRI remote control 5","scenes":[],"state":{"all_on":true,"any_on":true},"type":"LightGroup"},"3":{"action":{"bri":128,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":128,"scene":null,"xy":[0,0]},"devicemembership":["4"],"etag":"78922d4cf283a4a0fb1878aeefc46311","hidden":false,"id":"3","lights":["2"],"lightsequence":[],"multideviceids":[],"name":"Office manual color light group","scenes":[],"state":{"all_on":true,"any_on":true},"type":"LightGroup"}}`

and here are the devices 

`{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"e36857a2132d9a8fa5857defad0ae4d2","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":true,"daylight":false,"lastupdated":"2020-02-01T18:18:09","status":230,"sunrise":"2020-02-01T07:57:12","sunset":"2020-02-01T15:49:12"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:05:3e:33-01"},"4":{"config":{"alert":"none","battery":null,"group":"3","on":true,"reachable":true},"ep":1,"etag":"554606143e6ef72bf6b126a3f1aa4a5c","manufacturername":"IKEA of Sweden","mode":3,"modelid":"TRADFRI remote control","name":"TRÅDFRI remote control","state":{"buttonevent":3002,"lastupdated":"2020-02-01T19:01:52"},"swversion":"2.3.014","type":"ZHASwitch","uniqueid":"00:0b:57:ff:fe:c9:ac:7d-01-1000"},"5":{"config":{"alert":"none","battery":87,"group":"26926","on":true,"reachable":true},"ep":1,"etag":"3689de3637028537794e71fb55550fc4","manufacturername":"IKEA of Sweden","mode":3,"modelid":"TRADFRI remote control","name":"TRÅDFRI remote control Novi","state":{"buttonevent":2002,"lastupdated":"2020-01-30T20:45:01"},"swversion":"1.2.223","type":"ZHASwitch","uniqueid":"14:b4:57:ff:fe:3f:74:71-01-1000"}}`

I see that problematic switch is using group ID 3, which I will not be deleting for now even though I cannot select it from the UI.

tebby69, 2020-02-04:
Hi,
I HAD the same problem.
I solved it by pairing the remote with an light manually: Press the connect-button for some seconds near the light till it blinks, than press the main-button of the remote and switch the light off and on by the remote.
Now you can proceed in the Phoscon-App als usual...

dakipro, 2020-02-04:
I did that with zigbee2mqtt but something was missing, forgot what. 
Do you see both button and light in the Phoscon? And you can control light with phoscon, and control the light via button without the stick?

tebby69, 2020-02-04:
Yes and yes!
And I can remove and add lights in the Phoscon!

dakipro, 2020-04-09:
I now had one group disappear from Deconz ui, its only light has somehow changed the group to another while I was including new switches. I could still see the group in the rest response, and I could see that it still had one switch to it, but it was not showing in the admin. Here is the group ID
`
"26926": {
		"action": {
			"bri": 127,
			"colormode": "hs",
			"ct": 0,
			"effect": "none",
			"hue": 0,
			"on": false,
			"sat": 127,
			"scene": null,
			"xy": [
				0,
				0
			]
		},
		"devicemembership": [
			"5"
		],
		"etag": "deda378b85161d7758b3da908e8ffc72",
		"hidden": false,
		"id": "26926",
		"lights": [],
		"lightsequence": [],
		"multideviceids": [],
		"name": "Office light",
		"scenes": [],
		"state": {
			"all_on": true,
			"any_on": true
		},
		"type": "LightGroup"
	},
`

I fixed it by going to the homeseer JowiHue plugin which did show the group and lights, and I was able to move light back to the original group, and all works fine. 
So... It has to be something with the UI not showing the group properly, as homeseer+JowiHue did see all just fine

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

dakipro, 2020-06-06:
It is not resolved and all suggests that it is a UI bug since other UIs can work with/around it just fine.

Mimiix, 2020-06-26:
@dakipro To follow up: This is a phoscon issue?

dakipro, 2020-06-26:
To be honest I am not quite sure what phoscon vs Deconz is. Last week I purchased one more color light with a switch, I added it to the system, and ALL groups got fckd up and disappeared from the web UI. I had to do a combination of my [previous comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2388#issuecomment-581061042) and homeseer JowiHue plugin, and vnc to connect to the docker UI, and also a json reader to understand what the UI is showing vs what it is receiving from the network (vs what the system is actually having in database). 

And somehow, I do not know how, I got the system working by luck, even though Phoscon/Decons is showing me things differently then homeseer, and all this is different from what the web UI is actually getting from the server when I inspect network traffic from the console. I just had to repeat some assigning to groups, manage to brake my office and hallway lights, then had to do manual group assignment for them as well. 
Now I have three more Ikea lights to add to the system, but I am so reluctant that entire home will go offline, so I am not sure if I am willing to bother at all or migrate back to crappy Ikea gat. 

Luckily I work with web apps, so I can read network traffic to a point, but if a casual user tries to use the system as advertised with ikea, no chance it would work due to a lot of issues.

Mimiix, 2020-06-26:
@dakipro I can see your issue. Let me explain a bit:

Phoscon is the page you open in the browser. For an example: This is my phoscon:
[image]

deCONZ is the actual software doing stuff. In my case, its reached by VNC as i use the HA addon.  This is also the place everything gets it data from. Phoscon uses the same API JowiHue/homeseer uses. deCONZ is the server of this API. 

It looks like this:
[image]

There is even a old web page, but that isn't in use anymore much. But it could be useful for your groups:
[image]

If your issue is in Phoscon, let me know so i can move this issue towards their repository.

Mimiix, 2020-06-26:
In addition: The reason you see stuff in Homeseer and deCONZ is because Phoscon needs to add support for it seperately. The RESTapi that is used by homeseer is created by deCONZ and is used by Phoscon aswell.

Either way, Phoscon is just a utility like Homeseer. But Homeseer uses the api better. 

For more background info, [this is a good read](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies) .

dakipro, 2020-06-26:
Well, I do not know where the issue lies exactly, I guess it is Phoscon, but it could be that the fix is in deCONZ giving some wrong data. Or at the Ikea giving some false reading..

Mimiix, 2020-06-26:
On my screenshot you see the remote group in Phoscon. Is that what you are reffering to?

dakipro, 2020-06-26:
I think Phoscon is not showing groups properly, I tried to explain it in my [previous comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2388#issuecomment-581015942) 
json from the deCONZ (backend right?) contains few more groups that are either not selectable from Phoscon, or not even visible in Phoscon. Some of them do appear and can be manipulated from Homeseer though (but not all of the issues can be fixed from here either, in few cases I had to use deCONZ directly to bind devices to groups, sometimes I had to repeat binding process over and over in circles for groups, as sometimes they would just get unbinded, I lost track as to when exactly but it happened).
So everything points to Phoscon struggling to show all remote groups as in my attached jsons above

Mimiix, 2020-06-26:
I can explain whats going on here. I had a talk about this before and know whats going on:

There is 2 types of groups: 

- The remote group (on that screenshot in the comment you mentioned ), with the icon of the Ikea remote. I call it group type A
- a "General" group which can be edited. I  call it group type B

When i got the conbee in december, i was able to create group type A and B. However, i noticed that if i created A i was able to use the remote,  philips dimmer switch, as how it worked within the Hue system. This means that the buttons had the exact same behaviour.

If i created group B, i could assign keys and such, but it wouldn't work as it works in Hue. So the difference here is that group A works as it does on it's original ecosystem. Also: It doesn't "Group" them as Hue does. That being said, it is impossible for me now to get rid of the type A group in phoscon. 

This functionality was removed in Phoscon around Jan this year. Because it "caused confusion". Not my call, as i really don't like functionality removed. 

However, this is where the OLD environment kicks in(the black page in my comment). In there, you can assign devices to a type A group.  Just assign devices to the remote and it should work.

dakipro, 2020-06-26:
I might try it next time I encounter the problem. Do you know if it can be reached from docker/vnc? 

If it is removed intentionally, then it is not completely/properly removed, if you look at the second picture in my first post I can see the remote group, Phoscon just shows some error in console when I click on it (or it just ignores the click and selects the first group in the list). 
So it is either not working, or it is not properly removed from Phoscon. 

I cannot 100% confirm it now as I forgot a bit, but if I would to assign one light to this remote from my first photo via Homeseer or manually via rest-api, then the Phoscon would start working just fine. I can edit the group, I can assign new lights to it and all is great. It is just when the remote(or its group) is introduced to the system for the first time, Phoscon starts to act up. 
Usually Homeseer would be able to show and assign lights to remote(group) fine, and once I've added one light, then the Phoscon app would work fine. 

It looks like Phoscon is struggling with selecting this remote/group if there are no lights assigned to it. Either it should be fixed, or removed a bit better. Right now it looks like a bug

jan666, 2020-06-26:
I think https://github.com/dresden-elektronik/phoscon-app-beta/issues/132 is the same issue

Mimiix, 2020-06-26:
You can reach the old web page, by going into Phoscon and go to the "Help" page. There is a link to the old webUI.

the username is delight, the password is the same as your phoscon login.

I will forward this to the Phoscon repository and make sure @YKO-de looks into it.