dresden-elektronik/deconz-rest-plugin #2390: Deconz ignore previous API entriy after a restart.

Smanar, 2020-01-29:
Hello, I have something strange again. (Long version here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-578411698)

An user is making some code for a new device support.
When he make the inclusion, all is fine, but after a restart, the device stop working, and on deconz he have the NWK adresse instead of name.

After association, all is good :
[image]
After deconz restart
[image]

There is some log to https://github.com/dresden-elektronik/deconz-rest-plugin/files/4112125/err.log

All informations seem to be saved and loaded on log.
This device make a device announce every time we use it, it's something normal, there is another that do same, but works without problem.

Deconz use the uniqueID to reconize device ? so how this can happen ? (and at a moment deconz see the NKW change and update the sql database)
I can't say if it's from deconz, the new code or the device.

I take all possible ideas/suggestions ^^.

Kane610, 2020-01-29:
Can you enable debug for deconz integration and share the logs describing the device both from when it works and when it doesn't

Smanar, 2020-02-01:
No-one have already something like this ?

It's not like all device are lost. Because in this situation devices are out of network, but name in deconz appear.

And there is no message in log about an ignored entry, or something like that.

Smanar, 2020-02-08:
@manup , we are realy blocked.
I have compared the zll.db file before and after the restart, files are same.

There is something in deconz code that can make it ignore an entry present in database ? Or do I need to search in API code ?

manup, 2020-02-08:
Some devices needs to be configured in the `sqliteLoadAllSensorsCallback()` in database.cpp file, similar to when they are configured while joining them. I guess this part might be missing?

Smanar, 2020-02-08:
Thx, we will check that.

So for you the problem is from a missing code in API, even it works before the restart, some data would be used but not memorized ,

Smanar, 2020-02-09:
Ok, so was exactly that.
It miss the cluster WINDOW_COVERING_CLUSTER_ID.

Thx again.

closed 2020-02-09 by Smanar