dresden-elektronik/deconz-rest-plugin #2392: New device support request: Bitron remote control 902010/23

nikolauzi11, 2020-01-29:
Hi, I would like to ask to support the Bitron remote control 902010/23.
[image]
[image]
[image]

Thanks!

SwoopX, 2020-01-29:
Hm, it feels like something is missing there. Can you try to re-pair while pushing the buttons randomly?

nikolauzi11, 2020-01-30:
Something seems to wrong with my device. I tried to re-pair, but now the endpoint menu button is gone. I think, I'll send it back and get back to you as soon as I have a new one.

ebaauw, 2020-01-30:
deCONZ hasn't read the _Simple Descriptor_.  Use the other menu button to read it.  Make sure to wake the remote during read.

nikolauzi11, 2020-01-30:
It reads reads the node descriptor and the power descriptor, but I don't get the second menu button.
Sorry, I'm just getting started on deconz.

SwoopX, 2020-01-30:
You might wanna give that procedure a try: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2052#issuecomment-552202908

In your case (for bullet 1), reset the remote accordingly.

nikolauzi11, 2020-02-01:
I got it paired up again after I reset it. But the same attributes are always read as in my first post.
Unfortunately SwoopX's method did not work either. It did not bind after inserting battery, only after pressing button 2 for 10s as usual.

SwoopX, 2020-02-04:
Can you try pushing a button and immediately reading the simple descriptors in deconz GUI? It's under the 1st menu button you previously mentioned.

nikolauzi11, 2020-02-06:
Sorry for the late answer, I have reinstalled my environment in an lxc container with gui, so it is no longer necessary to plug the conbee stick back and forth
I tried that, but the attributes look the same as before.
How exactly can I see that the simple descriptors were read successfully?

SwoopX, 2020-02-21:
I guess sniffing the zigbee traffic would give the required answers...

nikolauzi11, 2020-02-22:
I'm afraid I'm not up to it.
I'd like to donate the remote to one of the developers.

ebaauw, 2020-02-22:
> I'd like to donate the remote to one of the developers.

I'll be happy to take a look at the remote, but it seems simple enough to support without sniffing.

> I tried that, but the attributes look the same as before.

This looks to be a standard wireless controller, that sends _On/Off_ and _Level Control_ commands directly to (a group of) light(s).  It uses a ZHA endpoint, so it probably won't select a group on factory reset.  You want to bind the client _On/Off_ and _Level Control_ clusters to a group.  Then add a light to that group, and the remote should control the light.  You might also want to bind the server _Power Configuration_ cluster to the coordinator, and setup attribute reporting for _Battery Percentage Remaining_.  Double check that it's reported in 0.5%.

If you whitelist the remote, the deCONZ REST API plugin should log the commands the remote sends to the group on the various button actions.  This should be enough to create a buttonMap.  Setting up the group and client cluster bindings has been done for many other switches and remotes (e.g. the IKEA Trådfri On/Off switch).

nikolauzi11, 2020-02-22:
> I'll be happy to take a look at the remote, but it seems simple enough to support without sniffing.

I would be happy to support the project, so I would like to send you the remote control.
How can I get the address, because there are no PMs on Github

ebaauw, 2020-02-22:
Please send me an email, or DM me on the homebridge Slack, https://homebridge-slackin.glitch.me/.

ebaauw, 2020-02-26:
Received the device, thanks @nikolauzi11.

From the figure on the Quick Start Guide, you need to hold button 2 to reset the device.  It took a good 10 seconds of holding, but after that it immediately paired with deCONZ.  I bound the _On/Off_ and _Level Control_ clusters to a group and indeed, can control the lights in the group.

The switch sends the following on pressing a button.

button | command | direction | size | transitiontime
------ | ---------- | ------ | -- | --
1 | _Step (with On/Off)_ | _Up_ | 10 | 0xFFFF
2 | _On_ | | | 
3 | _Off_ | | |
4 | _Step (with On/Off)_ | _Down_ | 10 | 0xFFFF

No double press, release, hold, or anything fancy.  Will be supported by `buttonevent` values 1002, 2002, 3002, and 4002.  

It doesn't report battery status nor voltage, despite the _Power Configuration_ cluster.  It does seem to report the battery level through the _Power descriptor_, though.  deCONZ shows the battery at 2/3rd, but Wireshark doesn't seem to like the value.  I would need to check the ZigBee documentation for the exact meaning of the field.  Also, the CR2032 battery is disposable, not rechargeable.
```
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0xc301
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xf893
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Power Descriptor Response, Nwk Addr: 0xf893, Status: Success
    Sequence Number: 239
    Status: Success (0)
    Nwk Addr of Interest: 0xf893
    Power Descriptor: 0x8221
        .... .... .... 0001 = Mode: Receiver Periodically On (1)
        .... .... ...0 .... = Available AC Power: False
        .... .... ..1. .... = Available Rechargeable Battery: True
        .... .... .0.. .... = Available Disposeable Battery: False
        .... ...0 .... .... = Using AC Power: False
        .... ..1. .... .... = Using Rechargeable Battery: True
        .... .0.. .... .... = Using Disposeable Battery: False
        1000 .... .... .... = Level: Unknown (8)
```

ebaauw, 2020-02-26:
Ah, it does support attribute reporting on the battery _Alarm Mask_.

nikolauzi11, 2020-02-27:
I'm glad to hear that:grin:

ebaauw, 2020-02-27:
Above commit adds support:
```
{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "9f54229fce3bf59b06d34875d4b97392",
  "manufacturername": "Bitron Home",
  "mode": 1,
  "modelid": "902010/23",
  "name": "Bitron Remote",
  "state": {
    "buttonevent": 4002,
    "lastupdated": "2020-02-27T21:41:23",
    "lowbattery": false
  },
  "swversion": "20150518",
  "type": "ZHASwitch",
  "uniqueid": "00:12:4b:00:09:55:b0:6f-01-0006"
}
```
Note the missing `config.group`.  This is because of the way the Bitron sends messages to groups, see #2503.  If you manually bind the client (grey) _On/Off_ and _Level Control_ clusters to a group, you can control the lights in that group directly, even when deCONZ is down.

nikolauzi11, 2020-02-28:
Thank you very much.
I will test it once it is committed and I have a new remote.

nikolauzi11, 2020-02-29:
Remote works.
Many thanks again.

Hypfer, 2020-04-24:
How did you pair it? I forgot the procedure and also apparently didn't document it anywhere

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix