dresden-elektronik/deconz-rest-plugin #2397: Delete Spirit

FlyingPersian, 2020-01-31:
Hello,

I'm running deConz in Hassio on my RPI. After updating the deConz addon, one of my Eurotronic Spirit TRVs didn't reconnect, so I deleted it via VNC (select node --> delete selected node). Unfortunately, even though it disappears from the VNC, it still isn't deleted. It doesn't disappear in Hassio (under Integrations --> deConz) and also re-appears in VNC after rebooting deConz. Not sure what to do here, but I think this prevents me from re-connecting the Spirit after resetting it. I'm using Raspbee with my RPI.

Hassio: 0.104.3
deConz: 5.1
Phoscon: 2.05.72 / 12/12/2019
Firmware: 26050500 (cannot be updated, fails silently and cannot be updated the alternaive way(s))

Smanar, 2020-01-31:
See here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2074
If you are lucky, you will be able to delte it in some days.

FlyingPersian, 2020-01-31:
Well that sucks! I think that's the reason I cannot re-add the device. Any idea what I can do now?

Smanar, 2020-02-04:
No, it's a problem, you can re-include it without problem event with the "ghost bug".

You too can include a device already in the network, the new setting will overwrite the old.

Take a look on name, if the name is black, your device is on the network, if grayed, it's propably the "ghost bug"

FlyingPersian, 2020-02-06:
As of now this is what it looks like. I re-installed deCONZ on my HA install and these four remained, not sure why. I re-added a TRV (Eurotronic Spirit), but it's neither showing up in deCONZ nor in HA.

[image]

Smanar, 2020-02-06:
Your devices have jennicMacPrefix, I think it's xiaomi device.

The Eurotronic spirit is a little bit complicated to install if you are unlucky (need 2 phases), the offcial issue is here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1098

FlyingPersian, 2020-02-06:
Thanks! I just reset HA again, it's just faster than trying to figure out the issue.

closed 2020-02-06 by FlyingPersian