dresden-elektronik/deconz-rest-plugin #2398: IKEA buld connected, but I cannot turn it on/off

hver, 2020-01-31:
Hi, I have Conbee II stick running in RPi4 as Home Assistant addon. I added Hue bulbs to Deconz and they work flawlessly. 
But I have a problem with my new Ikea bulb, I can add it to Deconz, but cannot control it. Deconz identifies the bulb as **"TRADFRI bulb E14 W op/ch 400lm", Version 1.2.214**. Maybe it worth to mention, that I can only add this bulb after 6x on/off cycle. 
I logged in to the Deconz app via VNC, and tried some cluster commands from UI, I can run all "read" commands, but if I try to issue some "write" commands, nothing happens - the bulb stays 100% bright. 
Do you have some ideas how can I get the bulb to respond to commands? Maybe it is just broken? 

[image]
[image]
[image]
[image]

MunkeyBalls, 2020-02-01:
I'm might be running into similar problems with deconz II (2.05.72) on rpi3 with the marthoc/deconz docker image. 

I have a TRÅDFRI bulb E27 CWS opal 600lm running version 1.3.009 and for some reason I can turn the light on, but cannot turn it off anymore afterwards. Weird thing is, after I change the color I can turn it off perfectly fine. I can reproduce this every single time.
The two other tradfri bulbs I'm using are working fine, but they different bulbs, namely TRÅDFRI bulb E27 WS opal 980lm lights running 2.3.007

I only just received my deconz, so I'm still getting used to how the apps work. As far as I can see my lights are all up to date (updated via deconz). I didn't figure out yet how to get to the cluster settings. Whenever I go to the cluster info via VNC it asks me to select a cluster, but not sure where I need to do this.

stormmurdoc, 2020-02-05:
Same probleme here with a OSRAM Classic A60 TW. Other bulbs are working finde.
V1.03.07

closed 2020-06-12 by stale[bot]