dresden-elektronik/deconz-rest-plugin #2399: EUROtronic EUR_SPIRITZ Wall Radiator valve control

Max0212, 2020-02-01:
Hello All,

Currently I am using 4 **zwave** EUROtronic's on my heaters but the battery life is very poor. I exchanged one of the with the Zigbee version and the battery life is much better. Less than  month for zwave against over 3 months with the zigbee version. Now I want to exchange the other three devices for zigbee as well.

The issue is that in the zwave version I am using the valve control parameter. I use an external temperature sensor to control the valve. To do this I made a simple PD controller script to set the valve position. It all works fine and it keeps the temp with 0.4 C.

The zigbee version does not allow the control of the valve yet, before I exchange all of them I wonder if it is possible to give deconz control of the valve position (and operating mode).

Thanks

SwoopX, 2020-02-04:
I feel like nobody can give an answer since the device is unknown and so are its capabilities. Is the external sensor directly connected to the device?

Maybe you should issue a device support request as mentioned in the wiki.

ezerez, 2020-02-11:
Support for this device has already been added. see topic #1098  I would also be very interested to control the valve directly. now it is only possible to set a temperature and the valve position is reported. According to the [manual](https://eurotronic.org/wp-content/uploads/2019/11/Spirit_ZigBee_BAL_web_EN_November-2019.pdf) there are options for remote sensing and set valve position that are Read Writeable. I dont know enough to program this myself but I hope it is possible and that somebody wants to take a look at it.

SwoopX, 2020-02-11:
Ah, got it. Focussed to much on the title. Yes, the device is basically supported and coincidentially, I got one lying around. Now, from deconz GUI, you should be able to set the valve position as desired. Also, the remote sensing value can be set in that way.

However, I know that there's still some work to do to have some functionality available and be able to set it "the easy way". That means some modifications on API level and is probably what you require to control the device as you'd like to. Also, I don't fully understand what effect any adjustment of the remote sensing value may have.

I recommend to read the full threat you mentioned (if you not already have) to get a better understanding what I mean.

ezerez, 2020-02-12:
Thx for the reply @SwoopX , I read through the topic some time ago but i saw that @ebaauw replied  a week ago some small issues need to be fixed before they can implement this. I am personally using it in Home Assistant. Seems there is some talk about stability issues but mine is running fine for 3 weeks now. Hope someone finds the time to fix the issues and implement it. If i can help someone with testing i will be glad to do it :) I am guessing the remote sensing value you just send a temperature value from an outside sensor and then it adjusts the valve position, comparing the outside temperature and the temperature its set at. Dont know for sure though its just a guess

Max0212, 2020-02-12:
Hi all,

Thanks for the replies, it seems more people would like to have control over the valve position on the zigbee version. Hopefuly this can be implemented. 
I use the Eurotronic zigbee in Domoticz, it has been stable for some months now.

Indeed I use an external sensor to adjust the valve position via a small PD controller script. (on the z-wave versions I have running).

SwoopX, 2020-02-13:
> I am guessing the remote sensing value you just send a temperature value from an outside sensor and then it adjusts the valve position, comparing the outside temperature and the temperature its set at. Dont know for sure though its just a guess

It, unfortunately, doesn't seem so and that's what I don't fully get. This is from official ZCL documentation:

[image]

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix