dresden-elektronik/deconz-rest-plugin #2400: [Device Support Request] Please ad support for Wattle zigbee smoke detector

htreo, 2020-02-01:
[image]

https://wattle.com/support-main/wattle-devices/smoke-detectors/smoke-alarm/

SwoopX, 2020-02-01:
Hi there,

this is not a Develco device. Based on your screenshot, I'd put my bet on a Heiman device (which is already supported). Please provide the screenshots as described in the wiki for requesting device support to be certains. Also make sure the attributes have been read before taking these.

htreo, 2020-02-03:
It looks like Heiman smokedector. But wattle is the brand you can buy in Scandinavia. But can't be seen in phoscon app /  but shows in the home assistant.

SwoopX, 2020-02-03:
> But can't be seen in phoscon app / but shows in the home assistant.

That's because Phoscon is closed source. As it is available in HA, I conclude there's nothing to do for us.

closed 2020-02-04 by htreo