dresden-elektronik/deconz-rest-plugin #2403: Heiman gas sensor HS1CG-M support

stickpin, 2020-02-02:
Hi,

I've received a new Heiman gas sensor HS1CG-M. I was able to connect it to deConz, but it seems that whitelisting is needed as some of the HS1CG series already supported.

The sensor is working well and detecting a regular natural "cooking" gas correctly.

Here are the details:
[image]

[image]

[image]

[image]

Please let me know if more details are needed.

Thanks a lot in advance!

SwoopX, 2020-02-02:
I wonder what your request is.

You say, it has been paired successfully and is correctly detecting gas. This leads to the conclusion that it is fully functional. What else should be done?

stickpin, 2020-02-02:
@SwoopX is not available on the API level. 
/sensors are not reflecting it.

SwoopX, 2020-02-02:
Interesting, because the sensor IS already supported.

andriej, 2020-02-02:
It is not, same on mine - can be seen on VNC and works as a repeater, can't use it any further on (Phoscon, HomeAssistant).

stickpin, 2020-02-02:
@SwoopX anything I can do on my side that will help to understand why deCONZ not exposing it through API?

andriej, 2020-02-02:
Same as in #1973

SwoopX, 2020-02-02:
@andriej 

> It is not, same on mine

It is. I don't argue on that.

@stickpin how have you paired the sensor, via Phoscon?

stickpin, 2020-02-02:
@SwoopX yes, I’ve used Phoscon.

SwoopX, 2020-02-02:
How exactly have you done it, via sensor search? if yes, did you get a green notification of success?

From my experience, pairing does not necessarily for out, even for supported devices, on the very first time. Trying to reset and re-pair the device mostly leads to success.

andriej, 2020-02-02:
I've tried to pair it through Phoscon like dozen times (seeing other Heisman devices supported) and none of it succeded.
I can only see it in VNC

stickpin, 2020-02-02:
I see that my manufacturer code detected as: 0xbbaa which i guess is leading to the issue. I will try to re-pair once again when I will get home. Will let you know later today if it’s succeeded.

SwoopX, 2020-02-02:
What makes you think it is related to the manufacturer code being 0xbbaa?

stickpin, 2020-02-02:
@SwoopX if my understanding of the code is  correct. In case for deCONZ to detect the device in a proper way, deCONZ should know from which manufacturer it’s coming from, in my case it should be 0x120b, this is a Heiman manufacturer code. After that it goes to device type. 0xbbaa it seems like Osram Plug manufacturer or something like this. So as far a i see, there is two ways to solve it: 1. Gas sensor should be identified with proper manufacturer code. 2. To make deCONZ to recognize 0xbbaa with type Gas Sensor.

Please correct me if i’m missing something...

SwoopX, 2020-02-02:
I'm afraid your understanding is based on wrong assumptions. I'll explain that later.

Quick shot: It might be related to the sensor being a full functional device. Seen that previously with another device...

stickpin, 2020-02-02:
@SwoopX You were right, re-pairing solved the issue.
Now devices detected as a smoke sensor and triggered correctly through Homebrdige-Hue.

Thanks a lot for your help!

closed 2020-02-02 by stickpin

andriej, 2020-02-02:
@stickpin what was the way you did the pairing? step by step as I've tried different approach and still end up with undetected sensor...

stickpin, 2020-02-02:
@andriej I did it few steps:

**Unpair:**
1. Un-plugged it from the power socket completely.
2. Plug it back and wait for approximately 3 minutes so the green light stay still.
3. Use a small hole on a side to unpair the device. Press and hold the button inside the hole for approximately 5 seconds.
4. Connect to deCONZ GUI via VNC and delete your Gas sensor.
If device appears again, delete it again. 

**Pair:**
1. Press and hold the button inside the hole for approximately 2 seconds. The led on a device will blick for 2-3 seconds.
2. Go to Phoscon and do a new sensor search.
The next steps were a bit strange... 
The device should re-appear in deCONZ GUI. Go to the device and read basic information from it.
Right after that Phoscon will say that the device is ready.

Check if you can see it in sensors.
For me it's represented in the following way:

```
    "38": {
        "config": {
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "183093b394920a24d0f89e88a07b67e9",
        "manufacturername": "Heiman",
        "modelid": "GAS_V15",
        "name": "GAS_V15",
        "state": {
            "fire": false,
            "lastupdated": "none",
            "lowbattery": false,
            "tampered": false
        },
        "swversion": "20150330",
        "type": "ZHAFire",
        "uniqueid": "00:40:22:e9:b3:24:45:15-01-0500"
    }
```

I hope it helps...

andriej, 2020-02-02:
@stickpin - the "read basic information" in VNC was the key. 
It did pair fully, even tho not visible in Phoscon.

Thanks!

(It should be in manual somewhere I guess, or at least maybe someone will google it)

SwoopX, 2020-02-02:
@stickpin Great work, also regarding your write-up. The short explanation I owe you from the previous post in short: Although Heiman seems to be an OWM doesn't necessarily mean they use their own hardware for devices. We've seen that quite a number of times. The device you have has an Osram manufacturer code and a jennix (not heiman) MAC address. Code-wise, this has been taken into account.

@andriej Glad it worked out for you as well. Unfortunately, you will no be able to see it in Phoscon unless DE implements it. Phoscon is different software and not open source.

michis0806, 2020-02-18:
Any chance to get this working on a headless installation?

SwoopX, 2020-02-19:
Check out the wiki on how to access headless setups

ruabo, 2020-11-10:
Sorry, total noob here, I'm thinking about buying that sensor; is it possibile to remotely mute the siren if on ? Or can I turn the siren on ? May I have a code snipplet ? Thanks

Mimiix, 2020-11-16:
@Gordan74 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

Gordan74, 2020-11-16:
Hello,
I am able to get this sensor in Deconz VNC see attachment #1.
I even gave name "GAS" but it was not visible as entity in Home assistant.
Dou you have and idea how to add it as a device/entity?
Kind regards,
Gordan

Mimiix, 2020-11-16:
I don't see any attachments.

Gordan74, 2020-11-17:
I guess that is because we are messaging through github notifications. Can
you send me your mail and I wil resend my screenshot with data...
Kind regards,
Gordan

Mimiix, 2020-11-17:
@Gordan74 You can post directly here: 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2403

Gordan74, 2020-11-17:
OK, Here is current situation.
My Conbee II is "Configuration tool 1" and there is "unnamed" device which is Gas detector appearing in VNC.
[image]

And here are details of node info for this device when I click on it.
[image]

How do I get to use it within HA? I cant find it under devices or entities.
I one of posts I saw someone finding it under binary sesors with name set to IEEE code.
I don't see this either.
Thanks for help,
Goedan

Mimiix, 2020-11-17:
The name suggests it's not supported yet . Can you click the right DOT and click "basic" Then click the "read"button?
 For more info https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Gordan74, 2020-11-17:
Yes, here it is...

[image]

Cheers,
Gordan

Gordan74, 2020-11-17:
wait wait..

Gordan74, 2020-11-17:
Here it is...

[image]

Mimiix, 2020-11-17:
@SwoopX Why isn't the above working?

Gordan74, 2020-11-17:
Is there any log file that I can get for you to troubleshoot?

Mimiix, 2020-11-17:
Nope, because as far as i can see it should work. So i have to wait for Swoop here

Gordan74, 2020-11-17:
Coll, Thanks guys!

SwoopX, 2020-11-17:
The device is supported but apparently, the sensor(s) has not been created. Try entering sensor search in Phoscon again and read the basic cluter.

Gordan74, 2020-11-17:
Hi,
I have tried it many times. I followed suggested route (unpair  and pair). When paired  delete within VNC and pair again. But to no avail.
When I click on add sensor (choose Other) it is searching and 3 minutes pass and i do not get message sensor
[image]
 ready (green). It just searches until time is over. then I can see it in VNC as shown above.

SwoopX, 2020-11-17:
Have you read the basic cluster attributes during search? Another option is to quickly "tickle" the device by briefly pressing the button used for device reset.

Gordan74, 2020-11-17:
sorry for stupid question, how can I read cluster during search. I should initiate Phoscon search and then on other browser tab look in VNC?

SwoopX, 2020-11-17:
Yes, like you already did before.

Gordan74, 2020-11-17:
Now it was discovered and got name Gas Sensor.

[image]

[image]

Gordan74, 2020-11-17:
what is next step?

Gordan74, 2020-11-17:
Found it!

[image]

Gordan74, 2020-11-17:
Thank you @SwoopX  and @Mimiix !
Faaboulous!