dresden-elektronik/deconz-rest-plugin #2404: DB table "device_descriptors" is spammed with duplicate entries

SwoopX, 2020-02-02:
All,

I'm currently struggling with the surprising fact that I cannot add any more devices to my network. Having around 100 sensor nodes and like 15 routers, that shouldn't be an issue. However, during my investigation, I surprisingly recognized that table "device_descriptors" shows roughly 2000 entries, although I only have 55 devices. More interestingly, more than 95% of the 2000 are duplicate entries of one particular Hue light.

That recordset seems to be re-written more or less randomly over time (got also some other valid entries here and there in between). Not sure if that is somehow connected with not being able to add anything new but it appears to be wrong anyhow.

Worth to mention that Phoscon says it's a LWG001, Version 1.29.0_r21169. It is not always on since it is in a lamp physically disconnected from any power by a rocker switch in the hallway (intentionally, let's not discuss the sense behind it). Feeld like maybe a "if exists" check is missing for the database.

Any thoughts or experiences in this regard? Much appreciated!

ebaauw, 2020-02-02:
See #967.  Probably created when rejoining the device under previous version.

SwoopX, 2020-02-02:
@ebaauw Thanks! Looks quite similar. Not sure if I got the "fix" already at that time. Anyway, was a reason to hack the database, clean up and get rid of the orphan devices.

Unfortunately, it didn't solve my challenge not being able to pair my other 2 devices.

closed 2020-02-19 by SwoopX