dresden-elektronik/deconz-rest-plugin #2405: Rasberry pi 3b+, Raspbee stop reponding

DKFrederik, 2020-02-02:
I have an issue, where my PI gets unresponsive after about a day (~24 hours]

The logs don't show much ( i've attached last log) 

[image]

I've tried installing varius images from phoscon.de and respbian and did a manual install of phoscon software as per instructed on the phoscon webpage. 

This happens evry day and the pi gets completly unresponsive so that I have to unplug the power cord and plug it back in! 

Best regards 
Frederik

mr-mbs, 2020-02-03:
Started using the raspbee a few weeks back on a rpi4. Installed HassOS and the deconz plugin. I notice the same behavior. At first it all works fine and sometimes it is unresponsive. This means all the devices are unavailable.

I need to restart / reboot to get it all working again of unplug the power cord.

Hope someone can help?

SwoopX, 2020-02-04:
@DKFrederik Even no SSH connection possible? If it is still possible, I'd be interested to know what the "wa" value is from the top command. My machine also dies every now and then with high wa, so pending I/O and I cannot find out if it's deconz or FHEM related.

Anyway, do you guys have the latest deconz version and firmware running?

ebaauw, 2020-02-04:
I’ve had issues with my Raspberry Pi 3B+ freezing (!) when it gets too hot.  Compiling the REST API plugin with `make -j3` would do that, but just running deCONZ (with GUI enabled) and two homebridge instances is fine.  The RaspBee partly covers the CPU, hindering the air circulation.  the official Pi case is terrible, offering no airflow at all.  The Pi 4 gets a lot hotter, but I haven’t yet tried a RaspBee in it.

I run a cronjob to log the CPU temperature and frequency (see `vcgencmd`) every five minutes to be able to check the Pi’s state before the freeze, after it has rebooted.  If the freeze does seem temperature related, you might try and place your Pi case on its side, attaching a heat sink to the CPU, or use a heatsink case with a GPIO riser.

DKFrederik, 2020-02-05:
> @DKFrederik Even no SSH connection possible? If it is still possible, I'd be interested to know what the "wa" value is from the top command. My machine also dies every now and then with high wa, so [...]

Unfortunately It does freeze completly, so I can not grap anything before I do a "power reboot". I did find that some times timestamps in syslog is all mixed up, So I did try a NTP troubleshoot, but with no luck so far.

DKFrederik, 2020-02-05:
> I’ve had issues with my Raspberry Pi 3B+ freezing (!) when it gets too hot. Compiling the REST API plugin with `make -j3` would do that, but just running deCONZ (with GUI enabled) and two homebridge [...]

@ebaauw 
That was my first though also, so i grab all temp. from two days (with a reboot in between) and logged them @ a external drive. At no time did the temp. rise above 62 degree. So I would think that it is within threshold.

My PI has heatsinks and a temp. controlled fan standing beside it, blowing air directly at the PI ( from the top down)

DKFrederik, 2020-02-05:
I would think it's weird that it's almost on the clock when a day has past (24 hours) then I need to do a power reboot.

closed 2020-06-11 by stale[bot]