dresden-elektronik/deconz-rest-plugin #2406: Cant detect iHorn Temperature and Humidity Sensor

Slothinger, 2020-02-02:
Hi,

I'm having trouble adding this [iHorn (Huawei) Temperature and Humidity Sensor](https://de.aliexpress.com/item/4000192768667.html?spm=a2g0s.9042311.0.0.397e4c4dzq1n8G) as a Sensor in the Phoscon App.
When i put the temp sensor into paring mode the deconz application detects this device:
[image]

The webinterface doesnt detect it and tells me "Failed to connect".
Is there something i can try or do i have to wait till this device gets eventually supported?

cheers

SwoopX, 2020-02-02:
Have you tried pairing more than once? It happens that those need a couple of attempts. Also, make sure you're close enough to a zigbee router device.

Slothinger, 2020-02-03:
yeah many times and with different devices running the software. conbee2 with my windows notebook, raspberry with raspian and with the image from the website.
zigbee router device = conbee2? (tried to switch to router mode)

SwoopX, 2020-02-04:
Ok, then let's check the details. Can you please provide the screenshots required as described in the wiki for requesting device support? Also, please make sure to read the attributes before taking them.

Slothinger, 2020-02-04:
[image]

Measured Value of the humidity looks correct in the attributes table.
but its still not possible to add it as a sensor in the phoscon webinterface (verndor = other).
thx for your help!

SwoopX, 2020-02-04:
Basic cluster is more important.

Slothinger, 2020-02-04:
[image]

ok should be this

SwoopX, 2020-02-04:
Sorry, but you forgot to read the attribute. Thanks.

Slothinger, 2020-02-04:
tried it but it doesn't work in the basic view.
is there another way to provide you this data?

SwoopX, 2020-02-04:
That's a bit strange. However, having a closer look at your initial screenshot, it seems that you run an older version of deconz. I'd recommend an update and giving it another try.

Slothinger, 2020-02-05:
[image]
[image]

SwoopX, 2020-02-05:
Just realized that I misread the vendor (thought it was Xiaomi, not Huawei). Those devices will not bee seen in Phoscon, as it is a different software and closed source on top of that.

Does the sensor work anyway with whatever middleware you might use? Otherwise, you must make a device support request out of it.

Slothinger, 2020-02-06:
I'll try if i can use it directly in home assistant but it would be nice if the device gets supported.
Should i recreate this issue as device support request?

SwoopX, 2020-02-06:
Check out HA first. Otherwise, ti would be sufficient to just rename the issue. I guess all relevant screenshots are already there.

Slothinger, 2020-02-17:
@SwoopX renamed the issue because i didn't found a way to get it into ha.
Would be great if there is a way to get it supported.
Thanks for your help so far!

SwoopX, 2020-02-25:
> So i deleted the device in deconz gui and after reconnecting and checking the correct temp and humidity values its still not possible to add the sensor in phoscon. (Sensors -> Add new sensor -> Other)

How did the pairing go while searching for new sensors, did it end with a green success message? If not, you may need to wake it up while pairing. See if there's any physical button and press it during the procedure. I hope that those sensors are not fiddled together like the Xiaomi ones because that would pretty much require traffic sniffing.

Even after a successful paring, it will not show up in Phoscon as it is a different software and we cannot do any integration there. Best way to verify if the sensors have been created successfully would be via REST API calls. Nothing automatically created in HA so far?

Btw, new version .74 is out since today so no need any more (for now) to manually compile the plugin.

rmotapar, 2020-02-25:
> How did the pairing go while searching for new sensors, did it end with a green success message? If not, you may need to wake it up while pairing. See if there's any physical button and press it [...]

@SwoopX Trying to see how to get the devices integrated into HA when they don't show up in the phoscon app. Is there a user guide or instructions somewhere that talks about this process? Also, I am using the deconz add-on in home assistant. How would I go about getting the latest version?

SwoopX, 2020-02-25:
Are you using docker?

rmotapar, 2020-02-25:
> Are you using docker?

I am using hassio which uses docker. Here's the link to the Deconz hassio add-on : https://github.com/home-assistant/hassio-addons/tree/master/deconz

Slothinger, 2020-02-25:
> How did the pairing go while searching for new sensors, did it end with a green success message? If not, you may need to wake it up while pairing. See if there's any physical button and press it [...]

I think it was the green dot was there, but i can confirm that tomorrow.
For the pairing i tried any combination (longpress, klicking, 1 click....) 
When i look at the cluster in deconz gui the temp and humidity updates the values faster when i press the button.

Last thing i tried today was the rest api thing:
Post { "devicetype": [Name] }
to get the key and then
GET --> http://[ip]/api/[API-Key]/sensors 
to list the sensors.
But there was only the default software day and night sensor and not my temp&humidity sensor.
I think it only lists devices from phoscon, right?

Wanted to try this Home Assistant Rest integration:
https://www.home-assistant.io/integrations/rest/

im using docker btw.

SwoopX, 2020-02-25:
Just to be clear, I meant sensor search in Phoscon, not deconz GUI. Successful pairing should look like this:

[image]

I don't have any experience with HA since I'm using a different middleware. However, since you haven't seen the sensors via the API call, it looks like they haven't been created. Xiaomi devices e.g. may be a little stubborn at times and require re-pairing multiple times. You may want to repeat pairing a couple of times to see if that hepls, if you haven't tried already.

Slothinger, 2020-02-25:
Ah ok, i tought they must be connected first in the deconz gui (with the green line). 
But yeah i tried the sensor search many times in phoscon with the "other" option as in your screenshot. Also tried the deconz option etc...

What do you mean with "it looks like they haven't been created"? When deconz gui sees the zigbee device and can read the values, it should be possible to read them via api right?

SwoopX, 2020-02-26:
> What do you mean with "it looks like they haven't been created"? When deconz gui sees the zigbee device and can read the values, it should be possible to read them via api right?

Unfortunately not. with the modifications made, there should have been at least 2 sensors created (temperature and humidity) accessible via API. As this is not the case, we need to find out why. Could have various reasons... But debugging, it's good that you can compile and test changes I provide to my repo.

Slothinger, 2020-02-26:
hm, the first test (Rest + pull battery) didnt work.

Here ist the output:
[ihorn_join.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4256375/ihorn_join.txt)

SwoopX, 2020-02-26:
Would prefer to stay with english, so the rest of the people may follow.

Thanks for the log. Are you 100% sure you're using deconz version .74 on your ubuntu? The log shows the previous version

> 15:42:51:221 Load config apiversion: 2.05.73 from db.

Slothinger, 2020-02-26:
Ah sry didnt noticed that there is a update available. Now im on 2.05.74

And here is the new file:
[ihorn_join.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4258159/ihorn_join.txt)

SwoopX, 2020-02-26:
Thanks again. Just to be sure, have you reset the sensor every time before pairing? I'd assume not because I don't see some particular data I'm after. Also, have you restarted deconz after the update?

Slothinger, 2020-02-27:
Yep, i rebooted the whole system.
Here are two logfiles, one with a sensor delete and one with a reset of the sensor:
[ihorn_join_reset_sensor.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4260451/ihorn_join_reset_sensor.txt)
[ihorn_join_delete_sensor.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4260452/ihorn_join_delete_sensor.txt)

SwoopX, 2020-02-27:
This still does not look right.

How did you reset the sensor? Have you hold a button somewhere on the device for like 10 seconds and what does the manual say about it. Point is, essential information regarding the pairing process is not there, but it must be...

Slothinger, 2020-02-27:
The manual is in chinese but with google translate i found out that i have to press the button 5 sec. 
I did reset the sensor in deconz gui, klick on the sensor and select reset in the top left corner menu bar. Same with the delete function...

I can show you that over teamviewer, otherwise i think i should dump that sensor and get another one...

SwoopX, 2020-02-27:
> I did reset the sensor in deconz gui, klick on the sensor and select reset in the top left corner menu bar.

That could eventually be the reason. I never tried this to reset a device to be honest, but please give it another go physically on the device. Usually, it gives you some kind of feedback upon a successful reset (e.g. some lights blinking or so).

After physical reset, give the sensor search in Phoscon another go with the debug output enabled. If it remains stubborn for whatever reason, we can see if we can have a session on the weekend or you send it over.

Slothinger, 2020-03-01:
The manual says 5 sec to reset the sensor and after that it enters network distribution state.
a single press updates the temp, humidity and mby battery values.

Here is antoher log where i tried that; 
[ihorn_join4.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4272413/ihorn_join4.txt)
Would be great if we can arrange such a session. (I joined the deconz discord mby we will meet there)

SwoopX, 2020-03-21:
Finally made it:

```
"4": {
    "config": {
      "battery": null,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "a15664738b4a66137336aa341cc16712",
    "manufacturername": "iHorn",
    "modelid": "113D",
    "name": "113D",
    "state": {
      "lastupdated": "2020-03-21T21:37:44",
      "temperature": 2170
    },
    "type": "ZHATemperature",
    "uniqueid": "00:15:8d:00:04:c4:74:d2-01-0402"
  },
  "8": {
    "config": {
      "battery": null,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "2e83d507f061b408d3b0254c4d391890",
    "manufacturername": "iHorn",
    "modelid": "113D",
    "name": "113D",
    "state": {
      "humidity": 2950,
      "lastupdated": "2020-03-21T21:38:04"
    },
    "type": "ZHAHumidity",
    "uniqueid": "00:15:8d:00:04:c4:74:d2-01-0405"
  },
  "9": {
    "config": {
      "battery": null,
      "on": true,
      "pending": [],
      "reachable": true
    },
    "ep": 1,
    "etag": "b657f3e9cb591ccdc99a1c780b2b68d5",
    "manufacturername": "iHorn",
    "modelid": "113D",
    "name": "113D",
    "state": {
      "alarm": false,
      "lastupdated": "none",
      "lowbattery": false,
      "tampered": false
    },
    "type": "ZHAAlarm",
    "uniqueid": "00:15:8d:00:04:c4:74:d2-01-0500"
  }
}
```

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix