dresden-elektronik/deconz-rest-plugin #2408: Sonoff Zigbee device (SONOFF BASICZBR3) is not visible in Phoscon.

AbDhops, 2020-02-03:
Recently I got [SONOFF BASICZBR3 DIY Smart Switch](https://sonoff.tech/product/zigbee-diy-smart-switches/basiczbr3). 
Its visible in deconz over VNC. Could also control the relay over the cluster. However, I am not able to see it in Phoscon app. Even after multiple scanning. 
Could it be because this is not yet supported and newly launched ?

Phoscon Version : 2.05.71 / 11/17/2019
Firmware version : 26490700

zladukas, 2020-02-04:
You should find it in lamps section.

SwoopX, 2020-02-04:
@zladukas don't thinks so, as it is a switch. Many devices will not show up in Phoscon, as they are not supported (by Phoscon). Note that Phoscon is a different software than deconz rest plugin and closed source as well. So probably, not much what can be done about it.

It's more important if the device is visible via REST API. If not at all, a device support request is required. Feels more like it.

Smanar, 2020-02-04:
Wired switch are see as light and this device is supported https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1950 but yes unfortunately there is so much devices invisible on phoscon for people that don't use home automation application.

zladukas, 2020-02-04:
Trust me, I'm successfuly using them.

SwoopX, 2020-02-04:
Alright, convinced ;)

realwax, 2020-02-05:
@zladukas Are you using a Conbee 1 & 2 and which version of Deconz? 2.05.72? I am pretty sure I tested it with whether 72 or 71 without success on Conbee 1. What did the trick? Thanks!

zladukas, 2020-02-05:
@realwax raspbee 2.05.72.
Try click these toggles in GUI, when pairing. I'm not really sure, but seems it helps.

[image]

SwoopX, 2020-02-05:
It's probably "just" the simple descriptors checkbox. It can help making all the clusters visible and also resolve issues where sensors have not been created in the API...

jscurtu, 2020-02-06:
Same here, no issues at all. I remember though, the first one I used I did reset the switch holding the button down long.. but didn’t have any issues with a second one.

[image]

Smanar, 2020-02-07:
Try compare your JSON for this devices. I don't think Phoscon use other data than the JSON to make a device visible or not.

jan666, 2020-03-21:
Hi,

I have the same problem with my SONOFF BASICZBR3 DIY Smart Switch. 

Pairing via. Paired via Phoscon but does not show up as light in phoscon. No new light is ceated in the REST API. But it works in deCONZ (off, on, toggle)

Any way to fix this?

[image]

[image]

Smanar, 2020-03-21:
Are you sure you haven't new light in the API ?

Can you show us the basic attributes (Cluster 0000) ? But for that I m seeing, you have the same device than other.

Have you tried to set again Phoscon in permit join, and recheck the box like previous post ?

jan666, 2020-03-21:
> Have you tried to set again Phoscon in permit join, and recheck the box like previous post ?

Ahh, that helped. Did not understand it earler.

To all other slow on the uptake: if you dont see your "SONOFF BASICZBR3 DIY Smart Switch" in Phoscon but in deCONZ: Search for Lights and click the checkboxes shown in @zladukas s post.

SwoopX, 2020-03-21:
@jan666 Can you please provide a screenshot of the node info panel? I recall such a behaviour if a required value is missing...

jan666, 2020-03-21:
@SwoopX 

[image]

[image]

SwoopX, 2020-03-21:
First quick look: the manufacturer code in the node descriptor panel could be the reason. I'll have a closer look in the evening. Thanks!

serrnovik, 2020-04-17:
Hello. Did you find the issue? Got my new sonof basic zigbee switches. Trying to connect/pair them in phoscon app. Nothing happening. Long push to button makes light blink several times. No errors when light is not found. Running Phoscon in docker on synology, 2.05.75 / 3/8/2020, Conbee  II. This is my first lights on Conbee. All other devices work fine (door sensors/temperature/buttons etc).

Smanar, 2020-04-17:
From zigbee2mqtt
>if brand new, when powered on it will attempt to pair to Zigbee2mqtt automatically. If not (or if has been paired before and needs to be re-paired) - press and hold the (relay) button on the top for about 5 seconds until the relay clicks and the red LED flashes several times. The device will then go into pairing mode and the blue LED will begin to flash. When connected, the blue LED will turn on solid. It should then be connected to Zigbee2mqtt. Pressing the button should activate the relay on/off as normal, and the red LED will be on/off.

Have you check directly the API, to see if the device was inside ?

serrnovik, 2020-04-17:
Actually it is (had some issues with vnc port on docker, but fixed it) [image]

Had to unplug and plug back to check if it is really the one. It has name 0xDA2F name on the picture. Another offline is the second sonof that is unplugged.

Smanar, 2020-04-17:
Yes, it's a router.

So have you used the trick in this issue, that help the device to be displayed in phoson ?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2408#issuecomment-582487956
and
Set phoscon in permit join again, and asking for basic attribute ?

serrnovik, 2020-04-17:
Ok, thank for help. After your comment I actually got to some success. 

During first reading of the post I didn't understand correctly. To sum up and for others.
1. When you just add switch using find lights in phoscon they are not displayed (yet inside paired as switch only)
2. You have to log into vnc (attention to firewall rules and ports configs if you use docker)
3. Log in with vnc to your device. And do following
[image]
4. Search for lights in phoscon again. 

After this your device name should change to on/off device.

Aslo it is now visible in phoscon
[image]

However I still don't see it in Home Assistant. Will investigate that tomorrow.

serrnovik, 2020-04-17:
Actually another strange thing.  I could not do the same thing with another switch. It also looks different in the api. 
[image]

serrnovik, 2020-04-17:
Update: 

Home assistant discovered them after reboot. 
Also sonoff switches do build a mesh. But so far only with each other (even though they are in the middle between other powered devices. Maybe it will take time. 
[image]

[image]

morfei1, 2020-05-25:
> Try click these toggles in GUI, when pairing. I'm not really sure, but seems it helps.
> 
> [image]

I can confirm walkaround it's working on .76. Thanks for share!