dresden-elektronik/deconz-rest-plugin #2410: Re-adding Aqara/Xiaomi sensors

FlyingPersian, 2020-02-03:
Hello,
I had to reinstall my Home Assistant install on my RPI. I had all my sensors (2x Spirit TRV, 2x Xiaomi Mijia Motion Sensors, 1x Aqara Door Sensor) connected to my Raspbee. After the re-install, I'm failing to add the door sensor and one motion sensor. 

When I reset the door sensor by holding the button, the LED just blinks fast for as long as I hold the button, after releasing it blinks one more time, but it's not connecting to by Raspbee. It's also not showing up when sniffing for Zigbee traffic. 

Same goes for the motion sensor. It blinks twice while holding the button (no matter how long I hold it), then blinks another time after I let go. It also doesn't show up while sniffing. 

The devices that do show up while sniffing are both TRVs, something called PhillipsL, which is probably the virtual light or something within Phoscon, my motion sensor when activated and once in a blue moon my Xiaomi temperature sensor. So basically the 2nd motion sensor and door sensor are never read, even when activated.

FlyingPersian, 2020-02-04:
I just realized that when I press the reset buttons of the two devices, I sniff a Beacon Request:

[image]
[image]

SwoopX, 2020-02-04:
Those can be quite stubborn. Have you given it a try pairing right next to your raspbee?

FlyingPersian, 2020-02-04:
> Those can be quite stubborn. Have you given it a try pairing right next to your raspbee (and more than once)?

Yes I have! Tried 30x already while holding them 10cm apart. I just reconnected the working motion sensor again and this is what I got:

[image]

So I suppose the Beacon Request should be responded to with the Ack (I believe it stands for ackowledged) and then the Association Request should be made, followed by the Association Response from the coordinator. Not sure why the Beacon request isn't met..

SwoopX, 2020-02-04:
Well, for the TVR, it's an Eurotronic?

> After rebooting my TRV wouldn't reconnect and I couldn't reconnect it after reset as well as the device was still ghosting around (visible in VNC).

If yes, that's actually the reason why decided they're not an option for me. However, people around here report that it would be sufficient to power cycle them in case they're not responsive any more.

For the beacons, my gut feeling is that you're heading into the wrong direction (might be wrong thoughas I didn't learn the pairing my heart). When you initiate sensor search, raspbee sens a permit join. Interesting would now be what happens afterwards (transmission of transport key, node descriptor send, etc.) for the devices in question. If you use wireshark, you may want to apply this filter "zbee_nwk" and draw your attention on the zigbee ZDP traffic.

Also, since I haven't read anything about further routing devices (e.g. smart plugs) other than the raspbee, communication to and from the sensor must susually always involve 0x0000 as source or destination.

FlyingPersian, 2020-02-05:
> Well, for the TVR, it's an Eurotronic?
> If yes, that's actually the reason why decided they're not an option for me. However, people around here report that it would be sufficient to power cycle them in case they're not responsive any more.

It is the eurotronics. Sadly, rebooting didn't help in my case. 
 

> For the beacons, my gut feeling is that you're heading into the wrong direction (might be wrong thoughas I didn't learn the pairing my heart). When you initiate sensor search, raspbee sens a permit [...]

I will try later when I'm home. Weirdly, when I re-added the one sensor, I didn't get anything regarding the transport key. There was no device announcement, only the stuff in the screenshot. Not sure why, maybe cuz I deleted it a minute before? 

> Also, since I haven't read anything about further routing devices (e.g. smart plugs) other than the raspbee, communication to and from the sensor must susually always involve 0x0000 as source or destination.

I only have sensors and the TRVs. 0x0000 is indeed the source.

SwoopX, 2020-02-05:
That's how it usually should look like, e.g.:

[image]

FlyingPersian, 2020-02-06:
Well tbh I never have the Device Announcement, but it's working nonetheless for all the other devices except those two. Even reinstalled HA to be sure. Here the log from every time it fails:

[image]

Smanar, 2020-02-06:
But the blinking are so strange.
Have you tried with new batterie ?

FlyingPersian, 2020-02-06:
> Have you tried with new batterie ?

Just tried the battery of the other motion sensor, isn't working. Same shit -.- I contacted the support, but I doubt anything useful is gonna come out of this.

FlyingPersian, 2020-02-06:
I FINALLY GOT IT!! A few months back I wasn't able to upgrade the RaspBee FIrmware, but now it worked. For whatever reason I was able to connect both devices after upgrading the firmware! Thanks for your help

closed 2020-02-06 by FlyingPersian