dresden-elektronik/deconz-rest-plugin #2413: Read only "Unknown" clusters

Kniesch, 2020-02-04:
On all my devices I just get "unknown" clusters. I tried resetting everything and reconnect the devices but it is still not working. I am able to set brightness or color to my light bulbs via deconz but since nothing is read I can't detect changes made by remote controls. I also tried older firmware versions as well as older deconz versions but its always the same.

[image]

I get the same for the Conbee 2 as well for all of my devices.
OS: Windows Server 2016
Phoscon: 2.05.71
Firmware: 256A0700

Thank you in advance

ebaauw, 2020-02-04:
Your `general.xml` is corrupted or the _ZCLDB_ setting (under _preferences_) is wrong.

Kniesch, 2020-02-04:
Thanks for pointing me in the right direction. In the ZCLDB settings there was nothing added. I added the general.xml file and after a restart everything was working fine again.

closed 2020-02-04 by Kniesch