dresden-elektronik/deconz-rest-plugin #2414: Sensors disconnect

dieterlessmann, 2020-02-04:
I have a handful of Xiaomi sensors around the house and they are constantly disconnecting or becoming unreachable so I have to delete them and re-add them.
I've changed the deconz stick to the new v2 but have the same problem.

Sometimes I get lucky and when the sensor is showing as unreachable, I close the contact and open it again and the contact sensor will come back online but most of time I'll reset all the sensors and after a day or two they are all offline again.
I've swapped Rasperry pi's, I've un-installed and re-installed the phoscon software. Currently running on 
version: 2.05.72  12/12/2019 
firmware: 264A0700

I've added an extension to the usb stick and that hasn't helped either, the sensors are within 5 meters of the Raspi and USB stick so it can't be the range

Smanar, 2020-02-04:
Have you take a look on your network on deconz, the color line between devices can be usefull for diagnostics.

SwoopX, 2020-02-04:
Could also be related to router devices, as those react quite sensitive on them being not available. They stick to the parent they were associate with, no matter what. Maybe some smart plug being disconnected from power?

Smanar, 2020-02-04:
I m agree with him, if you can show your network image from deconz, can be usefull.

dieterlessmann, 2020-02-06:
I'm trying to view the network but not able to get it up... 
I'm running the deconz through HASSIO and trying to view the networkmap but having a little trouble with that, currently I just have a blank screen

sunshineautomate, 2020-04-24:
> 
> 
> Could also be related to router devices, as those react quite sensitive on them being not available. They stick to the parent they were associate with, no matter what. Maybe some smart plug being disconnected from power?

This seems to be the answer I'm looking for to an issue I've been having. I have a Sonoff Zigbee that won't respond when I shutdown a light that it was associated with and it would have been the closest device when initially pairing. Is there a way to break that bond so the device is not reliant on this device running?

ebaauw, 2020-04-24:
> Is there a way to break that bond so the device is not reliant on this device running?

Devices conforming to the standard should find a new parent router.  Not all devices do, however, notable Xiaomi don't.  I have no experience with Sonoff.  For those that don't: factory reset the device, and re-pair it on the physical location where you'll be using it.

Morale: don't power down your Zigbee routers, in particular, don't use 20th century wall switches to control your smart lights.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

caseyjhol, 2020-08-04:
I'm running into a similar issue. I'm using the Sonoff SNZB-02 (TH01) temperature/humidity sensors. I've got one in my fridge and one in my freezer. They'll work fine for a day or so, and then eventually they disconnect. I have to take the sensors, add them again using Phoscon, and then put them back. They work again for a bit, and then the issue happens again. I've thought about getting another Zigbee device to act as a router and placing it closer to the refrigerator, but when the sensors are working, the signal seems to be fine (the lines connecting the nodes are more of a dark green, rather than bright green - will update with a picture when I get a chance).

dieterlessmann, 2020-08-04:
I've pretty much given up on zigbee. It's just too unreliable
The devices are still scattered around the house but I don't bother reconnecting them anymore. There's no point because it's just a matter of time before it stops working again. Fine if you only have one but all of mine go down together so it can't be a signal thing as two xiaomi contact sensors are within 2 meters of the dongle and they go down too. 
Zigbee is crap. throw it in the bin is my recomendation.

Smanar, 2020-08-04:
Zigbee is not crap, Xiaomi is crap.
And it depend of your mesh.

dieterlessmann, 2020-08-04:
You're probably right, all my devices are Xiaomi. I'll try a different brand

Mimiix, 2020-08-05:
@dieterlessmann I have to agree with Smanar, partly. 

Xiaomi devices are not 100% zigbee compliant. They work fine if they have "better brand" devices in the mesh, but if you only have xiaomi devices it is sticky business. 

If you have a proper mesh, it is one of the best wireless systems around IMHO.

dieterlessmann, 2020-08-05:
In that case I may have to give ZigBee another go with another brand. They are amazing when they are working and I miss having them so will look for some contact sensors online to replace my shitty Xiaomi. All my window, washing machines and mailbox notification automations rely on the open/close information so I've lost a lot of functionality not having them in my house. Thanks for the feedback 👍

Mimiix, 2020-08-05:
For any advice, you are welcome on our Discord! 👍

caseyjhol, 2020-08-05:
My understanding is the new Sonoff devices should be more zigbee compliant than their Xiaomi counterparts. TH01 (4) and TH01 are the two that keep disconnecting.

[image]

Mimiix, 2020-08-05:
@caseyjhol Having solely Sensors isn't a good mesh and probably end up in what you describe.

Typically you want a healthy mesh with routers and endnodes. 

Add some routers and see what happens. If you still have issues or need help, open a [User Question issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=User+Question&template=user-question.md&title=).

caseyjhol, 2020-10-25:
Just to follow-up for anybody who comes across this issue in the future, I added a couple Sonoff S31 Lite ZigBee plugs to my setup, and it resolved the issues I was having.

Oderik, 2021-09-24:
@caseyjhol thanks for reporting your solution!

So, do I understand this correctly? Cheap Xiaomi and Sonoff sensors only remember one "gateway node" and once that becomes unavailable, the sensor connection is permanently broken?

Placing a gateway node (sorry for possibly wrong terminology) nearby seems like a workaround that might help. But what if that's a light bulb that is accidently switched off or there's a power outage? Is there any chance we can make this more robust? I mean, I do see the sensor in the Deconz network, so there should be a chance to recover from the disconnected state, right?

I am experiencing the disconnection issue regularly and it breaks the sensor's functionality of course.

Mimiix, 2021-09-24:
@Oderik Let me change your terminology (so we get on the same page) first before continuing:

Gateway node = Parent . This could be the Coordinator (Conbee/Raspbee | Blue in deCONZ) or a Router (Any 220V device | Yellow in deCONZ). 
End node > Sensors/Battery operated devices.Grey in deCONZ. 

Xiaomi's tend to not switch parents, while they should. They can handle outages, but not excessivly. We often see that if the parent is out for a length of 1 hour, they drop out of the network. 

To add here, Conbee can be the "parent" for give or take 20 devices. These can be end nodes (sensors) and router(mains devices). If you have a network of only sensors, it gets unstable after a while because they don't mesh.  It is recommended to have a good mix of routers and sensors. 

For xiaomi's, its best to pair them on the place you are going to use them. But you need to make sure that you have enough Routers in the network.

Sonoff sensors do switch parent in my experience. 

Your issue could be that theres not enough routers. 

I hope this gives some clarity. 

Either way, If you need additional help i'd like to ask you to either post on the forums(forum.phoscon.de) or discord: https://discord.gg/3XGEYY9

Oderik, 2021-09-24:
Just in case anybody else stumbles upon this: thanks to the help of @Mimiix on discord, it looks like the Sonoff disconnection issues are solved for me.

All I had to do was to perform a firmware update of my RaspBee II.

I am running hass.io and today I learned that firmware updates are not possible from within that system. See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually for details!

Thanks again, @Mimiix!