dresden-elektronik/deconz-rest-plugin #2415: New light added with same amount of details as a sensor

Kane610, 2020-02-04:
This is all we get with lights right now 
{"e":"added","id":"43","r":"lights","t":"event","uniqueid":"00:17:88:00:00:00:00:00-0b"}

Could we get it with more details like sensors?
{
    "e": "added",
    "id": "10",
    "r": "sensors",
    "sensor": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "7088b28f8a8a2c786e6e48d95c547fa4",
        "id": "10",
        "manufacturername": "icasa",
        "mode": 1,
        "modelid": "ICZB-KPD12",
        "name": "ICZB-KPD12 10",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:0d:6f:00:10:65:8a:6e-01-1000"
    },
    "t": "event",
    "uniqueid": "00:0d:6f:00:10:65:8a:6e-01-1000"
}

ebaauw, 2020-06-03:
Missed this one, but is also mentioned in #2625 and fixed in #2553.

Mimiix, 2020-06-12:
As @Kane610 mentioned in the HA issue , it should be fixed in 0.77. Can you confirm it is fixed?

Kane610, 2020-06-12:
Anyone running hass and pairs a new light can verify it

Mimiix, 2020-06-12:
```
2020-06-12 18:49:40 DEBUG (MainThread) [pydeconz.deconzdevice] Dimmable light 16 created as 
{'etag': 'b653f35ee1c1b2e17e8424ad9a0cf6a4',
 'hascolor': False,
 'lastseen': None,
 'manufacturername': 'Philips',
 'modelid': None,
 'name': 'Dimmable light 16',
 'state': {'alert': 'none', 'bri': 0, 'on': False, 'reachable': True},
 'swversion': None,
 'type': 'Dimmable light',
 'uniqueid': '00:17:88:01:02:a3:4d:67-0b'}
2020-06-12 18:49:40 DEBUG (MainThread) [pydeconz.websocket] {"e":"added","id":"16","light":{"etag":"b653f35ee1c1b2e17e8424ad9a0cf6a4","hascolor":false,"lastseen":null,"manufacturername":"Philips","modelid":null,"name":"Dimmable light 16","state":{"alert":"none","bri":0,"on":false,"reachable":true},"swversion":null,"type":"Dimmable light","uniqueid":"00:17:88:01:02:a3:4d:67-0b"},"r":"lights","t":"event","uniqueid":"00:17:88:01:02:a3:4d:67-0b"}
2020-06-12 18:49:40 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new light.deconz entity: light.dimmable_light_16
```

closed 2020-07-11 by stale[bot]