dresden-elektronik/deconz-rest-plugin #2416: Rule execution seems faulty

esseki, 2020-02-04:
Hi,

I'm using Deconz with a Conbee II for a while now. Up to the point that I'm creating my own rules from the API.

With the latest stable release (2.05.71) I have noticed some weird behaviors from the rule engine :

### 1) Rule seems to be executed more than once 

Here is the rule I'm getting from the API **before triggering it** 
{
  "actions": [
    {
      "address": "/groups/17/action",
      "body": {
        "on": false,
        "transitiontime": 4
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/12/state/buttonevent",
      "operator": "eq",
      "value": "1002"
    },
    {
      "address": "/sensors/12/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/groups/17/state/any_on",
      "operator": "eq",
      "value": "true"
    }
  ],
  "created": "2020-02-03T22:31:10",
  "etag": "8026077baab091e5ab80a9138d98c0fc",
  **"lasttriggered": "2020-02-04T20:55:50",**
  "name": "SWITCH OFF Room Salon",
  "owner": "31FD8CEC70",
  "periodic": 0,
  "status": "enabled",
  **"timestriggered": 37**
}

This rule is supposed to switch off a group when a button Is pressed on a switch (Jung 4 gang switch) if at least one light of the group is on. 
We can see that before I press the button on the switch the rule has been triggered 37 times.

Just in case here is the state of the switch before I press the button :

{
  "config": {
    "group": "58571",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "8026077baab091e5ab80a9138d98c0fc",
  "manufacturername": "Insta",
  "mode": 1,
  "modelid": "WS_4f_J_1",
  "name": "Interrupteur Jung Salon",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2020-02-04T20:55:50"
  },
  "swversion": "10023202",
  "type": "ZHASwitch",
  "uniqueid": "00:0f:17:12:41:02:09:d3-01-1000"
}

Now if I press the button on the switch to trigger the rule (to switch off the group), the rule becomes :

{
  "actions": [
    {
      "address": "/groups/17/action",
      "body": {
        "on": false,
        "transitiontime": 4
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/12/state/buttonevent",
      "operator": "eq",
      "value": "1002"
    },
    {
      "address": "/sensors/12/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/groups/17/state/any_on",
      "operator": "eq",
      "value": "true"
    }
  ],
  "created": "2020-02-03T22:31:10",
  "etag": "167d7bdf570452e965a01ac13d535194",
  "lasttriggered": "2020-02-04T21:06:16",
  "name": "SWITCH OFF Room Salon",
  "owner": "31FD8CEC70",
  "periodic": 0,
  "status": "enabled",
  **"timestriggered": 50**
}

We can see that it has been triggered 13 times when I pressed the button only once on the switch. I would like to precise that with previous versions of Deconz the behavior was fine for this rule.

Just in case here is the state of the switch after I pressed the button :

{
  "config": {
    "group": "58571",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "167d7bdf570452e965a01ac13d535194",
  "manufacturername": "Insta",
  "mode": 1,
  "modelid": "WS_4f_J_1",
  "name": "Interrupteur Jung Salon",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2020-02-04T21:06:16"
  },
  "swversion": "10023202",
  "type": "ZHASwitch",
  "uniqueid": "00:0f:17:12:41:02:09:d3-01-1000"
}

### 2) Rule is not triggered the first time 

This issue is not new with 2.05.71. I have noticed it with previous versions of Deconz. But I never took time to report it properly.

{
  "actions": [
    {
      "address": "/lights/32/state",
      "body": {
        "on": true
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/10/state/buttonevent",
      "operator": "eq",
      "value": "4002"
    },
    {
      "address": "/sensors/10/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/lights/32/state/on",
      "operator": "eq",
      "value": "false"
    }
  ],
  "created": "2020-02-04T21:14:18",
  "etag": "4b111d4537893fe01ff02193569bbe6a",
  "lasttriggered": "none",
  "name": "SWITCH OFF/ON prise veilleuse",
  "owner": "31FD8CEC70",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 0
}

This rule is supposed to switch on one light (actually an Innr SP120 plug) when a button is pressed on a switch (still a Jung 4 gang switch).

Just in case here is the state of the switch before I press the button :

{
  "config": {
    "group": "17752",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "fca06f29fa52536a1e0ae3af75e74c81",
  "manufacturername": "Insta",
  "mode": 1,
  "modelid": "WS_4f_J_1",
  "name": "Interrupteur Jung Ch Enfants",
  "state": {
    "buttonevent": 4002,
    "lastupdated": "2020-02-04T19:05:06"
  },
  "swversion": "10023202",
  "type": "ZHASwitch",
  "uniqueid": "00:0f:17:12:41:02:09:e5-01-1000"
}

After pressing the button (4002) the state of the switch is the same : 

{
  "config": {
    "group": "17752",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "fca06f29fa52536a1e0ae3af75e74c81",
  "manufacturername": "Insta",
  "mode": 1,
  "modelid": "WS_4f_J_1",
  "name": "Interrupteur Jung Ch Enfants",
  "state": {
    "buttonevent": 4002,
    "lastupdated": "2020-02-04T19:05:06"
  },
  "swversion": "10023202",
  "type": "ZHASwitch",
  "uniqueid": "00:0f:17:12:41:02:09:e5-01-1000"
}

It does not seem to have detected the button pressed. Until I press another button (3002) : 

{
  "config": {
    "group": "17752",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "15331e5b860fc70decf50e28c157a80d",
  "manufacturername": "Insta",
  "mode": 1,
  "modelid": "WS_4f_J_1",
  "name": "Interrupteur Jung Ch Enfants",
  "state": {
    "buttonevent": 3002,
    "lastupdated": "2020-02-04T21:27:05"
  },
  "swversion": "10023202",
  "type": "ZHASwitch",
  "uniqueid": "00:0f:17:12:41:02:09:e5-01-1000"
}

And then finally if I press the first button (4002) it will be detected : 

{
  "config": {
    "group": "17752",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "af294dca63cd65efa24770e02ceddc79",
  "manufacturername": "Insta",
  "mode": 1,
  "modelid": "WS_4f_J_1",
  "name": "Interrupteur Jung Ch Enfants",
  "state": {
    "buttonevent": 4002,
    "lastupdated": "2020-02-04T21:29:24"
  },
  "swversion": "10023202",
  "type": "ZHASwitch",
  "uniqueid": "00:0f:17:12:41:02:09:e5-01-1000"
}

and the rule will finally be triggered : 

{
  "actions": [
    {
      "address": "/lights/32/state",
      "body": {
        "on": true
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/10/state/buttonevent",
      "operator": "eq",
      "value": "4002"
    },
    {
      "address": "/sensors/10/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/lights/32/state/on",
      "operator": "eq",
      "value": "false"
    }
  ],
  "created": "2020-02-04T21:14:18",
  "etag": "be404dd346afa22830c8c4c822af4dab",
  "lasttriggered": "2020-02-04T21:30:07",
  "name": "SWITCH OFF/ON prise veilleuse",
  "owner": "31FD8CEC70",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 1
}

Would any log be helpful to investigate those issue ? In this case just let me know how I can access those and I will provide.

Thank you in advance for your help.

ebaauw, 2020-02-04:
Ad 1: never seen this before.  Could you double-check what web socket notifications are issued when pressing the button?  You might want to run deCONZ with `--dbg-info=2` to see debug messages for evaluating rule events and triggering rules.

Ad 2: The sensor resource’s `state.lastupdated` hasn’t changed, so the rule doesn’t get triggered.  It should update with each button press, even when `state.buttonevent` remains unchanged.  I don’t know this switch - maybe it needs the first button press to wakeup; maybe it needed to reconnect to the network, causing the message to be lost; maybe there’s a bug in how the switch is supported.  Hard to tell without sniffing the ZigBee messages sent by the switch.

esseki, 2020-02-05:
Hi @ebaauw,

tank you for your quick response.
This evening when I restarted Deconz to activate the logs, I could not reproduce issue nb 1. I am running Deconz from an Arm board (comparable to a RPi) and it is so much reliable that I tend to never restart it. I should maybe have tested this simple trick before posting here. I apologize for that.

Regarding issue nb 2 : I agree with you that this is not a rule triggering issue since the button event does not seem to have been received. What I can tell you is : 
- the switch is this one : https://www.jung.de/en/online-catalogue/374871630/
- it is marked as supported in Deconz here : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices
- I don't think this is an issue with switch needing to wakeup as any other button from the same switch is working from the first press. 
- Would it be possible for you to explain me how to sniff the Zigbee messages to check if there is anything wrong with this switch support ?

Thank you in advance

ebaauw, 2020-02-06:
You’ll need a second ZigBee radio device with firmware supporting sniffing.  I use a ConBee with ZShark on a dedicated RPi, sending the sniffed packets to WireShark on my macBook.

closed 2020-06-12 by stale[bot]