dresden-elektronik/deconz-rest-plugin #2420: Tradfri Bulbs not changing color temp

MysticNynja, 2020-02-05:
Similar or exactly like this issue:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/629

I have 2 Ikea [bulbs](https://www.ikea.com/us/en/p/tradfri-led-bulb-e26-1000-lumen-wireless-dimmable-white-spectrum-opal-70408492/) and they can be configured and used within the Deconz Add On in Home Assistant.

When I have them outside of the add on I cannot change the color temp and all attempts to send events to them are not working. There seems to be some syncing issues between the two components as well.

Here is a .gif to explain what I mean
[image]

p0rys, 2020-02-07:
I have a similar problem with the colortemperature of the ikea lamps. I try to change to temperature (which is working in phoscon) from ioBroker via the "ct" object and then the lamp will switch to the opposite color. I think the problem is that the colormode is set to xy but every time i try to change it to ct the deconz adapter crashes...

Does anyone has the same problem or knows a solution/workaround? Thanks!

ebaauw, 2020-02-07:
`state.colormode` is read-only.  It's updated automatically when setting `state.ct`, `state.xy`, or `state.hue`/`state.sat`.

Flurkmark, 2020-02-08:
> Does anyone has the same problem or knows a solution/workaround? Thanks!

Is it related  to #1861 ?
You could try the workaround  in the thread.

MysticNynja, 2020-02-08:
I'll do this today and see if it works

fullcavediver, 2020-02-10:
I can confirm the problem with Ikea E27 WS opal 980lm / firmawe version: 2.3.007

jasimancas, 2020-02-10:
Same error here.

Mimiix, 2020-06-05:
As the original issue is resolved, this issue will be closed.

closed 2020-06-05 by Mimiix

johnekarlsson, 2020-08-02:
The issue is not solved, unfortunately!

Mimiix, 2020-08-03:
@johnekarlsson If there is: Please open another issue.

However: is it still relevant for you? as you said you moved away to hue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2526#issuecomment-667729396