dresden-elektronik/deconz-rest-plugin #2421: Centralite 3321-S Open/Close Sensor & Centralite 3325S-S Motion Sensor

bal0o, 2020-02-05:
Trying to move away from SmartThings and get everything on deconz - I have a couple of sensors that are not working. They are detected in the DeConz app, but not showing in the Phoscon web app. Strangely they are picked up in HA, and report basic status (on / off) OK. 

The first is a door open/close sensor - The sensor is this one - it appears as a CentraLite 3321-S, but is smart things branded that came with the hub;

[image]

Details below - hopefully this is what you need! If you need any more just let me know

[image]

[image]

Second is a Motion sensor from smart things doing the same - appears as a CentraLite 3325-S;

[image]

[image]

[image]

SwoopX, 2020-02-05:
The devices seem to be supported looking at the code. The reason why you don't see them in Phoscon is because it is a different software and closed source on top of that.

jjanderson, 2020-02-07:
These devices do not show up in the UI only via VNC. Additionally, when trying to delete them (i sold mine) they will simply not get deleted.... Please advise?

Thanks

SwoopX, 2020-02-07:
What exactly do you mean with UI?

"Orphan devices" are unfortunately quite common as there's currently no way to permanently delete them as far as I know. The only way is by hacking deconz' database and for that, you should know what you're doing.

jjanderson, 2020-02-07:
User interface being webui vs vnc where you can view the network.

When I delete them from the vnc application, i would expect them to be deleted and therefore the application should delete them from the db. If not then I would argue it should.

Otherwise, do tou know what db it uses? Maybe I can fiddle around with one if the backup files?

It is really annoying as I want rules in homeassistant based around “unavailable devices” and any orphan sensors will get pulled through...

SwoopX, 2020-02-07:
> User interface being webui vs vnc where you can view the network.

So I conclude you mean Phoscon. Devices need to be integrated seperately and there's unfortunately nothing we can do here.

> When I delete them from the vnc application, i would expect them to be deleted and therefore the application should delete them from the db. If not then I would argue it should.
> 
> Otherwise, do tou know what db it uses? Maybe I can fiddle around with one if the backup files?

Agreed, but that's the current situation. DB is sqlite, located in the home directory of the user running deconz (e.g. /root/.local/share/dresden-elektronik/deCONZ/zll.db). You may want to make a backup first to be on the safe side and be sure to have deconz shut down. The following tables should require amendments, if I recall correctly: devices, device_descriptors, device_gui, sensors, nodes. Good luck!

jjanderson, 2020-02-08:
Not being able to delete sensors that are intended to be connected to hardware with the software provided is a pretty fundamental shortcoming in my humble opinion. It’s like selling a door and then saying, yea you can only use it to go in, but once you are in you cant go out... sorry, that is the nature of the doors we sell....

Mimiix, 2020-06-05:
It seems this issue is solved. If it is not, please re-open!

closed 2020-06-05 by Mimiix