dresden-elektronik/deconz-rest-plugin #2422: Websocket is closed by unknown errors

stormmurdoc, 2020-02-05:
I use FHEM to connect to the deconz web service. After a while the connection is closed and reopened. Is there a way to better analyze if it is FHEM or deCONZ?

```
Platform: docker
Version: [marthoc/deconz] Current deCONZ version: 2.05.72

```
```
13:12:08:516 Remove websocket 192.168.1.101:56910 after error Unknown error,
13:13:06:474 Remove websocket 192.168.1.101:56940 after error Unknown error,
13:14:06:497 Remove websocket 192.168.1.101:56976 after error Unknown error
```
On the other side (FHEM)

`2020.02.05 13:14:06.498 3: deCONZ: websocket opened to 192.168.1.119:8443
2020.02.05 13:13:06.544 3: deCONZ: websocket: Switching Protocols ok
2020.02.05 13:13:06.474 3: deCONZ: websocket opened to 192.168.1.119:8443
2020.02.05 13:12:06.591 3: deCONZ: websocket: Switching Protocols ok
2020.02.05 13:12:06.500 3: deCONZ: websocket opened to 192.168.1.119:8443
2`
There are hundreds of messages a day. I would be very happy about a tip.
Thanks in Advance
Patrick

Smanar, 2020-02-05:
Good question.
You are not alone, I have see the same issue on many applications, without answer yet.
But I can't find a "common point", it seem like totaly random.

stormmurdoc, 2020-02-05:
It's good to hear that I'm not alone :-)
But at the same time it makes me sad to hear that it is already known and is not corrected. :-/
But I don't give up hope - maybe one of the programmers will have mercy! *written on the christmas wish list*

essera, 2020-03-13:
i have a new docker installation with conbeestick since 5 days.

I have the same issue too. 
The messages arrive every minute.

21:38:26:492 Remove websocket 192.168.178.137:39992 after error Unknown error
21:38:26:495 New websocket 192.168.178.137:40024 (state: 3) 
21:39:13:905 Current channel 25
21:39:13:936 Device TTL 2100 s flags: 0x7
21:39:26:555 Remove websocket 192.168.178.137:40024 after error Unknown error
21:39:26:558 New websocket 192.168.178.137:40052 (state: 3) 

The port is counting up - at the end there is no free port number avaible - what will happen then  ??

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive . If it is not, please re-open!

stormmurdoc, 2020-06-06:
How can I reopen it?

Mimiix, 2020-06-06:
I've re-opened it ;) I added the label to bring it under attention.

fs79, 2020-06-26:
Found this issue and have the same challenge.
Have running FHEM and deconz on Raspberry Pi4 4GB mit Debian Buster und get the message constantly after 5-6 minutes.
How can we solve it?
Do you need logs or something?

It seems that functionality is fine but don´t like the log flooding and think that suppressing the message is the solution.
Is it works as designed that the connection is closed.
Some question which could shed some light on this issue.

closed 2020-07-25 by stale[bot]

pljakobs, 2020-10-06:
hmm... this is still happening in my installation - for no apparent reason. 
I suggest re-opening this bug.

Mimiix, 2020-10-06:
@pljakobs the websocket is managed by the client. Meaning that the client(user) of the api is faulty.
Open a own issue if you feel this is incorrect.