dresden-elektronik/deconz-rest-plugin #2423: gibt es den phoscon.de Support noch?

horchi, 2020-02-05:
Ich habe vor ~ 2 Monaten eine Supportanfrage an den phoscon gestellt aber bislang noch kein Feedback erhalten. Gibt es den Support noch? 

Hier meine Mail von damals: 

ich verwende ConBee / Phoscon in Verbindung mit dem Home Assistent und habe ein paar Fragen dazu:

-   wo finde ich eine Erläuterung der Kommandozeilen Optionen von deCONZ?
-   wo finde ich eine Beschreibung der Optionen der config.ini und wo diese zu finden ist
-   laut Foren ist es möglich mittels --http-listen= die IP zu spezifizieren, das klappt insofern das dann an dieser nun der HTTP LISTENER lauscht, leider sucht Home Assistent nach wie vor an der falschen! Ursache scheint das es wohl einen Dienst im Internet gibt welcher die Clients verwaltet und deCONZ dort nach wie vor die falsche IP hinterlegt (mein Host hat viele IPs).  Wie steure ich welche IP publiziert wird?

    Ich habe die Verbindung mit Home Assistent nun zum funktionieren gebracht indem ich in der Datei ...../site-packages/pydeconz/utils.py (Zeile 83) meine IP fest eingetragen habe:
        for bridge in response:
            bridges.append({'bridgeid': bridge['id'],
                            'host': '192.168.200.111',
                            'port': bridge['internalport']})

-   Wie verhindere ich das nach hause telefoniert wird nebst publizieren meiner IPs ins Internet nach https://phoscon.de/discover und stelle den Service so um das er rein Lokal in meinem Netzwerk funktioniert?

Nebenbei, wie verträgt sich das zentrale speichern (im Internet) meiner IP welche mit mir in Verbindung gebracht werden kann ohne meine explizite Zustimmung mit der DSGVO?

Danke und Grüße
Jörg

Sjonnie2018, 2020-02-05:
I tried twice to contact them to get support with one issue. I looked at there support page where I found this:

> I have one more question… The answer was not yet there?
Our support will be **happy** to help you. 

I have no response either. Not even an e-mail confirming I sent the support request.

I guess they are simply not **happy** at this moment ;-)

About your GDPR issue: I am not a lawyer but I guess we have agreed tot that through the EULA.

manup, 2020-02-05:
Hi all, sorry for the current support situation. Our support is very busy answering mails and telephone calls. But to be honest since the ConBee II got popular like crazy in recent months we struggle to answer all mails and got overrun by the amount. So far we've put more people onto it (and devs helping out) and we are also looking to extend our support team:

By the way, we're hiring :)

https://twitter.com/phosconde/status/1213128357665296384

https://www.dresden-elektronik.de/stellenangebot/mitarbeiter-f%C3%BCr-die-technische-kundenbetreuung-m-w-d.html

Anyway it feels like for every mail which gets answered we receive 5 new mails. I'm not sure if mail can scale anymore so we are also thinking of provide support in other ways.

One thing that we working on a while now is to address the software issues which generate the most support requests, like ConBee not connected after restarts and "whole network lost" kind of issues. Hopefully the updates will be out soon and relax the issues.

>Wie verhindere ich das nach hause telefoniert wird nebst publizieren meiner IPs ins Internet nach https://phoscon.de/discover und stelle den Service so um das er rein Lokal in meinem Netzwerk funktioniert?

Currently there is no simple GUI / Phoscon App page for this, disabling the discover server will be added in the gateway settings later on, as well a proper GDPR notice on the login screen (with an option to disable the discovery there). The login page shows it's age but it's on our 2020 todo list to address this asap. As side note: we don't store information permanently, the discover server runs only in RAM.

Meanwhile the discovery service can be disabled in the sqlite database manually. This is done by setting the `announceinterval` in the table `config2` to 0.

deCONZ needs to be closed before updating the database file `zll.db`.

```
cd ~/.local/share/dresden-elektronik/deCONZ
sqlite3 zll.db 'UPDATE config2 SET value=0 WHERE key = "announceinterval"'
```

Sjonnie2018, 2020-02-05:
Thank you very much for your reply and explanation. It might be an idea to manage buyers expectations by stating something similar on your website. It is no big deal to wait or turn to others sources of support (such as over here) as long as we know what's going on.
I hope you will get things sorted (and a new colleague) soon because the Conbee II is simply brilliant :)

horchi, 2020-02-07:
very nice to hear (and see) that the development is going on :).
Thank you for the feature to disable the discovery service!

Regarding the 'Home Assistent' pydeconz plugin, does this alao belong to you? If so, is there a trick to setup the ip/port of the phoscon gateway without patching the code?

Mimiix, 2020-06-05:
I think the discussion is closed. If anything, please re-open!

closed 2020-06-05 by Mimiix