dresden-elektronik/deconz-rest-plugin #2425: state.reachable

rccardany, 2020-02-05:
Hello, 

i am using the conbeeII Stick on my raspberry pi
is there a possibility to deactivate the state.reachable or ingrease the time when state.reachable gets false?
I need this option for my project.

Thanks

JuanitoAlimana, 2020-02-05:
No

rccardany, 2020-02-05:
maybe you can say some more words than only "NO"
Can you tell me why?

Smanar, 2020-02-05:
Lol.
No, it's not possible without code modification (or new feature).

rccardany, 2020-02-05:
ok.. of course it is not to do this without code modification.
But maybe someone can tell me where i have to change the code.
I think this was the question at all

Smanar, 2020-02-05:
So the modification is only for you ?
You want to remove this field, or it keep a fixed value ?
I think deconz use this value for itslef too, so better to change only the API result.

Make a search on code with "RStateReachable"

rccardany, 2020-02-05:
it would be ok for me if the state stays always true. Or maybe stays true for xthousend secounds after switch off

maybe i should describe why i need this:
i am using the deconz app in iobroker and it works fine. 
when i switch off my philips hue lamp (connectet to deconz) the iobroker is switching off the power via homematic relay of the hue lamp (i need this to switch on the lamp with a normal wall swicht again)
when i directly switch on again the deconz app communicates to iobroker and there the homematic realy is switched on again. everything goes fine.

but when the hue is some secounds without energy the state.reachable turns into false.
the problem is that there is no communication between iobroker and deconz when state.reachable is false. so i am not able to switch on my homematic relay again. 

I need the permanent communication between iobroker and deconz to switch the relay correctly.

JuanitoAlimana, 2020-02-05:
I dunno.

Smanar, 2020-02-05:
There so much thing I don't understand on your procedure but if you want to have alaways true as reachable you can take a look in the rest.light.cpp, fonction DeRestPluginPrivate::handleLightEvent(const Event &e).

But for me it will change nothing. What is the application that will read this field ? better to change it instead of deconz api.

JuanitoAlimana, 2020-02-05:
...

rccardany, 2020-02-06:
@Smanar 
these are the first steps with the conbee stick at all - so please forgive me
i really would like to try this but i cant find the file rest.light.cpp on my raspberry pi

Smanar, 2020-02-06:
It'isn't on your raspberry.
And after modification, you need to recompile it, it's described on github home page.

But have you take a look on ibroker side ?

rccardany, 2020-02-09:
ok o got that thanks.
the problem begins before iobroker.
the deconz software is not communicating with the iobroker addon when state.reachable is false.
can you tell me what i have to change at the lines you mentiont DeRestPluginPrivate::handleLightEvent(const Event &e)

Smanar, 2020-02-09:
You can try around line 2073
>state[key] = item->toVariant();

rccardany, 2020-02-10:
i did not really get along with it
maybe i can describe my problem in another way

in the webinterface phoscon a light will become greyed out when the light is not reachable (for example because there is no power at the light) 
is it possible to increase the time of the timeout or never let the light grey out ?

Smanar, 2020-02-10:
You can try just after the loop

`state["reachable"] = true;`

rccardany, 2020-02-13:
it seams that i got one step further with it.
i have added (rest.light.cpp line 255)
state["reachable"] = true;  
now the lamp is always reachable in iobroker and also in the phoscon webinterface.
but when i hit the clickable lamp-button nothing happens.
do i need to activate something else?

Smanar, 2020-02-13:
But your lamp is reachable or not when you clic on the "lamp button" ?

rccardany, 2020-02-13:
Yes in iobroker i can see that the State reachable is true and in phoscon the light is not greyed out as before.
But when I hit the lamp - button Nothing happens - nothing in iobroker and nothing in phoscon.

Smanar, 2020-02-14:
No I mean, in reality, the hardware is online or not ?

rccardany, 2020-02-14:
Yes of course but for me there is an difference:
When the lamp is "really" reachable the State on is pushen to iobroker when i hit the on button
When the lamp is "only manipulated" reachable the State on is not pushen to iobroker when i hit the on button (then i can Not Switch the Power relay on to Power the lamp

But i found out something new 
When i put a lamp Info a Group and hit the Group-on Button the State on of the lamp is pushen to iobroker. It doesent Matter if the lamp is reachable it self. And acactly this is the Feature i would like to have with hitting the on button of the lamp

rccardany, 2020-02-20:
is there a possibility to switch on a light like switching on a group to solve this ?

Smanar, 2020-02-21:
I still don't understand. Why you can't set on a light ?

ebaauw, 2020-03-15:
`reachable` is utterly useless and best ignored, see #2590.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix