dresden-elektronik/deconz-rest-plugin #2426: Issues pairing IKEA LED1836G9 806lm bulbs

repvik, 2020-02-05:
I have three LED1836G9 bulbs that I simply cannot pair with deconz/conbee.

I reset the bulbs by switching them off/on 6 times. They all start pulsing slowly, but none of them pair. They keep on pulsing for a couple of minutes before they give up and return to normal brightness. 
I've tried pairing with the dongle physically resting on the bulbs, but to no avail. 
I've also tried resetting a bulb, then switch it off/on again (sometimes necessary with the other IKEA bulbs)

I have more than 20 other bulbs, and none of them have had this issue.

Smanar, 2020-02-05:
Have you try the touchlink feature ? You need have the bulb realy close the conbee.
In phoscon/help/old webapp

mr2web, 2020-02-07:
I'm experiences the same issue... and another issue with this bulb is that it gets really HOT in its base. That hot that one can not touch it. But that is not an issue caused by the deconz-rest-plugin though...

theninth, 2020-02-08:
I have the same issue with LED1835C6. I guess Deconz need some code update to support these new bulbs.

Smanar, 2020-02-08:
But it's strange, bulbs are working natively (most of them).
Nothing appear in deconz (and not Phoscon) ? Have you tried with deconz only ?

theninth, 2020-02-08:
Thanks so much for taking the time to answer! This stuff is beyond me, but I went to the deconz app and open the "permit join" dialog. I put in 254 s and pressed "set", "get" and "ok" (or in a different order, I don't know). Then I did a reset on the bulb by flickering on and off 6 times. It really looks like it paired, a new node in deconz appeared and the bulb flickered like it use to do on a succesful pairing. But I have no clue how to get any further  from here. Can I export the node info in some way and send it to you? Or how does this work?

theninth, 2020-02-08:
I tried pair it like 15 times earlier today in and I know I successfully had it resetted because it showed the signal flickering but it never worked. Now I did it through deconz, then in the old web interface and a third time in the phoscon UI. I have no clue what I did, but it's working!

repvik, 2020-02-08:
How did you pair it in the old interface?

theninth, 2020-02-08:
First I tried pairing it through Touchlink (under settings), it didn't find anything, so I tried "open network", now it appeared in the UI, but I could'nt control it. Then I did it in Phoscon and it started working.

repvik, 2020-02-08:
Opening network in the old interface worked now. No clue why.

theninth, 2020-02-08:
It's really strange. I also bought a new Trådfri Filament light today. I had the same problem pairing it. Now I tried doing the same three things; (1) Permit join network in Deconz app and reset. (2) "Open network" in old webui and reset. (3) Searched light in Phoscon and reset.

I have no clue which of the things that made it work, but it works...

Smanar, 2020-02-09:
@theninth if you can see the bulb in deconz and not phoscon it's a good new.
Can you show the picture of the node (with cluster and if you can basic attribute)

Once a device is in phoscon, you can make it add in api (but depend of api code, if there is a problem in code, it will not work).
1 - set phosocn in permit join.
2- return on phoscon, cluster 0000, press read attribute.
3 - you can too need to read some missing descriptor (the first circle in title bar)

DBa2016, 2020-03-06:
I'd like to second this. Tried every method - opening network (with reset), Touchlink with reset, Phoscon scan with reset - no success.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix