dresden-elektronik/deconz-rest-plugin #2427: Conbee cannot read aqara window/door sensor

klauskkinski, 2020-02-05:
Hi, folks,

I am new to the conbee world. I got a new conbee II stick and a couple of aqara window/door sensors. Based on what I have read it should be piece of cake to connect everything together. Except that it is not :-( 

So, I installed the deConz. No problem. It found the conbee stick without a problem. I paired then the door sensors with the conbee stick. 

The sensors connect, however the conbee cannot read anything useful. I also cannot set e.g. the name of the sensor. Basically beyond having a connection and reading the node info nothing happens. Phoscon does not recognize the sensor... Each sensor was bought from a different source; they are different physically, result is the same. I must be doing something wrong, but have no clue what. Help would be appreciated.

deConz v 2.05.71
USB firmware 264A0700

Screenshots are below... I can provide more info if necessary.

Thanks for helping,
Klaus

[image]
[image]

TilBumble, 2020-02-05:
Hi.

Perhaps you should try the phoscon app.
It is a interface running in a web browser, accessing your running deconz program.

It has a more user friendly approach and offers the possibility to add, rename and so on.
(sensors, switches and Lights)

You can find the description here 
[https://phoscon.de/en/app/doc](url)
in german and english.

ebaauw, 2020-02-05:
> The sensors connect, however the conbee cannot read anything useful.

These are battery-powered devices, that sleep most of the time, meaning they turn off their radio to preserve battery life.  You can wake them by briefly pressing the reset button (the LED blinks blue) or by moving the magnet.

They report their state through the _On/Off_ cluster.  When opened, the _OnOff_ attribute turns true; when closed, it turns false.  No need to setup any reporting nor bindings.

> I also cannot set e.g. the name of the sensor.

You can set the name through the API, see the API documentation (linked to from the README).

> Phoscon does not recognize the sensor.

I don't know if Phoscon supports these, but the REST API does.  Looking at your screenshot, the pairing didn't succeed in full: deCONZ hasn't read the descriptors.  You can try and re-pair the sensor, or read these manually from the left drop down dot on the node.  Make sure to wake the sensor and read the _Node Descriptor_ and _Simple Descriptor(s)_.  When successful, the right dot should appear, and you can use it to drop-down the endpoints and clusters.  Next, search for devices from Phoscon, and read the attributes of the _Basic_ cluster (making sure to wake the sensor).  That should create the REST API resource, and change the name displayed on the node.  See the User Manual under Help for details on how to interact with deCONZ through the GUI.

klauskkinski, 2020-02-06:
> Hi.
> 
> Perhaps you should try the phoscon app.
> It is a interface running in a web browser, accessing your running deconz program.

Hi, TilBumble,

Phoscon does not recognize the sensors at all. The sensor discovery fails. 

Regards,
Klaus

klauskkinski, 2020-02-06:
> > The sensors connect, however the conbee cannot read anything useful.
> 
> These are battery-powered devices, that sleep most of the time, meaning they turn off their radio to preserve battery life. You can wake them by briefly pressing the reset button (the LED blinks blue) or by moving the magnet.

The sensors are not in sleep mode -- I've been moving the magnets. One of the sensors also has a LED that responds to detecting the magnet.

> They report their state through the _On/Off_ cluster. When opened, the _OnOff_ attribute turns true; when closed, it turns false. No need to setup any reporting nor bindings.

Right, this doesn't seem to work at all. 
 
>  Looking at your screenshot, the pairing didn't succeed in full: deCONZ hasn't read the descriptors. You can try and re-pair the sensor, or read these manually from the left drop down dot on the node. Make sure to wake the sensor and read the _Node Descriptor_ and _Simple Descriptor(s)_. 

I tried multiple times. Many times... no luck... 

> When successful, the right dot should appear, and you can use it to drop-down the endpoints and clusters. Next, search for devices from Phoscon, and read the attributes of the _Basic_ cluster [...]

Phoscon does not recognize the sensors at all. 

I'm starting to suspect the conbee II USB stick. :-(

Regards,
Klaus

ebaauw, 2020-02-06:
> One of the sensors also has a LED that responds to detecting the magnet.

So they're different models?  Xioami have a history of changing the internals of their devices.  Do any of them show the right drop-down circle?

I've seen some intermittent issue that deCONZ won't send the _Node Descriptor_ requests, when the network is still open.  Did you try to read the descriptors while the network was closed (the search from Phoscon was finished/off)?

klauskkinski, 2020-02-06:
> So they're different models? Xioami have a history of changing the internals of their devices. Do any of them show the right drop-down circle?

No, none. 

> Did you try to read the descriptors while the network was closed (the search from Phoscon was finished/off)?

I'm not sure what do you mean by reading the descriptors? I may have done that just don't call it this way... ? Can you elaborate a bit, please?

Klaus

klauskkinski, 2020-02-06:
@TilBumble I definitely cannot read simple descriptors... When I push the butotn, it simply jumps out a second or two later

[image]

klauskkinski, 2020-02-06:
I have also tried to use the USB stick on the RPI and Windows 10.... same result... Not sure if it helps or not...

ebaauw, 2020-02-06:
Bummer.

Looking at the mac address, 0x00158d0003xxxxxx, and 0x00158d0004xxxxxx, they seem newer devices than my `lumi.sensor_magnet.aq2` sensors, which have 0x00158d0001xxxxxx.  Not sure if that means anything though.

> I'm not sure what do you mean by reading the descriptors?

Indeed, checking the square left of the _Read Node Descriptor_ and _Read Simple Descriptor(s)_.  These contain the ZigBee endpoints and clusters, which deCONZ uses to populate the right circle.  If these aren't read, the pairing process stalls, and no REST API resources are created.

> I have also tried to use the USB stick on the RPI and Windows 10.... same result... Not sure if it helps or not...

Did you rule out any radio interference?  No USB-3 devices connected, ConBee II connected to a USB-2 port, no bluetooth or wifi; USB extension cable between PC/RPi and ConBee II?

klauskkinski, 2020-02-06:
> Did you rule out any radio interference? No USB-3 devices connected, ConBee II connected to a USB-2 port, no bluetooth or wifi; USB extension cable between PC/RPi and ConBee II?

This is another avenue to investigate. The windows box has a builtin WiFi, but not bluetooth... However, the sensors are literally 5 cm from the USB stick. I'll have to go and buy an extension USB cord, but I'm not convinced it's going to change anything simply because the sensors almost touch the USB stick... Not today though, it's evening here and the stores are closed... Yup, a real bummer... I wish there was a good troubleshooting guide or something similar to at least pinpoint the direction of the problem (e.g. interference report, sensor handshake / reading log to see if there are any obvious errors, signal strength report <-- my lines from the sensor to the stick are green in deConz, but yeah, I know...). 

I'll try with the cable in the next day or so and see if it changes anything... 

Thanks for all the help, @ebaauw -- really appreciated.

Regards,
Klaus

klauskkinski, 2020-02-10:
Just to report -- got an USB extension cable... no luck, still can't completely read sensors... Looks like my Conbee is somehow broken or at least I can't get it up and running... 

Also, I have emailed info@ and support@ dresden Elektronik --- no response whatsoever so far (first email was sent 10 days ago). I'm getting a bit disappointed... 

Regards,
Klaus

etmrugl, 2020-02-10:
I have roughly the same issue. The door sensor is not recognized in the Phoscon app. It fails at 'Add new Seonsor'. I tried to pair 2, one of them is visible in deConz. The other is not visible at all.
[image]
If I open the door the sensor in deConz window on the raspberry it will blink blue.

I am also completely new to Zigbee and deConz. Any help is appreciated.

Regards, Ruud

etmrugl, 2020-02-11:
I think I found a workaround.
When I move the door sensor near the coordinator (the Conbee II) the pairing works perfect. So it might be the case that the doorsensor won't pair through a router.

regards, Ruud

klauskkinski, 2020-02-11:
> I think I found a workaround.
> When I move the door sensor near the coordinator (the Conbee II) the pairing works perfect. So it might be the case that the doorsensor won't pair through a router.

It doesn't work for me... My sensor is 5cm from the coordinator...

klauskkinski, 2020-02-15:
I am disappointed with the Dresden Elektronik support though. Bummer. First email to them was sent on 24 Jan, second one on 6 Feb. I have received one useless suggestion 15 min after I posted my previous complaint here and then they went quiet after my response. Really bad... 

I'm shipping the USB stick back to Amazon. It looks like a DOA item...  

Regards,
Klaus

klauskkinski, 2020-03-17:
Folks, I have received a new Conbee II stick and I am happy to report that everything works as it should. I have returned the old one to Amazon and received a refund.

closed 2020-03-17 by klauskkinski

Smanar, 2020-03-17:
Ok, so was hardware problem.

Happy for you ^^.

buzzdev, 2020-12-19:
So if i am having the same issues with ConBee 1 and Aquara sensors, even though i've flashed the latest firmware to ConBee, it means this is a hardware problem and i need to get a new ConBee 2 stick?

Mimiix, 2020-12-21:
@buzzdev I suggest opening a new user question issue.

SwoopX, 2020-12-21:
I share the solution this when I'm not on mobile. This is NOT an issue of the DE hardware.

SwoopX, 2020-12-21:
Solution can be found here, probably only steps 5 and onwards are required:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3528#issuecomment-743367042

Also, you will most likely need 2 persons for that: one continuously pushing the button on the device, the other one doing the required steps in deconz GUI.

If this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2427#issuecomment-584210999 is the result after you've done the procedure, you've done it either wrongly or unsuccessfully.