dresden-elektronik/deconz-rest-plugin #2428: Cannot login to Phoscon anymore

FlyingPersian, 2020-02-05:
hello
I'm running a RPI 3 b+ with Home Assistant and the deCONZ addon. Since today I wasn't able to login to the phoscon web front. It just  says login failed. Resetting the password doesn't work. Once I hit reset, a green button comes up to change the password, but pressing the button doesn't do a anything. Already removed the addon, rebooted the RPI and re-installed the addon, but Phoscon isn't reset; it just pretends as nothing happened. VNCing into the machine also shows all previous connections. Seems all the stuff is stored somewhere else and is not removed when removing. What can I do?

ncandido, 2020-02-05:
I have the same issue!

cobch, 2020-02-05:
@FlyingPersian  I have the same issue. I spent all my evening thinking I did something wrong but apparently there is something wrong and bigger that just me...

wucherpfennig, 2020-02-06:
Second that

WladyX, 2020-02-06:
I have the same issue! Anyone found an workaround?

michaelrosenberg300, 2020-02-06:
Same issue here. Can't get "password reset" to work either.

stewie777, 2020-02-06:
Same problem here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2430

I tried to uninstall and install the addon again and now a new "problem" comes out :(  as you can see from the screen, hassio says it was impossible to install deconZ, but if I refresh the page I find deconZ installed
[image]

stewie777, 2020-02-06:
I'm afraid that the only solution to solve the problem is a fresh hassio installation

Sim7b, 2020-02-06:
Or a clean install of deconz outside of Hassio.

alexdepalex, 2020-02-06:
I didn't solve it yet, but for me it seems to be related to the ingress feature of hassio which uses tokens. For Deconz I'm getting a `Invalid token access` in the hassio_supervisor container whenever I try to login to the gateway, even though the token matches the one in `addons.json`.

As a workaround, you can expose the Deconz API backend directly on the host. You can enable it by going to the Deconz add-on configuration, scroll down to `Network` and enter `40850` in the `Host` column for the API backend.

In the end, this indeed looks like a hassio issue.

alexdepalex, 2020-02-06:
Seems to be fixed already. You just need to wait for the new supervisor release.

https://github.com/home-assistant/hassio-addons/issues/1049

closed 2020-02-06 by FlyingPersian