dresden-elektronik/deconz-rest-plugin #2430: deconZ can't reset password.. pls help m8

stewie777, 2020-02-06:
hello everyone, as the title says I write because I can't reset the password of deconZ
I tried to uninstall and install the addon, I tried to restore hassio from backup to before the problem arose, but without solving the problem.  when I reset password it's impossible for me to click on "set new password".  is there a way to reset the password from the terminal?
The default password will be "delight" right? Well.. it doesn't work
thank you all in advance for your support

stewie777, 2020-02-06:
[image]
I can't click on it

acid115, 2020-02-06:
But this is a hassio problem! 
Did you try to login with default user/pw?
delight/delight

stewie777, 2020-02-06:
> Ma questo è un problema di hassio!
> Hai provato ad accedere con l'utente / pw predefinito?
> gioia / piacere

[image]

as you can see if I try to access the addon ui web I see this page where I can only put the password, where can I try to log in with username and password?  maybe I miss some steps

cobch, 2020-02-06:
I have the same problem. Updated HassOs yesterday. Impossible to reset nor to have a new log in session to enter default password? How can I reset the password/username of the Deconz webui when the reset function is not working?

stewie777, 2020-02-06:
is it possible that the only solution is to do a new hassio installation?  it is impossible that deconZ cannot be accessed from the terminal to change and reset the password

cobch, 2020-02-06:
I have already installed Hassio from an earlier backup but same problem.

Highjack77, 2020-02-06:
I have the exactely same problem since today. any solution yet ?

cobch, 2020-02-06:
Today everything is working again... I have not done anything. Really strange. It may may some long delay between the password reset in Deconz?

cobch, 2020-02-06:
Update to the supervisor  199. See #1049 for more information.

busshi, 2020-02-09:
Hello I just bought a Conbee II but I succeed only once to log in. Impossible to change the password neither to connect with the one I give on the first install...
How did you make it work ? I am using it on a raspberry with Domoticz running.

Mimiix, 2020-06-05:
This issue seems to be resolved. 

@busshi , if you still have a issue, please create a new one.

closed 2020-06-05 by Mimiix