dresden-elektronik/deconz-rest-plugin #2434: Support for Xiaomi Bluetooth Temperature Smart Humidity Sensor?

3PO, 2020-02-09:
Does deconz support Xiaomi Bluetooth Temperature Smart Humidity Sensors?

[image]

bastshoes, 2020-02-09:
It's different protocol. deCONZ supports only zigbee devices.

Sjonnie2018, 2020-02-10:
There is a ZigBee version for this device "made" by Blitzwolf: https://www.aliexpress.com/item/4000640062775.html

G3rry71, 2020-02-12:
> There is a ZigBee version for this device "made" by Blitzwolf: https://www.aliexpress.com/item/4000640062775.html

Is it really supported in Deconz?
I have this https://www.aliexpress.com/item/4000578239423.html and it is not fully discovered

Sjonnie2018, 2020-02-12:
I don't know because I haven't bought it yet (but I am tempted :-) ). It would be great if you could tell me/us what is supported and what not. Preferably with some screenshots of the device in Deconz.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix