dresden-elektronik/deconz-rest-plugin #2435: Ikea motion sensor refresh time/detection problem

3kgtt, 2020-02-09:
New to this forum and its my first post, som im sorry if im in the wrong section! 

Samsung V3 hub 

bought my first Ikea motion sensor and found out that its to damn slow in detection and its refresh time. So i thought it was faulty and bought a new one. 
Still it detectes within 4-10 minutes! 
I have put the motion in the house entery so it can turn on the light swich when someone comes but and light switch is on for 40 sec

Problem is that its no where as fast as Xaomi motion sensors i have and this sensor dose detect, but then it takes from 4-10 minutes before next detection! 

Is there any fix? 
I need an Ip rated motion sensor and this is sadly the only one i can find

SwoopX, 2020-02-09:
I don't quite understand what role deconz plays here. Can you provide more details?

3kgtt, 2020-02-09:
> 
> 
> I don't quite understand what role deconz plays here. Can you provide more details?

Your write i think since i dont even know what deconz are 
I have googled a lot about the Ikea motion sensors and found biggest threads here. 
Im sorry once agian if i wrote in wrong section since im a newbie here.

wvuyk, 2020-02-09:
Please check next time. This github is about control of devices with deCONZ, not control with smartthings? You might need to check for smarthings control here?

tubalainen, 2020-02-15:
Home Automation is daunting and quite confusing. No worries @3kgtt.

But the deconz is a different platform from Samsung hub and they have as far as I know affiliation.

So your issues with the xaomi sensor should be addressed to Samsung and their forums. Take care and good luck with your home automation!

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix