dresden-elektronik/deconz-rest-plugin #2439: Ajax plug showing as a light

JamieP83, 2020-02-12:
Hey all,
i have this plug: https://www.amazon.co.uk/ZigBee-certified-Automation-Philips-SmartThings/dp/B07JYQVV7V/ref=sr_1_1?keywords=ajax+plug&qid=1581512011&sr=8-1

it "connects" to my ConBee2 however it shows as a light not as a plug/switch?

is there a way to update this somehow?

[image]

[image]

Smanar, 2020-02-12:
Nope, you can't do anything.
Can you show the node from deconz (to see the device type) ?

Someone have make a PR to correct the same bug on some device, but IDK if yours is concerned, cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1090

JamieP83, 2020-02-12:
thanks
is that not the 2nd picture i added? showing it as a "router" sorry? let me know if not and i can grab another image if it helps

[image]

Smanar, 2020-02-12:
If you press the second circle on the node, you will have a new window with I think "home automation" and the device type on title.

JamieP83, 2020-02-12:
this one sorry?

[image]

Smanar, 2020-02-12:
Yep it's this one, thx.
But unfortunately, it's not the same device at all than in the previous issue, yours is ZLL, and another strange thing, it have level control.

So, it can be a bug in deconz, or you device can realy have a "on/off light" type, but in all case, to correct this, you need to edit the code.

I have already tried to do that without the device and I was totally wrong ^^.

It's realy a problem for you ?

ebaauw, 2020-02-12:
It's not uncommon for On/Off devices to expose _Level Control_.  They respond to _Move to Level (with On/Off)_ so they can be controlled by wireless switches that send these commands instead of _On_ and _Off_.

Smanar, 2020-02-12:
Haaaa ok, it's for that, I have see lot of device where the api ignore the state.bri.

Smanar, 2020-02-12:
BTW @ebaauw there is no way to have the deviceId (haEndpoint().deviceId()) without adding a debug line in the code ?
I can't see this value in deconz.

ebaauw, 2020-02-12:
It's shown in the GUI under the node.  For example, the innr SP 120:
[image]
You see on the first line:
- The endpoint (0x01);
- A picture for the profile: [ZLL];
- A picture for the device [?]
- The name of the profile: "Light Ling"
On the second line:
- The name of the device: "On/off plug-in unit" (this is `type` in the REST resource).

The names are from `general.xml`, look them up to find the profile ID:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4187c8d04d076327de562d1b2b46939ec747ef26/general.xml#L2767
and the device ID:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4187c8d04d076327de562d1b2b46939ec747ef26/general.xml#L2775
Be sure to look for the device under the right protocol.

Alternatively, sniff the _Simple Descriptor(s)_ response.

Smanar, 2020-02-12:
Yep but in the GUI it's not the real one, but the deconz one after processing.
You can change it with some line code.

ebaauw, 2020-02-12:
> And to display a plug image we need to make a "ugly hack".

I don't think you should try and change the value if the device ID in the `deCONZ::` structures, which are managed by the core programme, and shown in the GUI.  You should only override the value of the `RAttrType` resource, which is managed by the REST API plugin, and shown in Phoscon and any other REST API client.

JamieP83, 2020-02-12:
Thanks for the replies guys
Is there something I need to do or any more info?

Smanar, 2020-02-13:
IDK.
I can make code modification to try, I think it will be easy, but you need to try the modification yourself.
So it mean you need to recompile my code, test it and if it don't work retry another code, and ect ...
The procedure to compile the code is on github page.

So meditate if you realy want to do that, it s so usefull for you ?

If you want to do it I need to the attribute list on cluster 0000 (basic), after you press the "read" button.

And if you can the JSON of the device, to be sure it have a bad "type" on it. To have it you can use your home automation application (with log) or if you haven't one https://dresden-elektronik.github.io/deconz-rest-doc/
You can too have it with enabling deconz log (if you don't have too much device).
Or you can too edit the database if you have a sql editor.

JamieP83, 2020-02-17:
sounds good
is this the attribute cluster?

[image]

Smanar, 2020-02-17:
Missed ^^.

First clic on the second circle on the right, you will have a cluster list, then select the basic one (0000), it will update the left pannel you have on your screenshoot, don't forget to press the "read" button before taking the screenshoot.

Smanar, 2020-02-19:
@JamieP83 I need something like that https://user-images.githubusercontent.com/59974444/74603402-7e816a00-50b3-11ea-91ed-07d0d18fec3b.png

To know the model id of the device.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix