dresden-elektronik/deconz-rest-plugin #2441: issues with xiaomi mijia window/door sensors.

bhushan0710, 2020-02-12:
Hello All,

I had a working setup of conbee ii via deconz addon and several zigbee smart plugs, xiaomi temp/humidity senors, motion sensors, aqara cube, xiaomi switches, ikea switches and xiaomi door sensors.

since last couple of days my xiaomi door sensors have disappeared, 

If i pair it through phoscon it works fine, but after a restart to hassio or deconz, these xiaomi door sensors disappear and never appear again even if i wait for more than 24 hours.
i have even paired a new out of box sensor, it works when paired but after a restart its gone never to be found again by conbee ii

**Logs:**
22:51:53:514 ZCL configure reporting rsp seq: 96 0x000D6FFFFE306520 for cluster 0x0006 attr 0x0000 status 0x00
22:52:03:062 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:52:07:821 Current channel 11
22:52:07:833 Device TTL 3636 s flags: 0x7
22:52:08:741 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:52:12:843 verified group capacity: 14 and group count: 2 of LightNode 0x00124b001856bbed
22:52:12:843 0x00124b001856bbed found group 0xFFF0
22:52:12:843 0x00124b001856bbed found group 0x0001
22:52:13:628 verified group capacity: 255 and group count: 2 of LightNode 0x000d6ffffea4ea65
22:52:13:629 0x000d6ffffea4ea65 found group 0xFFF0
22:52:13:629 0x000d6ffffea4ea65 found group 0x0001
22:52:14:299 delay sending request 71 dt 0 ms to 0x00124B001856C6DF, cluster 0x0004
22:52:14:354 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:52:15:132 verified group capacity: 255 and group count: 2 of LightNode 0x000d6ffffe306520
22:52:15:133 0x000d6ffffe306520 found group 0xFFF0
22:52:15:133 0x000d6ffffe306520 found group 0x0001
22:52:15:905 verified group capacity: 14 and group count: 2 of LightNode 0x00124b001e7319ee
22:52:15:906 0x00124b001e7319ee found group 0xFFF0
22:52:15:906 0x00124b001e7319ee found group 0x0001
22:52:16:632 verified group capacity: 14 and group count: 2 of LightNode 0x00124b001491002e
22:52:16:632 0x00124b001491002e found group 0xFFF0
22:52:16:632 0x00124b001491002e found group 0x0001
22:52:32:854 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:52:50:057 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:53:03:250 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:53:07:822 Current channel 11
22:53:07:834 Device TTL 3576 s flags: 0x7
22:53:25:563 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:53:49:961 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:54:07:819 Current channel 11
22:54:07:831 Device TTL 3516 s flags: 0x7
22:54:12:879 GW firmware version: 0x264a0700
22:54:12:879 GW firmware version is up to date: 0x264a0700
22:54:19:754 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:54:33:758 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:54:39:457 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:54:57:216 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:55:07:822 Current channel 11
22:55:07:833 Device TTL 3456 s flags: 0x7
22:56:07:821 Current channel 11
22:56:07:832 Device TTL 3396 s flags: 0x7
22:57:07:819 Current channel 11
22:57:07:829 Device TTL 3336 s flags: 0x7
22:58:02:851 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:58:07:824 Current channel 11
22:58:07:836 Device TTL 3276 s flags: 0x7
22:58:08:470 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:58:12:872 GW firmware version: 0x264a0700
22:58:12:872 GW firmware version is up to date: 0x264a0700
22:58:21:746 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:58:32:949 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:59:02:762 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:59:02:812 saved node state in 1 ms
22:59:02:841 sync() in 27 ms
22:59:07:823 Current channel 11
22:59:07:833 Device TTL 3216 s flags: 0x7
22:59:17:959 0x00124B001856C6DF error APSDE-DATA.confirm: 0xA7 on task
22:59:32:859 0x00124B001856C6DF error APSDE-DATA.confirm: 0xA7 on task
22:59:39:359 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:59:41:957 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
22:59:56:809 0x00124B001856C6DF error APSDE-DATA.confirm: 0xA7 on task
23:00:07:820 Current channel 11
23:00:07:831 Device TTL 3156 s flags: 0x7
23:00:09:116 Announced to internet http://dresden-light.appspot.com/discover
23:00:50:160 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:01:07:822 Current channel 11
23:01:07:833 Device TTL 3095 s flags: 0x7
23:02:07:820 Current channel 11
23:02:07:831 Device TTL 3035 s flags: 0x7
23:02:12:863 GW firmware version: 0x264a0700
23:02:12:863 GW firmware version is up to date: 0x264a0700
23:03:07:823 Current channel 11
23:03:07:835 Device TTL 2975 s flags: 0x7
23:03:27:778 ZCL attribute report 0x00124B001E7319EE for cluster 0x0006, ep 0x01
23:04:03:247 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:04:07:818 Current channel 11
23:04:07:829 Device TTL 2915 s flags: 0x7
23:04:10:018 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:04:29:060 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:04:57:753 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:05:07:821 Current channel 11
23:05:07:833 Device TTL 2855 s flags: 0x7
23:05:09:155 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:05:20:354 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:05:35:115 0x00124B001856C6DF error APSDE-DATA.confirm: 0xA7 on task
23:05:53:960 0x00124B001856C6DF error APSDE-DATA.confirm: 0xE9 on task
23:06:07:820 Current channel 11
23:06:07:831 Device TTL 2795 s flags: 0x7
23:06:12:870 GW firmware version: 0x264a0700
23:06:12:870 GW firmware version is up to date: 0x264a0700
23:07:07:817 Current channel 11
23:07:07:828 Device TTL 2735 s flags: 0x7
23:07:52:469 ZCL attribute report 0x000D6FFFFEA4EA65 for cluster 0x0000, ep 0x01
23:07:52:478 ZCL attribute report 0x000D6FFFFEA4EA65 for cluster 0x0000, ep 0x01
23:07:52:551 ZCL attribute report 0x000D6FFFFEA4EA65 for cluster 0x0000, ep 0x01
23:08:07:815 Current channel 11
23:08:07:826 Device TTL 2675 s flags: 0x7
23:08:27:756 ZCL attribute report 0x00124B001E7319EE for cluster 0x0006, ep 0x01
23:09:07:820 Current channel 11
23:09:07:831 Device TTL 2615 s flags: 0x7

**My setup:**
Running hassos 3.9 with version 0.105.3 on proxmox on a NUC.
Zigbee channel 11, Conbee version 2.05.72
I use mesh wifi on channel 6 for 2.4GHz and channel 40 for 5 GHz

bhushan0710, 2020-02-13:
Turns out to be issue with some specific zigbee smart plugs which dont play well with xiaomi sensors.
In my case the plugs were from Panasonic Vetaar.
every thing works well if the panasonic Vetaar zigbeel plugs are kicked out of the network.

closed 2020-02-13 by bhushan0710

eserero, 2020-10-20:
hi @bhushan0710 , can you explain how you identified the problem? I have the same problem now, every time I restart deconz, all my xiaomi mijia window/door sensors disappear from deconz, sometimes a couple come back, but most disappear forever. I had this working fine for months now, but I recently decided to move a few more sensors from my xiaomi hub to conbee (a few smoke sensors, temp sensors and window sensors) and now I have this problem.  The funny thing is then when I start deconz, all the devices immediately come up EXCEPT the mijia window sensors - if they appear at all, they will come after a few seconds.
The devices disappear from deconz, but not from the web app, and if I pair them again - they come back to the same place they were before on deconz,  but with the default name.

Anyone have any idea what to do?

bhushan0710, 2020-10-21:
Hi @eserero 
For me it was a zigbee plug which was causing this issue.
Some zigbee devices esp. thr ones which operate on 110v or 220v and act as repeaters do not play well with xiaomi sensors. If you have added such a device try removing it permanently from the network, Also if i may ask are you on latest firmware for conbee2? I believe with the latest version of firmware the conbee 2 has improved a lot.

eserero, 2020-10-29:
Thank you @bhushan0710 . I think I will open another issue on this. in my case, there is nothing that should be causing this, all my devices are xiaomi and they all worked perfectly well on the xiaomi gateway. the only ones that stop working are the mijia door sensors. the newer version of the sensors (the square ones) work just fine, but not the rounder sensors. 
Also the behavior is strange, they disappear from deconz, but not from the web app. I think it is something related to conbee, not the sensor itself