dresden-elektronik/deconz-rest-plugin #2442: Support for Wireless Temperature and Humidity Detector / Sensor

G3rry71, 2020-02-12:
I am running ConBee II, Deconz Addon in HASSIO and would like to know if this sensor is supported: https://www.aliexpress.com/item/4000578239423.html

[image]

Measured Value of Temp, battery and humidity looks correct in the attributes table.
but its still not possible to add it as a sensor in the phoscon web interface
[image]
[image]
[image]
[image]
[image]
[image]

Please let know how I can help!
Thanks

SwoopX, 2020-02-12:
You must read the attributes before taking the screenshots. Additionally, please provide one of the node info panel.

Darkwin101, 2020-02-12:
Ok I Try to help, i Have the same Sensor from Tuya
[image]
[image]
[image]
[image]
[image]

G3rry71, 2020-02-13:
Here they are, thanks
[image]
[image]
[image]

Sjonnie2018, 2020-02-13:
> Ok I Try to help, i Have the same Sensor from Tuya
 So you are saying the device is found and supported in Deconz REST api? I would like to know because I am thinking of buying the Blitzwolf version for my setup.

TIA

Darkwin101, 2020-02-13:
No it is not supported i want to help with the screenshots

Sjonnie2018, 2020-02-13:
Aha! Clear! I will postpone buying the sensor then😉

SwoopX, 2020-02-16:
Guys, can you please provide another screenshot of the power configuration cluster but with the attributes read? Thanks!

Darkwin101, 2020-02-17:
[image]

G3rry71, 2020-02-17:
Here is mine:
[image]

SwoopX, 2020-02-18:
@Darkwin101 Thanks! I'll provide a pull request soon.

WhistleMaster, 2020-04-19:
@G3rry71 How is this device running so far ? Any issue ? Could you share your opinion on that sensor ? Thanks !

mmychu, 2020-04-22:
I have the BlitzWolf version and is not discovered by Conbee2/deconz at this moment

Darkwin101, 2020-04-22:
I have no Problem with the Tuya and the Blitzwolf Version in Deconz. It is not showing in Phoscon App but there it is  not supportet at this Time

mmychu, 2020-04-22:
Any chance to see this sensor in Home Assistant?

Darkwin101, 2020-04-22:
i use openhab and there is no Problem with this Sensor since Version XXX...74

SwoopX, 2020-04-22:
@mmychu Maybe you should try to re-pair the device. It should be visible in HA like any other sensor, if pairing has been successful.

mmychu, 2020-04-22:
@SwoopX can you help me with pairing? I try keeping the top button for 5-10 seconds and don't see any new Node in VNC network. I always used PhosconApp discovery, first time doing by VNC.
Should I open network for new devices? How to do it in deconz GUI via VNC?

SwoopX, 2020-04-22:
Always pair via Phoscon. Start sensor search and reset the device. Success should be indicated by a green message.

mmychu, 2020-04-22:
@SwoopX works like a charm in Home Assistant. Thank you very much!

G3rry71, 2020-04-25:
> @G3rry71 How is this device running so far ? Any issue ? Could you share your opinion on that sensor ? Thanks !

Hi, it working correctly now! I can see it in HA and in VNC network. Still not showing up as a sensor in deCONZ Phoscon web interface.

Thanks a lot!

SwoopX, 2020-04-25:
Nice. 👍

Can we close the issue then?

mmychu, 2020-04-25:
Yes. Thank you

G3rry71, 2020-04-26:
Thanks for supporting

closed 2020-04-26 by G3rry71

SargonofAssyria, 2020-10-16:
I am using zigbee2mqtt and there is a bad update frequency of Temperature and Humidity.
Temperature is only updated if change is 0.7 degrees ore more and Humidity only if change is 2% or more.
I wonder if that is the same here and if not how it is solved.

G3rry71, 2020-10-16:
> @G3rry71 How is this device running so far ? Any issue ? Could you share your opinion on that sensor ? Thanks !

Hi @WhistleMaster , working very well, not an issue after months. I have temp, humidity and battery level in HomeAssistat

easybeat, 2020-11-20:
I've got them but they seem to have different Firmware compared to the Blitzwolf device. The temperature change is only reported at a change > 0.7 degrees. The Blitzwolf reports temperature changes of 0.1 degrees.

Is there somewhere this can be set/configured?

Thanks for help.

Beat

answer35, 2021-02-11:
> > @G3rry71 How is this device running so far ? Any issue ? Could you share your opinion on that sensor ? Thanks !
> 
> Hi, it working correctly now! I can see it in HA and in VNC network. Still not showing up as a sensor in deCONZ Phoscon web interface.
> 
> Thanks a lot!

Hi I have the same situation with one sensor. Any trick to fix it ? Because I have a second one that I bought first which was automatically detected in Phoscon, DeconZ, and HA but that new one cannot come visible on Phoscon.

SAOPP, 2021-02-24:
hey guys, I added it to my conbee 2, and I don't see it in phoscon, but this sensor is presens in gui and in ha, it's correct behaviour? 'coz I saw guys are posted  screenshots from phoscon with this sensor... any comments?

answer35, 2021-02-28:
> hey guys, I added it to my conbee 2, and I don't see it in phoscon, but this sensor is presens in gui and in ha, it's correct behaviour? 'coz I saw guys are posted screenshots from phoscon with this sensor... any comments?

Hi, I don't know exactly what can be wrong with this type of sensor but it is random issue, I have two sensors like that. First one showed up on Phoscon, second doesn't. But it is in HA & deCONZ GUI.

G3rry71, 2021-02-28:
Hello,
mine still working both in. Phoscon and HA.Only minor issue is it update
not so frequently.

Here it is in Phoscon under sensor:
VendorHeiman
ProductMultipurpose Sensor
Versionunknown
Battery100 %
Temperature20 °C
Humidity46 %
Refreshed02/28/2021 11:44:08 AM

HA:

Filtra entità

Filtra stati

Filtra attributi
 sensor.sensors_anticamera
<https://casagerry.no-ip.org:8124/developer-tools/state#> 20.6 on: true
unit_of_measurement: °C friendly_name: Temperatura Anticamera icon:
mdi:thermometer device_class: temperature
 sensor.sensors_anticamera_2
<https://casagerry.no-ip.org:8124/developer-tools/state#> 46.4 on: true
unit_of_measurement: % friendly_name: Umidità Anticamera device_class:
humidity

deConz:

[image: Schermata 2021-02-28 alle 11.51.01.png]

Hope this help.

Thanh-Binh, 2023-03-01:
I have the same sensor in ZigBee  but produced by Moes, but can not see it in HA.
Could you please help me by HA integration?
Thanks

Mimiix, 2023-03-01:
> I have the same sensor in ZigBee but produced by Moes, but can not see it in HA. Could you please help me by HA integration? Thanks

Please open an new request with requested information in the template.