dresden-elektronik/deconz-rest-plugin #2443: Sensors freeze / stuck / not refresh

IsraelShok, 2020-02-13:
Hi, I just connected my new Conbee before 3 days and paired a few Xiaomi sensors.
everything works great for a day.
After a day, all the sensors not refreshed and freeze in the same last status.
I connected to extension cable, and move it away from the router and raspberry, but with no success.

distance Conbee stick from sensors: 5-7 meters
stick channel: 15, tried to change the channel with no success.
version 2.05.72
firmware 264A0700
working with Hassio as deCONZ add-on

Please help me with this issue
Thanks

CHornhues, 2020-02-14:
Hello, I have the same Problem. Very annoying. Xiami Sensors. They show green and were seen in the Conbee App. But they don't update values... Batteries all new.. as devices are new.

[image]

[image]

What can we do? I think it is an firmware issue..

TuomasPakkanen, 2020-02-15:
Having the same kind of issue with a hue motion sensor.

xxLeoxx93, 2020-02-25:
Same here! But for me only the Sqare "Aqara" version seems to have this issue. The round version works well here so far.
Do you have a Xiaomi Gateway running at the same time?

CHornhues, 2020-02-25:
Nope. Aquara Gateway didn't work due to restriction of Servers. I removed all Aqara and zigbee devices. Replaced all of them with z-wave devices. No problems anymore.

xxLeoxx93, 2020-02-27:
Alright, I got Ikea Tradfri bulbs and they work great as a repeater. It just seems like the stick doesn't have a good long& stable range.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix