dresden-elektronik/deconz-rest-plugin #2444: deconz sends ColorControl command over and over again

krombipils, 2020-02-13:
Hi,
I'm new to Zigbee and I guess I don't understand a lot of the concepts. Nevertheless I'm trying to debug some problems (especially lost commands) with my simple Conbee Setup. Therefore I started deconz with dbg_aps=1 and I noticed an interesting behavior:

When I turn on my Osram Outdoor Lantern from the web interface, the command to the Color Control cluster is send over and over again (as long as the lantern is one).
I attached a short logfile: [deconz.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4199040/deconz.txt)

Initially a command to the Level Control cluster 0x0008 is send. Then a command to the Color Control cluster 0x0300 is send. This command is repeated as long as the lamp is on.

Is this behavior by design?

Thank you!

ebaauw, 2020-02-13:
I think what you’re seeing is the REST API plugin polling the light, by sending a _Read Attributes_ command.  You need to increase the debug level or use a sniffer to see the actual command.

The plugin only polls the _Level Control_ (0x0008) and _Color Control_ (0x0300) clusters when the light is on.  I’m not sure if you’ve filtered the log, but you should see a command to the _On/Off_ (0x0006) cluster and _Level Control_ cluster as well.  When the light is off, you only see _On/Off_ commands.

krombipils, 2020-02-13:
The log is not filtered. There is only an initial Level Control command followed by a Color Control command, No On/Off command.
But there is a  APS-DATA.indication (On/Off (0x0006))  cluster immediately after the  Level Control command.

I hanged the debug level to dbg_aps=2 and now I see more/other messages, but they do not make any sense to me, e.g there is a a command for the 0x031 cluster, but this cluster does not exist according to deconz:

`19:47:13:946 APS-DATA.request id: 67, addrmode: 0x03, addr: 0x7cb03eaa00ae47b6, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00`

[image]

Am I missing something?

ebaauw, 2020-02-13:
> But there is a APS-DATA.indication (On/Off (0x0006)) cluster immediately after the Level Control command.

OK, so the light supports attribute reporting, which apparently has been setup for 0x0006 and 0x0008, but not for 0x0300.

> there is a a command for the 0x031 cluster, but this cluster does not exist according to deconz

Profile 0x0000 is ZDP.  They encode the commands as clusters - best ignore these messages and focus on profile 0x0104 (ZHA) or 0xc05e (ZLL).  You should be seeing `asdu` lines in the log, after the APS-DATA lines.  These contain the message payload.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix