dresden-elektronik/deconz-rest-plugin #2445: Namron Zigbee dimmer 400W, Scene/Switch

olemr, 2020-02-13:
Hello.
I'm on:
[image]

Have a question regarding this device:
[image]
[4512700 INSTRUCTION MANUAL.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4201439/4512700.INSTRUCTION.MANUAL.pdf)

Just added it as a light, and it joined just fine after a power-up.
Can control the light just fine, but if I use the connected push(WAF)-switch turning the light on or off, the "on:" state takes 5-10s to be updated.

[image]

I see in the manual, that the device supports:
[image]

Would/should it be possible to add a switch/sensor device as well, with a quicker response?

ebaauw, 2020-02-13:
The delay is caused by the REST API plugin polling the lights for their state.  This looks suspiciously like another Sunricher OEM dimmer, so it should support attribute reporting.  This needs to be configured, so the dimmer sends updates on state changes, eliminating the delay.

The dimmer would need to be whitelisted in the code, for the REST API plugin to configure attribute reporting on pairing.  Best configure it manually in the GUI, see the user manual (under _Help_).  Bind the _On/Off_ and _Level Control_ clusters to the coordinator, and configure attribute reporting for _OnOff_ (1/300) and _Current Level_ (1/300/1).

> Would/should it be possible to add a switch/sensor device for the push-button as well, with a quicker response?

No, the connected switch isn't exposed over ZigBee.  Neither is the minimum dim level.

olemr, 2020-02-13:
Thanks.
Will try this during the weekend.
Will report back.

olemr, 2020-02-14:
This looks very promising!
Only things that didn't match your description 100% was the cluster names.
[image]

Thanks for helping out!

ebaauw, 2020-02-14:
> Only things that didn't match your description 100% was the cluster names.

The screenshot contains the **attribute** names of the _Level Control_ cluster (0x0008).  You need to set the reporting configuration for 0x0000, _Current Level_.  _On Level_ is the default level when you turn on the dimmer; value 255 means the previous value.  The _OnOff_ attribute (also 0x0000) is in the _On/Off_ cluster (0x0006).

> Will the two bindings I just performed be part of a backup? (where is it stored?)

On the dimmer, in non-volatile memory.

olemr, 2020-02-15:
A bit confusing the first time around.
Just discovered that the config I did yesterday har left the device uncontrollable.
Trying once more :-)

olemr, 2020-02-15:
After a factory reset, I got it working.
For other newbies (like me) coming here, and who are also a bit unsure even after reading the manual I will recap what I did.
Binding cluster 0006 & 0008 drag them to the Source, and the 01 cluster of the controller to the target. Bind them both:
[image]

For cluster 0006, double click on On/Off and set values 1/300
[image]

For cluster 0008 double click on 'Current level' and set attributes to 1/300/1:
[image]

Viola!

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix