dresden-elektronik/deconz-rest-plugin #2447: How to change device id in the Web UI or via Rest API

Liese2007, 2020-02-14:
Hi,
in Home Assistant I get following error after reconnecting a sensor (Aqara Window Open/Close):
`Platform deconz does not generate unique IDs`

How can I access the IDs ? Via a Database file or via the Rest API?

Best regards,
Jörg

Smanar, 2020-02-14:
Hello,
Can you take a look on the JSON of the device ? I think you can find it in HASS log.

I can't understand if the device miss an uniqueID or if you have 2 time the same.

Liese2007, 2020-02-15:
Sorry @Smanar , I am a Noob.... Just started in December. May you explain where a JSON output of the device can be found?

The error log shows the following kind of fault several times:
`
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 415, in _async_add_entity
    raise HomeAssistantError(msg)
homeassistant.exceptions.HomeAssistantError: Entity id already exists: binary_sensor.openclose_21. Platform deconz does not generate unique IDs
2020-02-14 09:49:13 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 415, in _async_add_entity
    raise HomeAssistantError(msg)
homeassistant.exceptions.HomeAssistantError: Entity id already exists: binary_sensor.water_5. Platform deconz does not generate unique IDs
2020-02-14 09:49:13 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
`
Does the Log extraction help?

Regards,
Jörg

Smanar, 2020-02-16:
Not realy for me ^^.
You don't have specific log for the deconz plugin ?
Or tale a look here https://dresden-elektronik.github.io/deconz-rest-doc/
You can have a JSON list with all your sensors with something like that
`http://ip_deconz:por_tdeconz/api/api_key/sensors`

And you can see the guilty.

Or you can try at random with removing sensor from Phosocn.

Liese2007, 2020-02-16:
I tried to get access to RestAPI before, but the GATEWAY UNLOCK button is not available in my Phoscon Frontend.

Any hint :-)
Or need to open othe rissue maybe?

Regards,
Jörg
[image]
[image]
[image]

Smanar, 2020-02-19:
Oups sorry I have forget you.
I have see someone else have answer to you https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2469

And yep, sorry the site is not in date.

But it s not the easier way, I think you can find an API key somewhere in the plugin config in HA.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

Davst, 2021-05-27:
I'm having this same issue and looking for a sollution.

Smanar, 2021-05-27:
What is your issue ?