dresden-elektronik/deconz-rest-plugin #2448: New Device Support Request: SALUS TRV10RFM

kaihen, 2020-02-14:
I got some stuff  from the vendor [SALUS](https://salus-controls.com/) (see at the bottom). It would be great if you could support the TRV10RFM.

Cluster Info - Basic
[image]

Cluster Info - Identify
[image]

Cluster Info - OTAU
[image]

Node Info
[image]

The Log shows:
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4206602/log.txt)

I don't know if it helps but I also found following file at my research for the TRV:
[SAL2EM1.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4206587/SAL2EM1.txt)

The following products counts also to my inventory:
- UGE600
- 5x TRV10RFM
- 5x VS20WRF
- OS600
- 2x SW600

SwoopX, 2020-02-15:
Generally, the device can be supported but I think, you'll not get happy with it. Based on the screenshots showing the available clusters, there is no therostat cluster available. This kinda fits into what I've previously read about the devices not having a local temperature sensor for valve adjustments. That means they'd always rely on an external sensor...

I'd assume this is where the manufacturer specific cluster FC00 comes into play. Unfortunately, we don't know which attributes it has and what they are for. This means they need to be reverse engineered by sniffing and analyzing the zigbee traffic, preferably with the vendor gateway available.

Sorry for not having better news.

kaihen, 2020-02-17:
Would it help if you got the firmware/update file for the Universal Gateway (UGE600)?

SwoopX, 2020-02-18:
Not really. It's communication what's important.

deennoo, 2020-02-22:
> Would it help if you got the firmware/update file for the Universal Gateway (UGE600)?

I have this device too, and we try to add them to Zigate.

Main probleme is cluster specific, Salus (computime) use is how language on Zigbee. even on connection Salus made thing very difficulte.

SwoopX, 2020-06-18:
Closing this due to incomplete information. Please re-open if required.

closed 2020-06-18 by SwoopX

benkuly, 2021-05-18:
Maybe this two links could help, where some guys sniffed the communication between SALUS devices:
https://github.com/zigpy/zha-device-handlers/issues/571
https://github.com/Koenkk/zigbee-herdsman-converters/issues/1524

Mimiix, 2021-05-18:
@benkuly Thanks for sharing this information. However: As Swoop mentioned we are missing some info from deCONZ which is needed to add this device.