dresden-elektronik/deconz-rest-plugin #2449: Cleanup of supportedDevices in de_web_plugin.ccp

SwoopX, 2020-02-14:
I'm just wondering if we shouldn't shorten the list of supported devices a bit in that sense that we shape the entries more general (as it has already been done occasionally).

As an example, we have quite a number of supported Xisomi devices, all beginning with "lumi". Following the proposed, we could reduce the number of entries from somewhere around 25 to maybe 6, focussing on the different vendors and mac prefixes. Benefit would be that a new entry would not be required any more, if a device has, let's say, a country extension (which can be the case for lumi.plug devices). Under the assumption, other parts of the other parts of the code are equally written, it could potentially lower implementation efforts a bit while broadening initial support for other devices. It might also solve one or the other issue some users experiencing.

Also, we only have one place in the code where that list is checked or picked up respectively. There, a more generic check on the modelId is conducted anyway through "startsWith", so no real need to be that explicit anyway. Of course, it might be required to stay as explicit from the modelId perspective as it currently is elsewhere in the code.

What do you think?

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/f1fb5f9acd1f748a6d86f6b7dd8464244949b0cf/de_web_plugin.cpp#L7238-L7260

Smanar, 2020-02-15:
For me it depend of "API V2".
Not realy usefull to edit the code and have a chance to make a mistake if we can start to worjk on a new API in some months.

BUT, ATM, not a lot of news about it ...

ebaauw, 2020-02-15:
The problem is not the whitelisting of devices in `supportedDevices[]`.  If it were, we could simply skip the check altogether.  Unfortunately, adding a device to `supportedDevices[]` does **not** automagically add support for the device.

The problem is the lack of standardisation how devices use the ZigBee standard, and/or devices simply deviating from the standard.  These exceptions are handled "everywhere" in the code, resulting in the "there's always one more place to whitelist a device" experience.

API v2 should indeed do a re-design from the ground up, so that devices incl. all their capabilities and quirks only need to be whitelisted in one place.  Preferable in a JSON file, so we would no longer need code changes to support new devices with capabilities already supported.

SwoopX, 2020-02-15:
> The problem is not the whitelisting of devices in supportedDevices[]. If it were, we could simply skip the check altogether. Unfortunately, adding a device to supportedDevices[] does not automagically add support for the device.

Yeah, I now. However, if I understand the code correctly, it's relevant to enable sensor creation. It's just one piece of the puzzle but could slightly decrease the number of raised issues though.

> The problem is the lack of standardisation how devices use the ZigBee standard, and/or devices simply deviating from the standard. These exceptions are handled "everywhere" in the code, resulting in [...]

Couldn't agree more. I wonder why there's not any activity around that...

closed 2020-06-05 by SwoopX