dresden-elektronik/deconz-rest-plugin #2450: Deconz GUI not autostarting

daniel-lo, 2020-02-15:
Hi all, sometimes I notice that deconz is not running .. this is also the case after a reboot of the image. I then connect to the desktop via tightVNC and start the deCONZ app .. after that all works well.

I'd like deCONZ-gui to start automatically. I went through the instructions with sysctl but - if that is the way - something went wrong. Don't know too much about Linux.

Or should I rather put the deCONZ app is some kind of autostart folder like on Windows?

ebaauw, 2020-02-15:
See #1736, near the end.  deCONZ crashes when started before the graphical environment is running.  Use a timer to delay the startup after boot.

daniel-lo, 2020-02-15:
@ebaauw 

Thanks .. you mean Nastra's comment which ends with:
sudo systemctl enable deconz-conbee-reset.timer

or is it your "override.conf" thing?

ebaauw, 2020-02-15:
No, it’s the timer.  It’s actually in #1535.

daniel-lo, 2020-02-15:
So I did this with 1min and 2min:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1535#issuecomment-520122255
and rebooted .. but phoscon app will still not connect (neither my iobroker deconz instance). So I figure the gui service is not started.

Smanar, 2020-02-15:
On my side I create a file ~/.config/autostart/deCONZ.desktop

ebaauw, 2020-02-15:
> So I figure the gui service is not started.

Did you verify this over VNC?

Make sure the timer is enabled and the `deconz-gui` and `deconz` services are disabled (using `sudo systemctl enable` or `disable`).

> neither my iobroker deconz instance

Don't know how that one, but does it retry automatically when deCONZ isn't available?  It will be started way before deCONZ with the timer.

daniel-lo, 2020-02-15:
@Smanar could you elaborate on the content of deCONZ.desktop ?

daniel-lo, 2020-02-15:
@ebaauw 

yes .. as soon as I start the app via vnc/gui it works again.

systemctl only has deconz-update.service listed.

After entering "sudo systemctl enable deconz-gui.timer" I cannot find anything related to that in the systemsctl status list.

Something is wrong :-(

And yes .. iobroker retries by itself.

daniel-lo, 2020-02-15:
I use LXDE .. what seems to work is:

nano ~/.config/lxsession/LXDE/autostart

Add this to end:
@deCONZ

daniel-lo, 2020-02-15:
@ebaauw 
@Smanar 

THANKS!!!

closed 2020-02-15 by daniel-lo