dresden-elektronik/deconz-rest-plugin #2451: REST API with curl

cautious-bison, 2020-02-15:
Hi,

With the API, I can query states, but I cannot have any action done.
My set-up is done with a conbee 2 controller and a ikea Tradfri switch outlet. The outlet is in group 1.
From the website, the switch on/off of the group 1 is ok, so I am suspecting that only my call with curl is faulty here.

Here is the code I am using and is fine as it only queries :

```
API_KEY='sDHEUFRsdhfdkusflSDFSDF'`
API_ADDR='http://127.0.0.1'

curl -POST -d '{"username": "'"${API_KEY}"'", "devicetype": "my application"}'  ${API_ADDR}/api

# Get info
curl ${API_ADDR}/api/${API_KEY}/config

# Get info for groups
curl ${API_ADDR}/api/${API_KEY}/groups
curl ${API_ADDR}/api/${API_KEY}/groups/1 

curl -PUT -d '{"on" : true}' ${API_ADDR}/api/${API_KEY}/groups/1/action
```
Here is the request and the answer
```
curl -PUT -d '{"on" : true}' ${API_ADDR}/api/${API_KEY}/groups/1/action
curl -PUT -d '{"on" : false}' ${API_ADDR}/api/${API_KEY}/groups/1/action

[{"error":{"address":"/groups/1/action","description":"method, POST, not available for resource, /groups/1/action","type":4}}]t
```

I am probably doing something wrong and missing a tiny bit of information that will make my calls work ! Any pointer will be super appreciated.

ebaauw, 2020-02-15:
The error says method POST?!

cautious-bison, 2020-02-15:
Yes.
And the error is the same with a PUT method.

Smanar, 2020-02-15:
You have an error message about POST when using PUT ?

cautious-bison, 2020-02-15:
Yes, I do have this error whether it's POST or PUT.

I am running the server from docker (following this installation  [procedure](https://phoscon.de/en/conbee/install#docker) ).

ebaauw, 2020-02-15:
See `man curl`: you need to specify `curl -X PUT` or `curl -X POST`.

cautious-bison, 2020-02-15:
Thx !
That was the bit I was missing !

closed 2020-02-15 by cautious-bison