dresden-elektronik/deconz-rest-plugin #2453: Add support of GS SKHMP30-I1 smart plug (manufactured by Heiman)

andreypolyak, 2020-02-15:
Hi team,
Recently I bought [GS SKHMP30-I1](https://www.gs.ru/catalog/umnyy-dom/umnaya-rozetka-gs-skhmp30-i1/) smart plug which is only sold in Russia by one of the satellite TV operators. This smart plug is in fact produced by Heiman and is a copy of [Heiman's smart plug](http://www.heimantech.com/product/139.html). However, it has a different manufacturer and model id in the Basic cluster.
I was able to add this smart plug but it only supports On/Off commands, no power metering is available (I'm using Home Assistant).

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-02-16:
What load was applied when you took the screenshot, 388W or 38,8W?

andreypolyak, 2020-02-16:
Hi @SwoopX,
It was 38,8W

andreypolyak, 2020-02-27:
Hi @SwoopX ,
I updated to 2.05.74, but still can't see power usage neither in Home Assistant nor in Phoscon. I have deleted smart plug from Phoscon and then added it again. Something wrong?

Thanks!

SwoopX, 2020-02-27:
Hm, not really what I expected. Could you please try once again, but this time after deleting the device in Phoscon also physically reset the device? Meanwhile, I double check the code if something went wrong.

andreypolyak, 2020-02-27:
Hi @SwoopX,
please ignore my last message. I restarted HA and deConz, deleted the device from Phoscon once again and Power and Consumption sensors appeared in Home Assistant. Thank you so much!

SwoopX, 2020-02-28:
Could you please double check the consumption value is reasonable? Might be that I made a mistake and set the comma wrongly.

andreypolyak, 2020-02-28:
Hi @SwoopX ,
to be honest, I don't understand how consumption value is calculated. Should it show consumption over the last hour? Or real-time consumption? For test purposes, I put an LED bulb to the smart plug. The bulb is marked as 23W and that's approximately what I get as power value (25-26W). However, consumption value is shown as 0.134-0.154 kWh. 

Thanks

SwoopX, 2020-02-28:
It is the total power consumed since last device reset, so it should constantly grow over time. With the LED, it takes some minutes for the value to change.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix