dresden-elektronik/deconz-rest-plugin #2454: Can't pair any devices + Firmware: Not Connected

xxLeoxx93, 2020-02-15:
Hi there,
I'm running a Pi4 headless with the conbee stick 2 on a USB 2 port with an extention cable. Domoticz, PiHole, and OMV are also used on the pi if that matters.
I did the configuration as stated in the setup instructions. I also dissabled the gui version.
Now the webinterface is working fine so far and I can see the gateway. However I've two problems:
1) the firmware is always saying "not connected" to some reason
2) I can't pair anything. Now I know the stick is working because I tried it on my windows machine -> Worked perfectly.
3) I can't swich the channel (also working on windows).

when I run sudo systemctl edit  deconz I get this:
`# /etc/systemd/system/deconz.service.d/override.conf
[Service]
ExecStart=
ExecStart=/usr/bin/deCONZ -platform minimal --auto-connect=1 --http-port=8085 --dbg-error=1 --dbg-info=2
`

Any ideas what could be the reason?

Smanar, 2020-02-15:
Hello.
When you have the firmware with "not connected" can you try this command pls (to check if you haven't too many instance running)
>ps ax | grep deCONZ

and to check hardware connexion
>GCFFlasher_internal -l

xxLeoxx93, 2020-02-16:
Hi, thanks for your repy.
`root@raspberrypi:/home/pi# ps ax | grep deCONZ
  774 ?        Ss     0:05 /bin/bash /usr/bin/deCONZ-update2.sh
  804 ?        Ss     0:02 /bin/bash /usr/bin/deCONZ-WIFI2.sh
  805 ?        Ssl    1:29 /usr/bin/deCONZ -platform minimal --auto-connect=1 --http-port=8085
13112 pts/0    S+     0:00 grep deCONZ
` --> Only one instance assuming I understood the result correctly.

`root@raspberrypi:/home/pi# GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2149599  | ConBee II
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee
`

Smanar, 2020-02-16:
I think 2/3% is something normal, I have same with GUI version, and I don"t see anything strange on your report.
Retry the first command the next time, you have the firmware "not connected" again.

As you already have tried other tips (USB2, cable extention) I don't have more ideas. And it works fine on Windows .... A power supply problem ?

Have you the last firmware version ?

Perhpas if you can have some deconz log for another person ? It's strange it work with command line but not with service.

xxLeoxx93, 2020-02-16:
Im running it on a extensioncable + USB2 already and I was able to pair a sensor once. So I guess it should work in general.
I saw also in the log that when I try to swich the channel from 20(currently) to 15 or any other I get a sucess message in the webinterface but the log will then post every 5 seconds somthing like "Channel 20, should be 15, initializing channel switcher".

manup, 2020-02-16:
Does the issue still persist with deCONZ version 2.05.73?

xxLeoxx93, 2020-02-16:
@manup What issue? I've installed this version and like I said I can see the Firmware information now but pairing still doesn't work at all.

Smanar, 2020-02-16:
You crash a third application when using deconz ?
And if you make tries with only deconz, it works ?
Can you make a tries with debug enabling on deconz (paring your device) ?

xxLeoxx93, 2020-02-16:
@Smanar 
I'm using 2.05.73. How can I enable debugging for pairing?
I've stopped domoticz but still no luck.
I've also changed the ports again but Domoticz still crashing when I search for a sensor in Deconz :(

Smanar, 2020-02-16:
With sometking like 
>deCONZ -platform minimal --auto-connect=1 --http-port=8085 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1

Take care lot of logs with this command line.

It's out of context, but fastly, domoticz crash when you start it (when the plugin start), or when you (try to) add a device in phoscon ?  I realy don't understand how one can interfere with the other (just using REST command).

xxLeoxx93, 2020-02-16:
Hey, Domoticz crashes when I try to add a device, otherwise it runs fine.
While searching for a reason I found this issue:
[(https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1699)]
There the port only changed when removing -platform minimal. I tried that but it doesn't start at all, 
`root@raspberrypi:/home/pi# deCONZ --auto-connect=1 --http-port=9595 --ws-port=9696`
`qt.qpa.screen: QXcbConnection: Could not connect to display`
`Could not connect to any X display.`

When I start deCONZ manualy I also see "PROTO: CRC error" sometimes and starting/stoping the service gives me a "The unit file, source configuration file or drop-ins of deconz.service changed on disk. Run 'systemctl daemon-reload' to reload units.". Not sure if that is related to my pairing issue.
I'll try with debugging later and report back!

SwoopX, 2020-02-16:
When you make any changes on the systemd files, then a 'systemctl daemon-reload' is required. Nothing unusual, just do it and that's it.

The error you mention is typically related to a headless Raspbian. Is it so in your case? If yes, check out the wiki on how to connect anyway.

Lastly, I've seen the "PROTO: CRC error" thingy in some posts over time here, which seems to be an unresolved spoiler causing troubles.

xxLeoxx93, 2020-02-17:
But do I need the GUi for pairing? I thought that the Webinterface is enough.

SwoopX, 2020-02-18:
Well, with the GUI, you could try to put it in a network. Generally, it would give you a couple of more options at hand.

xxLeoxx93, 2020-02-19:
I guess I'll plug the pi into my monitor on friday, then I can check those settings aswell. Thanks for the hint!

xxLeoxx93, 2020-02-23:
Alright, pluged-in to my monitor. Anything special I should check for?

Smanar, 2020-02-23:
For the moment make tries without domoticz running.

Have you another machine to test ? For exemple a windows machine ?

Have you now a firmware displayed on phoscon/gateway ?

xxLeoxx93, 2020-02-23:
Hey there,
on windows i was able to pair a light and sensor strait away without any issues.
The issue with the firmware is also not there anymore with my rasberry (264A0700).
Intrestingly I also managed to pair a motion sensor with my rasberry, however only when I start it directly, not as a service and no motion is reported back afterwards (works fine on my windowsmachine). I also started with "--dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1", however i get so much reported that I'm not sure what I'm looking for.

Smanar, 2020-02-23:
So to resume it works if you delay the deconz startup ?

xxLeoxx93, 2020-02-23:
Actualy no. So I stoped my manualy started deconz and started the deconz service (1) -> After 5 min I still can't change my lights at all. 
Stopping the service and excecuting the same line from the service script directly (2) -> works right away!

1)
`systemctl edit deconz`
`# /etc/systemd/system/deconz.service.d/override.conf`
`[Service]`
`ExecStart=`
`ExecStart=/usr/bin/deCONZ --auto-connect=1 --http-port=9595 --ws-port=9696 -platform minimal`

2)
`deCONZ --auto-connect=1 --http-port=9595 --ws-port=9696 -platform minimal`

Not sure why 1) isn't working at all and 2) is.
Regarding the sensor: I realized that it's just delayed by ~15 sec. Not sure thou if that is cause by the sensor/interference.

xxLeoxx93, 2020-02-24:
**Update, I think I found the issue: The changed ports aren't working probably.**
Today I rebooted with everything dissabled. Then started deconz manualy (like in 2 above). The webapp appears on port 9595. Then I started domoticz (port 8080) and it failed beacuse port 8080 was already in use. So my guess is that the port settings in decons aren't working as they should!

Smanar, 2020-02-24:
Ha ?
Realy bad ....

Have you tried to use

`sudo netstat -nltup`

to be sure ?

xxLeoxx93, 2020-02-25:
That's the result:
```
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:9696            0.0.0.0:*               LISTEN      29355/deCONZ
tcp        0      0 0.0.0.0:6144            0.0.0.0:*               LISTEN      7792/domoticz
tcp        0      0 127.0.0.1:4711          0.0.0.0:*               LISTEN      880/pihole-FTL
tcp        0      0 0.0.0.0:139             0.0.0.0:*               LISTEN      613/smbd
tcp        0      0 0.0.0.0:5355            0.0.0.0:*               LISTEN      334/systemd-resolve
tcp        0      0 192.168.178.55:5357     0.0.0.0:*               LISTEN      602/python3
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      609/lighttpd
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      880/pihole-FTL
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      566/sshd
tcp        0      0 0.0.0.0:8090            0.0.0.0:*               LISTEN      7792/domoticz
tcp        0      0 0.0.0.0:9595            0.0.0.0:*               LISTEN      29355/deCONZ
tcp        0      0 0.0.0.0:445             0.0.0.0:*               LISTEN      613/smbd
tcp6       0      0 ::1:4711                :::*                    LISTEN      880/pihole-FTL
tcp6       0      0 :::5355                 :::*                    LISTEN      334/systemd-resolve
tcp6       0      0 :::139                  :::*                    LISTEN      613/smbd
tcp6       0      0 :::80                   :::*                    LISTEN      609/lighttpd
tcp6       0      0 :::21                   :::*                    LISTEN      900/proftpd: (accep
tcp6       0      0 :::53                   :::*                    LISTEN      880/pihole-FTL
tcp6       0      0 :::85                   :::*                    LISTEN      585/nginx: master p
tcp6       0      0 :::22                   :::*                    LISTEN      566/sshd
tcp6       0      0 :::445                  :::*                    LISTEN      613/smbd
udp        0      0 0.0.0.0:9898            0.0.0.0:*                           7792/domoticz
udp        0      0 0.0.0.0:35549           0.0.0.0:*                           359/avahi-daemon: r
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           359/avahi-daemon: r
udp        0      0 0.0.0.0:5355            0.0.0.0:*                           334/systemd-resolve
udp        0      0 0.0.0.0:38715           0.0.0.0:*                           602/python3
udp        0      0 127.0.0.1:323           0.0.0.0:*                           549/chronyd
udp        0      0 0.0.0.0:1900            0.0.0.0:*                           29355/deCONZ
udp        0      0 0.0.0.0:53              0.0.0.0:*                           880/pihole-FTL
udp        0      0 192.168.178.55:68       0.0.0.0:*                           179/systemd-network
udp        0      0 0.0.0.0:68              0.0.0.0:*                           485/dhcpcd
udp        0      0 239.255.255.250:3702    0.0.0.0:*                           602/python3
udp        0      0 192.168.178.255:137     0.0.0.0:*                           596/nmbd
udp        0      0 192.168.178.55:137      0.0.0.0:*                           596/nmbd
udp        0      0 0.0.0.0:137             0.0.0.0:*                           596/nmbd
udp        0      0 192.168.178.255:138     0.0.0.0:*                           596/nmbd
udp        0      0 192.168.178.55:138      0.0.0.0:*                           596/nmbd
udp        0      0 0.0.0.0:138             0.0.0.0:*                           596/nmbd
udp6       0      0 :::5353                 :::*                                359/avahi-daemon: r
udp6       0      0 :::5355                 :::*                                334/systemd-resolve
udp6       0      0 ::1:323                 :::*                                549/chronyd
udp6       0      0 :::33614                :::*                                359/avahi-daemon: r
udp6       0      0 :::53                   :::*                                880/pihole-FTL
```

Are you running Domoticz on port 8080 and use arleady port 80 for something else?

SwoopX, 2020-02-25:
Looks ok from a used ports perspective. Are you sure you previously haven't mixed up port 8080 with 80?

Port 80 is occupied by the pihole webserver (lighttpd)...

Smanar, 2020-02-25:
And you have nothing on port 8080.

On my side I have 
```
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      864/deCONZ
tcp        0      0 0.0.0.0:5900            0.0.0.0:*               LISTEN      448/vncserver-x11-c
tcp        0      0 0.0.0.0:8088            0.0.0.0:*               LISTEN      864/deCONZ
tcp6       0      0 :::5900                 :::*                    LISTEN      448/vncserver-x11-c
tcp6       0      0 :::5008                 :::*                    LISTEN      864/deCONZ
tcp6       0      0 :::8080                 :::*                    LISTEN      473/domoticz
tcp6       0      0 :::443                  :::*                    LISTEN      473/domoticz
tcp6       0      0 :::6144                 :::*                    LISTEN      473/domoticz
udp        0      0 0.0.0.0:1900            0.0.0.0:*                           864/deCONZ

```

xxLeoxx93, 2020-02-27:
No, domoticz defnitly did run on 8080. If i select 8080 for domoticz again and try to add a device in Phoscon, domoticz still crashes. I can repeate the issue over and over again so I don't thing it can be a unlucky moment. 
It works perfect not with Domoticz on port 8090 so for me it's fine. Let me know if I should still try out something else fo find the root-cause if it helps :)

SwoopX, 2020-02-27:
Does it instantly crash? Maybe you got the time to check the ports again before it crashes and post the output here. Obviously, there must be a conflict somewhere...

xxLeoxx93, 2020-02-27:
Tried that a minute ago, but it crashes so fast and the output looks always the same.
I'll do some more investigating this weekend. What's still not working correctly is running it as a service instead of excecuting it directly (in a screen window so I can close my SSH session and it keeps running).

SwoopX, 2020-02-27:
To be honest, I wonder a bit about the file which is opened in your very first post. I'm used that service definition files are found here: "/lib/systemd/system/deconz.service"

This is my output:

```
swoop@skynet:~ $ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-02-25 21:36:50 CET; 2 days ago
 Main PID: 5018 (deCONZ)
   CGroup: /system.slice/deconz.service
           └─5018 /usr/bin/deCONZ -platform minimal --http-port=8080 --ws-port=8888
```

closed 2020-04-26 by xxLeoxx93