dresden-elektronik/deconz-rest-plugin #2455: HELP!!!! Aqara TW LED (ZNLDP12LM) not connecting to Deconz

ravet007, 2020-02-16:
I have 2 Aqara TW LED (ZNLDP12LM). A conbee II plugged into the Rasp 3b+. I've successfully added 3 sensors and all work fine.  
The Aqara lights I cannot connect whatever I try. I reset the lights, and turned them 5x off/on till I get 3 blinks, but still no luck. 
Someone please help or point me in the right direction. 
Thanks

ravet007, 2020-02-16:
Solved: I turned on the Gateway and Mi app and selected the bulb and it appreared in deConz. One that is. 2nd one I'm still trying.

ravet007, 2020-02-16:
Now solved, restart of Hassio needed every time a device is found and added. It will also not find until a restart has been done. Strange but a restart before/after worked.

Mimiix, 2020-06-05:
As it is solved, i'll close this issue :)

closed 2020-06-05 by Mimiix